rbaron/b-parasite #2: Moisture readings overflow

AlexSzlavik, 2021-05-12:
I haven't dug into this myself yet, but noticed this morning that one of my plants moisture reading "overflowed". When looking at the code, I saw that moisture readings are actually fit to a polynomial and the final values from the equation. Haven't done any debugging yet, so probably not much to go on, but opening the issue as something is clearly miss-behaving. Will follow up as I get time. 
I'm also not convinced that the moisture reading is right in the first place. It's continuously trending up, yet I haven't added water to the soil since adding the sensor. 

Love this project, thanks a bajillion for open sourcing. 

[image]
[image]

rbaron, 2021-05-12:
Hi @AlexSzlavik,

It's super cool that you made some boards yourself!

The overflow issue should have been fixed in [this commit](https://github.com/rbaron/b-parasite/commit/f0b3294f0f77db032b2625e03c2066ef8001ea5c). The bug was multiplying the percentage by 2 << 16 instead of (2 << 16) - 1, so it could overflow an `uint16_t`. Do you know if you're running a version that has that commit?

About the "initial upwards trend", this is a phenomenon I also experienced with a couple of boards (and so did a colleague to whom I gifted one). For example, check out the first data I have from when I initially put a few sensors in the ground:

[image]

(The orange curve is even sensitive to the air humidity through the day!)

This is a little noisy, but there's definitely some initial upwards trend there. I don't know for sure why, but some hypotheses are:
* Moisture penetrating in the solder mask layer tiny pores
* Some water capillarity/surface tension that causes water droplets to start accumulating on the sensor surface
* Simply the soil accommodating itself around the sensor after being compressed by inserting the board, increasing the "contact" it makes with the board

Another interesting data point is that I recently deployed two new sensors, but I coated them completely (even over the components, except the temperature sensor) with [this protective spray (PLASTIK 70)](http://www.kontaktchemie.com/koc/KOCproductdetail.csp?division=&product=PLASTIK%2070&ilang=en&plang=en). There I haven't observed this initial upwards trend (the yellow bump is when I watered it a little bit):

[image]

I'll try and replicate this again with new coated boards with different types of soil. I'd also love to see how your sensor behaves in the next days/weeks.

Thanks!

AlexSzlavik, 2021-05-13:
Fantastic response, thank you! 
Sorry about posting without the patch, I was too excited when I finally put it together that I forgot to pull the latest 🤦 . Will run latest and check again. 

As for the upward trend, I suspect you are onto something with capillarity effect. When I pulled the sensor out it was noticeably "wet". That is, far more wet than I'd have expected on this particular plant. I've attached a couple images to show the level of "wet" dirt stuck to the capacitance section. I also think that there may be some issues with my soldermask. The second image clearly shows "bubbles" forming under the soldermask, likely moisture build up. I have some conformal coating I could apply around the base to waterproof it, but I'm worried what the impact on the capacitance reading will be. 
[image]
[image]

rbaron, 2021-05-13:
The board looks super cool in black! Thanks for the photos.

I've also seen this surprising moisture accumulation around the sensor and that got me thinking about this effect.

Those bubbles I haven't seen, but I think it's one more reason for us to use a layer of conformal coating. I've tried to recalibrate the sensor after applying the spray and I found the difference to be negligible. I'm still using the same values for both coated and uncoated boards. I imagine the added layer is so thin that the variation in the dielectric is super tiny.

It seems the location of the sensor (closer/further from the root) and specially how much contact it makes with the soil have greater effect on the read values than the protective coating.  Still, I think even those greater effects are okay for the level of precision we are aiming for here.

Let me know if you end up coating the boards - curious to learn about your findings!

hubikj, 2021-05-13:
I read an interesting article about capacitive soil moisture sensors and the author recommends sealing the PCB edges with epoxy:
[image]
Source: https://thecavepearlproject.org/2020/10/27/hacking-a-capacitive-soil-moisture-sensor-for-frequency-output/

rbaron, 2021-05-13:
Thanks for sharing @hubikj. That's a good reference.

Before using the conformal coating spray, I was applying a layer of transparent nail polish to the sides of the PCB. That seemed to do the trick, but epoxy/conformal coating seems to be the best solution.

AlexSzlavik, 2021-05-13:
Thanks for the discussion and the resources. I've applied a thing film of silicon based conformal coating to the edges and the flat surface. There was no noticeable change in sensitivity (as you said, on the scale we care about). I've replaced the sensor and ran it overnight. I still see some initial creep, but the rate is definitely lower. I'll leave it for a day or two and come back with the results.

MrEinstin, 2021-05-25:
I have been wondering about the effect of conformal coatings on at least the lower sensor part of the pcb as far as any negative effects in it's measurements. Also, some coatings are more prone to moisture ingress than others. There are also other factors to consider if you were to coat the board entirely as far as reparability. Obviously some areas need to be masked so that they aren't coated (humidity sensor, battery holder, programming pads). I happened to run into this comparison from VSI. https://vsiparylene.com/2020/09/10/the-essential-conformal-coating-comparison-guide/

AlexSzlavik, 2021-07-02:
So I changed the coating to a Polyurethane based coat. This has been working really well! I haven't tried the silicon based one again since, but I think I may have also messed up the initial curing process on that coating. Eitherway, as was pointed out in this thread, coating the edges is absolutely key.

closed 2021-07-02 by AlexSzlavik

rbaron, 2021-07-03:
I'm really glad to hear that @AlexSzlavik. Thanks for coming back and updating the thread!

chucknorris101, 2022-08-14:
Thanks for pointing me this way @rbaron! 
@AlexSzlavik For the poly coating, is it just the edges or giving it a nice full dip coat?

AlexSzlavik, 2022-08-15:
I coated the entire device which worked alright. @rbaron also wrote a blog post about how he did his coating which was great!