rbaron/b-parasite #3: How to order/Build

Flipez, 2021-05-18:
Hi and thanks for all the work!

I am interested in building some of them myself. Looks like I have to order pretty much everything myself. Could you give some hints/links on how to do that?

I've worked with a lot of sensors and smaller electronics in the past but never actually build/order my own pcbs. So I am a bit lost on that.

Thanks in advance!

drewzarn, 2021-05-18:
Seconding this somewhat, even a screenshot of what options to pick in the JLCPCB order form would be huge! I zipped up the gerber files and uploaded them, but I don't know what all options to choose to end up with a usable board. Thanks!!

jaimevisser, 2021-05-19:
Or even getting some prepackaged kits would be awesome. Soldering everything together is not a problem, but having to source all the hardware and pcb's is the hardest part here.

billypeanut, 2021-05-19:
Maybe if there's a big interest, a Kickstarter or something could be set up to do a big batch build/order of certain parts and reduce costs.

jaimevisser, 2021-05-19:
I think most people have multiple plants to monitor so even with a few people you'd get a considerable amount of parts that can be bought in batch.

rbaron, 2021-05-19:
I'm super happy to see everybody's interest here and thanks for the kind words! I am thinking about putting together a wiki in this repo with:
* Steps and screenshots on how to order it from JLCPCB
* How to solder the nRF module & battery clip
* How to protect the board & components from moisture

I'll ping this thread once it's up.

I agree with @jaimevisser. Sourcing the parts is not trivial - specially during this crazy chip shortage. In the 3 batches I've ordered so far, I had to source the nRF52 module from aliexpress at a higher price, as JLCPCB ran out. The battery clip also has to be sourced separately, since it's mounted on the underside of the board.

Dza-byte, 2021-05-21:
I too am very interested in this! Commenting so I can get the ping when the wiki goes up.  No experience with putting boards together myself

MrEinstin, 2021-05-23:
I'm wanting about a dozen or so of these myself. I was also looking at the files and saw the gerber files but I didn't see the schematic? Perhaps I'm overlooking it somewhere. I'd also like to see a USB port for easier programming.

rbaron, 2021-05-24:
Hey everyone,

I took a first stab at documenting the ordering process in the [How to order - PCB fabrication and SMT assembly](https://github.com/rbaron/b-parasite/wiki/How-to-order---PCB-fabrication-and-SMT-assembly) wiki page. Let me know what you think! Please feel free to contribute too.

It mostly covers my experience with ordering from JLCPCB, but I'd love to hear other peoples' experiences with other fab houses.

rbaron, 2021-05-24:
Hi @MrEinstin,

You can find the schematic in [kicad/parasite.pdf](https://github.com/rbaron/b-parasite/blob/main/kicad/parasite.pdf).

As for adding a USB connector, I think it's a great idea and something that also crossed my mind, but I ended up not going for it. On one hand, the nRF52840 chip we use does offer a USB stack which we could use. For using that for programming, I believe we'd need a bootloader like [Adafruit_nRF52_Bootloader](https://github.com/adafruit/Adafruit_nRF52_Bootloader). You'd still need a SWD programmer for burning it in first. It also adds a few hundred milliseconds to each run, chopping the battery life in ~half and/or requires a switch to toggle between boot modes, which adds a moving part to the board. I really wanted to keep things simple and as cheap as possible that we could deploy and leave it running for a long time, so I decided against it.

For development, I've been using one these [pogo alligator clamps](https://www.aliexpress.com/item/4001188042214.html?spm=a2g0o.productlist.0.0.55347aff5RofQ5) and I find it's super convenient to quickly hook it up and program/debug while testing.

drewzarn, 2021-05-24:
@rbaron This is great, thank you! I just priced out 15 boards and it's just under $32 with assembly. JLCPCB is short of the SHTC3 as well at the moment.

rbaron, 2021-05-24:
Awesome @drewzarn. Please keep us posted!

drewzarn, 2021-05-28:
@xlfe Were you able to source the chips that JLCPCB was missing?

Senbei123, 2021-05-29:
Nice project!!!

xlfe, 2021-05-29:
Only on back order for the temp sensor. AliExpress had heaps of the E73s

bit-bat, 2021-05-30:
@rbaron Thank you for putting the wiki together! This will help me save 10s of hours. 
I have a request though - could you add the nRF chip as well in the BOM & placement files for JLCPCB? 

This can save us some pesky SMT soldering when the chip is back in stock (and until then it doesn't hurt). I am planning to order around 30 of these (multiple pots and garden) and soldering that many SMT chips by hand is not fun.

rbaron, 2021-05-31:
Thanks @bit-bat. I will update the nRF chip. I can't wait for this chip shortage to be over 😰

rbaron, 2021-05-31:
Hi @bit-bat,

I updated the BOM and pick-and-place files with the E73 module in 753e5d832587b0f753ab5e64ed213f535deabbdf.

Since we can't order the SMT assembly with this part yet, we need to pay extra attention to the order when it does become available. This part requires special attention for SMT assembly because the footprint is not super standard. In practice, it's very likely JLCPCB engineers will take a look and fix the position and orientation of this component if there's an obvious error, or they might refuse the order if the footprint doesn't match. In that case, we'll probably need to find a way to use their specific footprint for it (sort of available [here](https://lcsc.com/product-detail/Bluetooth-Modules_Chengdu-Ebyte-Elec-Tech-E73-2G4M08S1C_C356849.html) for EasyEDA - not sure if it's possible to download it for KiCad).

I will keep an eye on it and put an order for a trial batch once the E73 module is available. I'll report back here. If you're the first one putting an order with the E73 module, just pay extra attention and things should be good.

argupta23, 2021-06-08:
@rbaron 
Thanks for your efforts around pulling this all together and sharing it with the community.
I am interested in ordering some as well.
I do have a couple questions
1. Does JLCPCB provide testing of the assembled product? Assuming they would need some of your firmware to do that
2. Can the firmware you have listed here be shared with JLCPCB?
Please advise.

@bit-bat 
While I understand the issue with chip shortages, can you please share some details around the total cost of each unit and what is the sweet spot in terms of quantity to make it cost effective?

rbaron, 2021-06-10:
Hi @argupta23,

I don't think JLCPCB offers such a service, but I may be wrong. I imagine big fabrication houses might do that for really big orders, but I don't have experience with that. For the orders I placed there, I soldered, programmed and tested the boards myself.

argupta23, 2021-06-11:
@rbaron 
Appreciate the feedback. given the shortages on SHTC3 and the NRF modules, I will anyway end up in the same situation.
With regards to other discussions ongoing in different threads on possible alternatives, are you planning to make any changes to the current version of the PCB? If yes, can you share when will it be available and what would the updated parts list look like?

Please advise.

rbaron, 2021-06-12:
@argupta23, I think a good plan is to update the slightly older and cheaper SHTC3 with some other [Sensirion SHT*](https://jlcpcb.com/parts/componentSearch?isSearch=true&searchTxt=sensirion). I will update the design, do a trial batch and update the code accordingly.

For the nRF module, I am not so sure yet it's worth changing the design. Even though JLCPCB doesn't have it in stock, I've been getting them from aliexpress and I still see them listed for ~5-6 USD, which is okay given the circumstances. [This wiki from nrfmicro](https://github.com/joric/nrfmicro/wiki/Alternatives#modules) lists some alternative modules, but I think they're all in a similar situation. I love the nRF52840, but it's a lot beefier than we need. Another possibility is to dial back and use something less powerful and cheaper, but that would incur considerable design and code change, and it's also not clear what the short term availability of that would be. We'd also lose the possibility of exploring ZigBee and on-chip USB for saving ~2 USD per board. At hobbyist's volumes, I don't think that pays off.

So, all that is to say that currently, on my mind, I think it's worth updating the SHTC3 and sourcing the nRF module separately.

argupta23, 2021-06-14:
@rbaron 
thanks for the feedback.  I will go down the path of ordering the parts separately and put some builds together when I get them.

bit-bat, 2021-06-15:
> @bit-bat
> While I understand the issue with chip shortages, based on the order you placed. can you please share some details around the total cost of each unit (complete with the pending parts cost)?

Hi @argupta23. I did not place the order as I am waiting for the chips to be available. Soldering 50 SMTs with ~30 pads each is a lot of work that I am not ready to commit (yet). But I did walk through the steps and at the end the PCB (with most components) was under $1 / unit when placing order for 50 units. This price doesn't include 2 chips - SHTC3 and nRF

argupta23, 2021-06-26:
@rbaron 
Is there a way I can reachout to you directly?
Thanks

Tirocinante01, 2021-06-28:
hello I'm sorry to bother
I read the discussion and saw all the links and blueprints but I was wondering
is there any way I could get a few b-parasite already built and soldered?

i sadly have no experience on doing this stuff and I dunno any closely company that could print/build these out for me
im not looking for big numbers but just 3-4 for a project of mine

semperstew, 2021-07-13:
Please forgive my ignorance if this has been discussed elsewhere. I'm trying to follow your instructions in the wiki for ordering the PCBs from JLCPCB. After uploading the zipped gerber files, I am unable to add the order to cart because the specification is missing "dimensions". I had thought those would be somewhere in the gerber files. Do you have a default set of dimensions that you would recommend?

[image]

rbaron, 2021-07-16:
Hi @semperstew,

I believe this field has been automatically filled every time I put an order with them, just after the zip file has been uploaded. 

Would you try again and let us know if that worked?

semperstew, 2021-07-19:
@rbaron I created a fresh clone and tried again. Seemed to work much better this time. A lot more of the data was filled in on the JLCPCB order form. Sorry about the false alarm.

rbaron, 2021-07-20:
No worries @semperstew. Glad to hear you got it working!

derwana, 2021-07-31:
Regarding the SHTC3 shortage: i ordered myself some cheep sensormodules from [AliExpress](https://www.aliexpress.com/item/4000352981649.html) and desoldered the SHTC3. After preheating the little daughter board with a thick soldering tip the chips came of with no problem.

[image]

argupta23, 2021-08-03:
Can somebody please share instructions on how to program these boards?
I do have a Jlink mini on order. 

Appreciate the help 

thanks

andreapx, 2021-08-07:
> 
> 
> is there any way I could get a few b-parasite already built and soldered?
> 

I'm interested to to buy a fully built version of the b-parasite and maybe even of the w-parasite (I want to install some of these in my garden and a single BLE bridge will not work).
Thanks!

rbaron, 2021-08-08:
Hi everyone,

The current semiconductor shortage is making things a little difficult. I hope I can put together some boards soon.

Meanwhile, I added a new wiki page on [How to Program](https://github.com/rbaron/b-parasite/wiki/How-to-Program) the boards.

argupta23, 2021-08-08:
Should somebody put in an effort to pull together a small batch, I am curious to hear what price range do people think the assembled version of the b-parasite module should be in? 

@Tirocinante01 @andreapx 
and what quantities are we talking about? 

Thanks

andreapx, 2021-08-09:
>
> @Tirocinante01 @andreapx
> and what quantities are we talking about?

It actually depends from the price if it's simila to what @drewzarn said (about 2$/piece, but actually even 4$/piece), I'd buy 5/6 pieces, If it's more in the order of 10$/piece, I'd buy 1 or 2 pieces.

hubikj, 2021-08-10:
I would buy 10 pieces if the price is reasonable.

rwmccoy, 2021-08-10:
I would buy 10 pieces at any reasonable price.

argupta23, 2021-08-11:
> > @Tirocinante01 @andreapx
> > and what quantities are we talking about?
> 
> It actually depends from the price if it's simila to what @drewzarn said (about 2$/piece, but actually even 4$/piece), I'd buy 5/6 pieces, If it's more in the order of 10$/piece, I'd buy 1 or 2 pieces.

@andreapx 
thanks for the feedback on price expectancy.

Given the chip shortage and current state of things, the raw material cost itself is around $13-15. Put in the additional efforts to assemble, flash, calibrate which is about 30-60min. 

Having said that I don't think that anybody can match the prices you are looking at. 

A sweet spot to make it commercially viable would be in the range of $23-25 + shipping. There could be some discount on purchase of 5+ units. 
 
what @rbaron  has IMO is a good start and can be extended down the line.

@hubikj @rwmccoy 
Does the price meet your expectancy? and would you still be looking at 10 pcs?

Thoughts and feedback are welcome.

hubikj, 2021-08-11:
I think it is OK price for a fully assembled, flashed and calibrated units. I think I would still buy 10, definitely 5+.

rbaron, 2021-08-11:
For context, @argupta23 reached out to me via email and I sent over these estimations, which he relayed here. We are by no means associated nor do I have any interest in pursuing any commercial partnership at this point.

I can provide some more insight into the economics of my last few orders and explain where these figures come from.

Let's take this batch of **30 boards** b-parasites as an example.

# JLCPCB costs
[image]
* PCB manufacturing: $9.80
* SMT parts & assembly: $57.11
* Shipping: $21.80
* **Total: $88.71** (w/o discount)

# nRF52840 module
This module would cost around $8 if sourced at JLCPCB (+ smt setup fee), but they are unavailable. On AliExpress, you'd normally find them for around $6.
For 30 boards, that would be **$180**

# Battery clip
JLCPCB also does not assemble the battery clip on the bottom of the board, so we have to source it separately. They are cheep, but the shipping is significant.

# Tax
I expect to be charged around $25 for orders this size.

# Total cost
For 30 boards: $180 + $88.71 + $25 = $293.71
Per board: $293.71 / 30 = $9.80 ~ $10 / board

# Manual work
* Hand soldering the nRF52 module
* Testing everything (out on water, out in the air)
* Hand soldering the battery clip
* Applying the conformal coating
* Packaging

We can easily pipeline this project and roughly estimate 10min/board (less if everything works on the first try; more if debugging is involved). You can plug in how one hour of work is priced and compute accordingly. For the sake of the argument, let's call it $5/board.

# The indie hacker multiplier
To make these indie hardware projects remotely economically viable, considering hardware issues, delays, returns etc, it's usual to multiply the cost by 2.0 - 2.5x ([source](https://youtu.be/UwrkfHadeQQ)).

That would bring the end user price to around ($10 + $5) * 2.something = **$30-$35**

# Final words
All of these values can be significantly brought down by optimizing the design, setting up huge (~10k+) orders, cutting features, picking a less capable RF module. These are all interesting options, but currently what brings me joy is keeping this a hacker friendly, completely open project. I would love for people to have a few of these in their houses to tinker with, hack it, extend it, learn and have fun, instead of optimizing for cost.

rwmccoy, 2021-08-11:
I might dial back to 6 as the price goes up just to get to the discount point. I see the issues with setting the price, especially with manual assembly and calibration required. Thanks for the insight into the process.

RAYs3T, 2021-08-16:
I'm also interested in buying about 5 Units (complete or partially assembled).
So if anyone is going to order a few or even assemble them I would be happy to take part in the order (living in Germany).

Purii, 2021-08-24:
Same for me. Also living in Germany and would be highly interested in buying around 3-5 units

andreapx, 2021-10-09:
Ok, I'm up for 2/3 pieces!

rbaron, 2021-10-18:
@icanhazpython, I just pushed a commit removing the C_0805_2012METRIC resistor from the BOM file. It shouldn't have been there. You can safely ignore it.

janpfo, 2022-01-08:
@rbaron I see that you mention "Applying the conformal coating", what conformal coating do you recommend to be apply?

Also, JLCPCB didn't have the red LED. It will not impact the final product to skip this part?

rbaron, 2022-01-09:
@janpfo I've been using this [Plastik 70 frmo Kontact Chemie](https://www.farnell.com/datasheets/2723407.pdf).

The missing LED should be okay, but you can essentially select any other red 0402 LED from JLCPCB.

rbaron, 2022-01-11:
@janpfo In the render, we can see Q4 connected to the SoC pad and to GND through the resistor R10. So the Anode should be on the left (side connected to the SoC pad) and the cathode on the right (connected to GND through R10). I believe this is configuration in the (corrected) right render, assuming the red vertical bar represents the cathode.

janpfo, 2022-01-11:
> @janpfo In the render, we can see Q4 connected to the SoC pad and to GND through the resistor R10. So the anode should be on the left (side connected to the SoC pad) and the cathode on the right [...]

I looked at the schematics, and it looks like the original is correct. If the numbers on the schematics and render is correct. The "1"-side should be connected to pin 8. 

[image]
[image]

rbaron, 2022-01-11:
@janpfo I would try clarifying with JLCPCB which pin represents the collector (pin 1 for us) and the emitter (pin 2 for us) in their footprint. Hopefully both us and them assign the same pin numbers to the footprint, but I'd still double check. Please let me know what they reply.

janpfo, 2022-01-11:
> @janpfo I would try clarifying with JLCPCB which pin represents the collector (pin 1 for us) and the emitter (pin 2 for us) in their footprint. Hopefully both us and them assign the same pin numbers to the footprint, but I'd still double check. Please let me know what they reply.
> 
> Looking at their [product page for the phototransistor](https://www.lcsc.com/product-detail/Ambient-Light-Sensors_Everlight-Elec-ALS-PT19-315C-L177-TR8_C146233.html), it seems like the vertical bar represents pin 1, and not the electrical characteristic (like in a diode symbol):
> 
> [image]
> 
> In this case I believe the **left** render is indeed correct (pin 1 on the left).

Looks like the original render was correct. Collector on pin 1 and emitter on pin 2.

a83r, 2022-03-29:
Hi, 
help would be appreciated. 
JLCPCB "corrected" the layout. I see they changed part Q4 and U2 is this correct? As I understood that Q4 has the correct position what is with U2?

First is the original

[image]

Second is corrected by JLCPCB

[image]

a83r, 2022-03-29:
> Hi, help would be appreciated. JLCPCB "corrected" the layout. I see they changed part Q4 and U2 is this correct? As I understood that Q4 has the correct position what is with U2?
> 
> First is the original
> 
> [image]
> 
> Second is corrected by JLCPCB
> 
> [image]

Ok, I checked myself with PCBEditor and found that U2 is in the second picture correct but Q4 in the original.

rbaron, 2022-03-29:
@a83r,

can I ask you to please crop/resize those images? Thanks.

On the comment above yours, @janpfo and I discussed this same issue with Q4. The conclusion was that Q4 was originally correct. @janpfo did you manage to order the boards? Was the orientation of Q4 correct? Thanks!

U2 seems to be correct in the corrected version - the dot should align with the footprint.

a83r, 2022-03-29:
@rbaron It looks like that JLCPCB spins U2 in the original picture (maybe a bug?) but shows it correctly in the corrected version. But in the DMF overview, they don't state that they changed Q4 only U2. 

I cropped the pictures, sorry I was little panicked as I didn't expect them to ask such questions....

This is how my DMF Overview looked like:
[image]

 I told them that Q4 was rightly placed in the original upload as well as U2

rbaron, 2022-03-29:
No worriees @a83r. It's usual that they analyze and suggest these adjustments. Just to be clear, U2 seems correct in the _corrected version_. The difference is in the red dot. In the second (correct version), it correctly aligns with the footprint dot. The convention is that the dot identifies the pin 1 of the package, so they should match.

janpfo, 2022-04-10:
> 

Sorry for my

> @a83r,
>
> can I ask you to please crop/resize those images? Thanks.
>
> On the comment above yours, @janpfo and I discussed this same issue with Q4. The conclusion was that Q4 was originally correct. [...]

> @a83r,
>
> can I ask you to please crop/resize those images? Thanks.
>
> On the comment above yours, @janpfo and I discussed this same issue with Q4. The conclusion was that Q4 was originally correct. [...]

Sorry for my late reply. Can confirm that 4/10 boards that I ordered works. 

If you are soldering the nrf chip, don't do the same mistake as I did. Use flux when soldering and apply solder to the pad before soldering the chip. Learned that the hard way 😅

drspangle, 2022-04-28:
I just placed a batch order for 15 of these. I had to order the BTLE chips from AliExpress, but I managed to get a significant bulk discount. Hoping I'll be able to get at least 10/15 brought up. I'm also working on ordering the cases from Shapeways, they're very expensive as a one-off order so I'm hoping I can get a bulk discount. Anyone know where I can get the cases 3d printed (FDM or SLA are both fine) for cheap?

RAYs3T, 2022-04-28:
@drspangle Any chance you want to sell 2-3 of these? :) - I could also offer to print you some cases in black or white PLA with my Voxel 3D printer.

a83r, 2022-04-28:
@drspangle 

Hey, I ordered 15 cases at [craftcloud ](https://craftcloud3d.com/) I paid for all of them with PETG (translucent) 46€. Mabye you give it a try aswell?
For future reference, you can buy the chips from Ebyte directly on [alibaba](https://www.alibaba.com/product-detail/E73-2G4M08S1C-Small-Size-Integrated-ARM_62096319088.html?spm=a2700.shop_plser.41413.12.8c1676b9BNaVZd) I paid for 15 pieces 96$ + Shipping + VAT

Get yourself a good amount of solder flux as well, it helps a lot to solder the chips to the board.

[26 more replies]