rbaron/b-parasite #5: Alternate Parts Availability

MrEinstin, 2021-05-25:
Right now as most have seen the nRF52840 ([E73-2G4M08S1C](https://www.ebyte.com/en/product-view-news.aspx?id=445)) as proposed is currently unavailable for the most part and the ones that are available have obviously went up in price. From what I can determine this particular version of the part, the advantage is the fact that it is pre-mounted on a daughter board which makes hand soldering easier and has a ceramic antenna already attached.

The Nordic nRF52840 SoC itself is available at LCSC ([C190794](https://lcsc.com/product-detail/RF-Transceiver-ICs_Nordic-Semicon-NRF52840-QIAA-R_C190794.html)) for pick and place manufacturing and at a better cost with 14,000+ currently available with more reportedly coming in. Using the Nordic in this fashion also gives you access to more of the IO's that are available.

From what research I can find, the ceramic antenna is what is considered a medium gain antenna. Better range and performance could be achieved by designing in an IFA pcb copper trace antenna that would also occupy near the same footprint of the ceramic antenna.

rbaron, 2021-05-25:
Hi @MrEinstin,

I would absolute _love_ to try and use the standalone nRF52840. It's something I've never done but I'd like to learn how to do it properly. I do have some reservations and some questions I need to learn how to answer:

* RF is tricky business. We have to pay attention to impedance matching, ground planes, steer clear from high frequency signals. I see people doing it in prototypes (specially for esp32s), and videos like [this great one from Phil's lab](https://www.youtube.com/watch?v=14_jh3nLSsU) make me think it's possible, but still looks very involved, and I believe we'd have to move to a 4 layer board, adding to cost;
* We do have reference designs in the nRF52840 datasheet that we can simply copy, but if the RF performance is bad, I think debugging is outside of the possibilities for me and most hobbyists (I'd love to be proved wrong here);
* It adds around a dozen new components to the BOM (resistors, capacitors, high frequency oscillator, inductors). Even though these components are relatively cheap, this also adds to the setup fee cost of SMT assembly;
* Fabrication capabilities. There are two packages of nRF52840:
  * WLCSP I believe requires laser cut microvias, which afaik is not offered by JLCPCB and I believe is prohibitively expensive in other fab houses
  * QFN might be doable, but I'd need to look closer at clearances and traces widths
*  Certification. With a module like the E73, I believe we get FCC (and friends) certification for free, which is nice. Getting a new RF design approved I believe is also outside of the realm of possibilities for hobbyists. I imagine this is the reason for even big names like Adafruit use ready modules in their Feather lines;

Even if we can pull this off, I have a feeling I would come out as more expensive for the volumes we usually order, and would only start offsetting the added fixed costs in the dozens of thousands boards. For reference, the standalone nRF52840 LCSC part you linked sells by $8.58, while a E73 module goes for ~$6 on aliexpress. It's hard to compete.

I am very interested in learning what's possible here and I'd love to hear more peoples' opinions on these points. Thanks for getting the conversation started.

lukas-kurka, 2021-08-10:
Hi @rbaron, I’m also strugling to source [E73-2G4M08S1C](https://www.ebyte.com/en/product-view-news.aspx?id=445). Have you tried [E73-2G4M08S1E](https://www.ebyte.com/en/product-view-news.html?id=1035)? Those seems to be widely available these days. Beside some little differencies in consumption the only significant difference I can spot is halfed flash and ram size. To what size does your code compile to?

lukas-kurka, 2021-08-12:
Well I’m hobbyist and much more into to SW than HW (this would be first time for me ordering custom pcb). I know the chip is different, however when I compare the datasheets side by side on [ebyte](www.ebyte.com) website I would say those modules are exactly the same (not counting the chip) including the pinout.

jhbruhn, 2021-09-02:
I have tried this successfully: #13 adds support for it. The Code has remained unchanged, only some defines and the linker script had to be adjusted for the different MCU.

rbaron, 2021-09-02:
I just merged #13. Fantastic work @jhbruhn! 🎉

rbaron, 2021-10-16:
Thanks everyone for the ideas and work here - super happy that we now have support for an extra nRF module.