rbaron/b-parasite #7: Exchange SHTC3 with more abundant SHT3X/SHT20

SchmidL, 2021-05-30:
Hi, 

found your project yesterday and love it!
Today was checking the feasability of the current version and wanted to ask why you went for the SHTC3 instead of the rather more often used SHT3X series, especially the SHT31.
The only reason I can find is the bit easier way to solder the sensor due to the packaging of DFN-4 instead of DFN-8 and the pricing - Is that right?
Another option would the the SHT20

The downside is right now that the sensor is really hard to source as on https://lcsc.com/product-detail/Temperature-Humidity-Sensors_Sensirion-SHTC3_C194656.html

rbaron, 2021-05-31:
Thanks @SchmidL!

It's unfortunate that the SHTC3 is out of stock. I originally chose it because it was cheaper and the number of parts in stock has super high. The fact that's a DFN-4 or DFN-8 doesn't matter too much I think, since we're using SMT assembly. Both these are very tricky to solder by hand, unless you use solder paste/toaster oven.

I will take a look at the other options. Most likely we'll have to rewrite some of the code that talks to it, but it should be a small change.

SchmidL, 2021-05-31:
Hi @rbaron, 

Thanks for the fast reply! Yeah thats also what I thought with the soldering. But was curious if I saw some super power saving specs or something!
I mean if the supply gets restocked, then I would be totally in for the cheaper option!

closed 2021-05-31 by SchmidL