rbaron/b-parasite #9: How to program

volzb, 2021-06-27:
HI, 

Nice work!!

I'm looking for guidance on how to program the devices, I have an Atmel ICE that has SWD would that work? 

Are there any cheap programmers that could easily be used?

kuerb, 2021-06-28:
I did manage to upload code using one of these bluepill programmers (~2€), as described in the [nrfmicro Wiki](https://github.com/joric/nrfmicro/wiki/Bootloader). They have no connection to the reset pin and I ended up buying a J-Link Edu Mini (~20€) because I couldn't get them out of an infinite loop (that I accidentally put into the code myself) otherwise. (Be aware, that the J-Link Edu Mini has a 2-row 1.27mm pitch header with no VCC pin, which I thought was a bit annoying to adapt.)

argupta23, 2021-06-28:
@kuerb @volzb 
I have a quick question. 
From where did you source the SHTC3 and the E73... modules? My experience with AliExpress sellers has not been pleasant.

kuerb, 2021-06-28:
@argupta23 I don't think this issue is the right place for this discussion. but so far, i've ordered the modules from 4 different aliexpress sellers without any issues. the shtc3 was in stock at lscs when i ordered the pcbs.

argupta23, 2021-06-28:
@kuerb 
If you feel comfortable, can you please share some links of sellers you have worked with? 
I know lcsc is out of stock on both, else I would I have gotten it from them as part of my order.
Appreciate the help.

rbaron, 2021-06-28:
Thanks @volzb. And thanks @kuerb for chiming in.

I second @kuerb. I think the J-Link Edu Mini is a good choice for hobbyists.

AlexSzlavik, 2021-07-02:
Adafruit offers the J-Link mini for a reasonable price. Sparkfun also has them.

rbaron, 2021-07-03:
Another new and promising alternative is @atc1441's [ESP32-based SWD flasher](https://github.com/atc1441/ESP32_nRF52_SWD). I haven't tried it myself, but it's looks super sweet. I'm looking forward to giving it a go.

argupta23, 2021-08-07:
@kuerb @rbaron 
Can I please request some help on how one should,
1. wire-up the J-link to the b-parasite board?
2. steps to program the board?

Thanks in advance for your help.

rbaron, 2021-08-08:
Hi @argupta23,

I set up an initial [How to Program](https://github.com/rbaron/b-parasite/wiki/How-to-Program) wiki page.

I hope it helps.

argupta23, 2021-08-08:
@rbaron 
Thanks, I will give it a shot and let you know if I run into issues.

AlexSzlavik, 2021-08-09:
> Hi @argupta23,
> 
> I set up an initial [How to Program](https://github.com/rbaron/b-parasite/wiki/How-to-Program) wiki page.
> 
> I hope it helps.

@rbaron Fantastic job on that wiki article! Heavy Kudos

william-1066, 2021-09-03:
@rbaron   - awesome project.  I did get 30 made (my first every PCB order - I had a lot of faith in you :-) plus your documentation made it very easy) and have started manually soldering the sensirion and ebyte chip on to boards - yes ... I am desperate to get this out into my greenhouse!

With no prior experience of this tool set or JTAG "thingy" I finally managed to get my first board working - awesome feeling.

[21:13:20][D][sensor:131]: 'b-parasite 01 Battery Voltage': Sending state 3.30800 V with 3 decimals of accuracy
[21:13:20][D][sensor:131]: 'b-parasite 01 Air Temperature': Sending state 22.18700 °C with 1 decimals of accuracy
[21:13:20][D][sensor:131]: 'b-parasite 01 Air Humidity': Sending state 60.05402 % with 1 decimals of accuracy
[21:13:20][D][sensor:131]: 'b-parasite 01 Soil Moisture': Sending state 25.61951 % with 1 decimals of accuracy

I could not flash via  make  (even after realising that "make flash-softdevice" should be "make flash_softdevice".  I kept getting "JLinkARM DLL reported an error" but was able to flash manually using JLinkExe loadfile, first the softdevice then the firmware.

I am not sure if I need to switch the flash mode to SWD as I am using a Segger J-Link and maybe the default is JTAG mode.  I could not find a way around this.

Anyway - thanks a million for your efforts here - 29 boards to go !!!

rbaron, 2021-09-04:
Hey @william-1066,

Thanks for the kind words. It makes me really happy to see people learning and hacking with b-parasite.
Are you soldering the temperature sensor by hand? That's pretty impressive!

As for the `make` error, could you please paste the whole output? It's interesting that it works with `JLinkExe` but not `make`.
Some thoughts:

1. Have you seen the [Deep Sleep & The Reset Trick](https://github.com/rbaron/b-parasite/wiki/How-to-Program#deep-sleep--the-reset-trick) on the wiki? Did you try `make flash_loop` and briefly shorting the reset pin to ground?

2. Under the hood the Makefile does:
```
flash: default
	@echo Flashing: $(OUTPUT_DIRECTORY)/$(TARGETS).hex
	nrfjprog -f nrf52 --program $(OUTPUT_DIRECTORY)/$(TARGETS).hex --sectorerase
	nrfjprog -f nrf52 --reset
```

The [nrfjprog](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fug_nrf_cltools%2FUG%2Fcltools%2Fnrf_command_line_tools_lpage.html) is a program from Nordic that interfaces with J-Link programmers. In theory it should set the correct protocol params (SWD, speed).

william-1066, 2021-09-04:
Hi @rbaron 

I am soldering the temp sensor and processor by hand - they were out of stock when I ordered.  The temp sensor is a bit tricky, but nothing watching a couple of you tube videos could not fix :-)

Thanks for the pointers.  With the third one I build I will try to create some logs and post here after a bit more of my own troubleshooting.

I have one other question - I seem to be getting the same MAC address on both sensors.  The top one is the new sensor I added.  Have I missed something in the documentation?

```
[18:48:34][D][sensor:131]: 'b-parasite 01 Battery Voltage': Sending state 3.16000 V with 3 decimals of accuracy
[18:48:34][D][sensor:131]: 'b-parasite 01 Air Temperature': Sending state 20.88900 °C with 1 decimals of accuracy
[18:48:34][D][sensor:131]: 'b-parasite 01 Air Humidity': Sending state 59.10797 % with 1 decimals of accuracy
[18:48:34][D][sensor:131]: 'b-parasite 01 Soil Moisture': Sending state 7.31964 % with 1 decimals of accuracy
[18:48:34][D][esp32_ble_tracker:620]: Found device F0:CA:F0:CA:00:01 RSSI=-57
[18:48:34][D][esp32_ble_tracker:641]:   Address Type: RANDOM
[18:48:34][D][esp32_ble_tracker:643]:   Name: 'prst'
[18:50:05][D][sensor:131]: 'b-parasite 01 Battery Voltage': Sending state 3.07900 V with 3 decimals of accuracy
[18:50:05][D][sensor:131]: 'b-parasite 01 Air Temperature': Sending state 21.46000 °C with 1 decimals of accuracy
[18:50:05][D][sensor:131]: 'b-parasite 01 Air Humidity': Sending state 93.80951 % with 1 decimals of accuracy
[18:50:05][D][sensor:131]: 'b-parasite 01 Soil Moisture': Sending state 72.26715 % with 1 decimals of accuracy
```

esp config.

```
sensor:
  - platform: b_parasite
    mac_address: F0:CA:F0:CA:00:01 
    humidity:
      name: 'b-parasite 01 Air Humidity'
    temperature:
      name: 'b-parasite 01 Air Temperature'
    moisture:
      name: 'b-parasite 01 Soil Moisture'
    battery_voltage:
      name: 'b-parasite 01 Battery Voltage'
```
Thanks 

William

rbaron, 2021-09-05:
Hey @william-1066,

You can change the MAC address before compiling the code by editing [PRST_BLE_MAC_ADDR](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/config/prst_config.h#L26).

william-1066, 2021-09-05:
Hi @rbaron 

Thank you.  So managed to change the MAC and the name - all looking good so far.  

Still can't get the make file to flash.  It may be an Ubuntu thing.

I thought it was going to be tricky to catch the CPU waking up without JLink dll working and using the make option with a flash loop, but I was able to use JLinkExe to halt the CPU and re-flash with a new MAC.  

Again, awesome piece of work.  I have some suggestions which I will add to another issue.

William

rbaron, 2021-10-16:
Thanks everyone. I think most points are covered in the [wiki](https://github.com/rbaron/b-parasite/wiki/How-to-Program). Please feel free to open a new issue in case you have problems.

closed 2021-10-16 by rbaron

SchmidL, 2021-11-15:
> Another new and promising alternative is @atc1441's [ESP32-based SWD flasher](https://github.com/atc1441/ESP32_nRF52_SWD). I haven't tried it myself, but it looks super sweet. I'm looking forward to giving it a go.

I tried this SWD flasher and it works so far like a breez. I could unlock the nRF chip and working now on the flashing of the firmware. Any idea of how I can make flashing the softdevice and actual firmware easier? - until now i just `cat` them together and flashed them all at once. Unfortunately could not check the MQTT messages, as my other setup is also evolving... Or would it make sense to flash the two parts separate with an offset for the b-parasite offset? No idea if there is any issue when using cat together firmwares. Happy for any input :D

rbaron, 2021-11-16:
@SchmidL, I believe the usual flow is to flash them separately. Also notice that you only need to flash the SoftDevice once. Then you can flash the firmware as many times as you want.

atc1441, 2021-11-16:
The problem is that the ESP32 SWD Flasher currently is not able to flash at an offset. 

One file needs to be created first from SD and firmware.

will add an offset feature to the Flasher but i can not say when, it has so much more thinks that could be added