rbaron/b-parasite #10: Offset correction

kuerb, 2021-07-14:
Some of my sensor are reporting an moisture offset of around 2 – 5 %. Is it possible to correct that value? These values seem to be a lot less stable than the 0 % readings I get from the others, when they aren't in soil. Is there already some kind of compensation?

rbaron, 2021-07-14:
Hi @kuerb,

In theory, it's possible to calibrate each b-parasite individually. The [get_soil_moisture_percent](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/prst/adc.c#L95) function (and associated comment) describes how the sensor is calibrated and how you can redo it yourself. 

In practice, variations in the fabrication, soil type, air humidity, temperature, presence/type of protective coating will have variable small effects in the precision of the readings. For this types of capacitive soil sensors, I believe chasing after single digit precision is beyond of the point of diminishing returns. I myself use a common calibration for all my boards and I encourage users to do the same - life is much easier that way.

rbaron, 2021-07-27:
@kuerb, I'll close this issue, but please feel free to reopen it.

closed 2021-07-27 by rbaron

kuerb, 2021-07-28:
Yes no worries. Reprogramming each sensor individually is a bit much. I was hoping for a correction within the scanner or HA.

I haven't tried it yet, but my idea was to use a [plant component](https://www.home-assistant.io/integrations/plant/) in HA and add the offset via a value_template into the configuration.yaml. (Maybe this helps someone else when looking for an answer for this.)