rbaron/b-parasite #11: Split sensor and RF

SchmidL, 2021-07-30:
hi, 

I still follow you project very tightly and will soon order my first batch! Unfortunately, I should have order the nRF when there where still more widely available, but at that time the temp sensor was on short. Anyway... 
Will waiting I was thinking if it would be possible to split the sensor (parasitic cap) from the rest of the electronics.
1. Would make the waterproofing of the electronics much easier, as I can have the nRF outside of my pots, and just the pcb for sensing in the pot.
2. I have some metal pots, which work kind of as a faraday cage, very bad for the bt-signal.

When checking out the schematics, I see just one connection path (SENS_OUT) to the nRF. So i was thinking just to elongate this path via a cable. Now I'm a bit wondering if it is really that easy, or do I miss something? Thank you anyway for your amazing project!

rbaron, 2021-08-01:
Hi @SchmidL,

That's an interesting idea. I think there are a few ways to split the sensor and RF part, depending on where you cut the design. For example, if you want to remove just the nRF module:

[image]

1. The `SENS_OUT` is the output, which is sampled by the nRF, like you described;
2. The `PWM` is the input to the sensor, a square wave;
3. There is a "fast discharge" circuit that sort of shorts the `C_sen+` to ground in the PWM low signal. This is done to ensure the `C_sen+` starts at 0 V in the next PWM high phase;

With the current design, I believe you would have to route these 3 signals (plus GND). 

Another idea is to remove just the actual parasitic sensor traces (`C_sen+` and `GND` signals), and keep all the input/output & fast discharge circuitry on the outside, together with the nRF:

[image]

For this, you could even literally cut the current PCB design to try it out:

[image]

As a last note, if you're willing to route some wires from the pot to outside, one idea is to go one step further and use the same nRF to reading multiple sensors. 

I hope this helps!

SchmidL, 2021-08-02:
Hi rbaron,

Thank you very much for your extensive reply and suggestions! Also a very nice idea is to use multiple sensors on one nRF.
I also really like your suggestions for different separation ideas. In my decision phase i don’t really care to have 3 or less wire, although just cutting off the parasitic capacitor was the one I like the most. I will come back when I have implement the different pcb designs.

Your immediate help is really appreciated!!

rbaron, 2021-09-30:
Hey @SchmidL, I'll close this issue since there's no activity. Feel free to reopen it!

closed 2021-09-30 by rbaron