rbaron/b-parasite #19: Community Gallery

rbaron, 2021-10-16:
I would love to build a small and friendly community of users, and it's really heartwarming to see pictures of your b-parasites. If you have some and would like to share, I invite you to post them in this issue.

I'll start. I believe this is a type of red [Alocasia](https://en.wikipedia.org/wiki/Alocasia):

[image]

And a little bonus. This is a fun [Calathea](https://en.wikipedia.org/wiki/Calathea) time-lapse experiment I ran a while ago:

https://user-images.githubusercontent.com/1573409/137588841-b8cd7b42-9407-4758-8eef-e2b859f3cd9b.mp4

jhbruhn, 2022-03-07:
Finally got around to making some pictures of my sensors! All of them are in cases, some made with transparent resin and some with traditional PLA. These are the photoresistor-based sensors, although I plan to make more from the newer design as soon as E73 modules / nrf52 chips are available again (whenever that might be).

[Epipremnum](https://en.wikipedia.org/wiki/Epipremnum_aureum):
[image]

Rare E73-...-CX variant with external antenna; semiconductor-crisis makes you do interesting things:
[image]

[Hypoestes](https://en.wikipedia.org/wiki/Hypoestes_phyllostachya):
[image]

rbaron, 2022-03-09:
That's awesome @jhbruhn. Which transparent filament is that in the first pic? It looks so nice!

jhbruhn, 2022-03-10:
That is Anycubic Eco Resin Clear, but with a layer of spray coating on there to restore the clearness and have it less diffused.

rbaron, 2022-04-28:
Thanks for sharing @JoelWise!

JoelWise, 2022-04-28:
partial conformal coating!

[image]

[image]

drspangle, 2022-04-28:
@JoelWise Looks fantastic. I opted for the black PCB too! Did you coat both sides, and what parts were masked/uncoated? This is acrylic conformal coat, sprayed on, right?

JoelWise, 2022-04-28:
> @JoelWise Looks fantastic. I opted for the black PCB too! Did you coat both sides, and what parts were masked/uncoated? This is acrylic conformal coat, sprayed on, right?

MG Chemicals 419D acrylic brushed on both sides. I just masked right below the battery contact. There is a visible line where the coating ends. TBH I'm not happy with the result. After some time in the soil there was a lot of white bubbling under the coating. I'm not sure if water is seeping in or there is just some separation while remaining sealed. Spraying is real mess that I don't have the space for. If I were to try coating again I would do a dip process using the manufacturer instructions on the coating datasheet. Also beware that the fumes from the coating liquid are 10x worse than nail polish. Do this outside and wear a serious respirator.

JoelWise, 2022-04-28:
some slightly abused citrus seedlings:
[image]

macourteau, 2022-06-16:
First b-parasite monitoring a Dracaena plant that doesn't get enough attention:
[image]

Black silkscreen, spray-coated MG 419E acrylic conformal coating. 🙂

perja12, 2022-08-03:
[image]

Made my first version of the b-parasite with a couple of changes. I replaced the programming header with Tag Connect footprint and added in a reed sensor (not soldered yet) that can be used to f.ex. trigger a reading. I saw that this was discussed in one of the issues. I plan to make a PR for the reed sensor change if it works.Thanks for a nice project!

SystemFuchs, 2022-08-23:
[image]
[image]

Made my first two parasites with an [Ibex antenna](https://www.amazon.de/gp/product/B07DC97PMY/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1). Used the [E73-2G4M08S1CX](https://www.cdebyte.com/products/E73-2G4M08S1CX) instead of the E73-2G4M08S1C. Did no do real range tests but 15m through a glasshouse & massive glass door to the ESP32 is no problem.

gniezen, 2022-09-28:
[image]

I ordered mine assembled from JLCPCB. They did indeed get in touch to confirm the orientation of photodiode Q4. I used a MHP30 mini hot plate to attach the E73-2G4M08S1C (that I bought from eByte's AliExpress store). I then soldered on the MY-2032-05 battery clip by hand. 

I programmed the module using the nrRF52840DK. Turns out it needs have the battery in even if Vcc is connected! Looks like it uses it as a Vref?

[image]
One thing I was wondering about though - why does it appear as if the soil moisture is increasing over time if I didn't water it?

rbaron, 2022-09-28:
Hey @gniezen, thanks for sharing, that looks great.

About the increasing readings, we discussed it in [this comment](https://github.com/rbaron/b-parasite/issues/2#issuecomment-839928191) and [this issue](https://github.com/rbaron/b-parasite/issues/50). The tl;dr is that we think it has to do with water creeping into the board, and applying a layer of coating helps, which I highly recommend -- see [this twitter thread](https://twitter.com/rbaron_/status/1491467468996431873).

gniezen, 2022-09-29:
@rbaron Ah, that makes sense! Now I'm just hoping you find the time to open-source that PCB rotisserie! :wink: 

I forgot to mention that I'm using the new BTHome integration, and it was so nice and easy to set up. Now waiting on some ESP32 boards so that I can set up [Bluetooth Proxies](https://esphome.github.io/bluetooth-proxies/) all around the house. I'm also hoping that they'll fix the Unavailable bug (https://github.com/home-assistant/core/issues/78702) soon, or maybe I should just set my sensors to send data every 5 minutes instead.

rbaron, 2022-09-29:
@gniezen Happy to hear the BTHome integration worked out of the box. Home Assistant folks will be happy to hear that too.

I just pushed the design files for the rotisserie [here](https://github.com/rbaron/b-parasite/tree/main/experimental/rotisserie). Disclaimer - I see this more like a silly experiment than anything serious 😜

oleo65, 2022-10-22:
Finally, after following this project for almost two years now, I got my first batch of `b-parasites`. 😍 In the end I learned alot about PCB manufacturing, soldering, chip shortages and even had to reactivate my limited `C` knowledge. Thank you all for this great project and I will hopefully be contributing back to the community in the future.

In the meantime some impressions of already soldered and coated parasites.

As I have ordered a bigger batch, I still have some spare ones if anyone is interested, either bare metal or fully assembled and flashed. Based in Germany but shipping within EU would probably be fine. 😉 

[image]

[image]

[image]

kzyapkov, 2022-10-22:
> I still have some spare ones if anyone is interested

Someone is :) Since your email is not public, please write to kiril.zyapkov@gmail.com, thanks!

hubikj, 2022-10-22:
> I still have some spare ones if anyone is interested, either bare metal or fully assembled and flashed

Hello @oleo65, I am very interested in the fully assembled spare ones. I am from the Czech Republic and I also have an address in Germany, so it should be easy in terms of shipping. Please contact me at hubik.j@gmail.com. Thank you!

TMillesich, 2022-10-23:
> "I still have some spare ones if anyone is interested"

@oleo65 if there's still a fully assembled and flashed one left i'd be VERY happy to order one!
Greetings from Austria --> mrwho@hallowelt.at

31670253, 2022-10-24:
Hi @oleo65 , I was wondering of you can maybe answer a few questions regarding the manufacturing of the PCB’s if you don’t mind:
1.) Through who did you order your PCB’s.
2.) How long did you had to wait for them after ordering. 
3.) Cost per PCB (+size of batch), can be in your currency will convert. 
4.)Lastly Which components did you had to assemble yourself ? 

Sorry for asking in this thread, just figured you posted the comment 2 days ago so the pricing would be the same still. 

Kind regards 
Niel

oleo65, 2022-10-24:
Hi Niel,

I ordered at JLCPCB, mainly because the POS/BOM files are tailored for their stash. Once you understand all the aspects needed for PCB ordering and assembly, the overall experience was quite good.

The production time was around 5 days if I recall and then another 6 days for delivery. A lot faster than I expected.

One requirement for me was, that I did not want to solder any of the tiny parts on the front side *including* the microcontroller. And this was the hard part during ordering since they are not in stock at any of the manufacturers.

I ended up pre ordering them at JLCPCB and had to prepay for them, once they arrived I could order the PCB fully assembled.

To get down to a reasonable price per unit I ordered a bigger batch of 50 parasites. The microcontroller was by far the biggest chunk of cost with around 8 € per piece. In total I came down to roughly 12.50 € including shipping / tax and the additional battery clips. For smaller batches I suppose you will land more in the range of 18-20 € since you have substantial one time fees per order.

For starters I would recommend configuring everything on the JLCPCB site and you will get a quote for your desired batch size.

31670253, 2022-10-25:
Thank you @oleo65 for your fast reply, I saw that the microcontroller is a bit pricey, One could rather go for a Zigbee chip alternate, which I discovered on Github, https://github.com/smartboxchannel/Plant-Watering-Sensor-Zigbee, because of its range advantage... Anyways, thanks for the information.

kind regards

Pferdebockwurst, 2022-11-01:
> Hi Niel,
>
> I ordered at JLCPCB, mainly because the POS/BOM files are tailored for their stash. Once you understand all the aspects needed for PCB ordering and assembly, the overall experience was [...]

Hey Ole, I am from Germany and I'd really like to buy a couple of sensors. I'd be happy if you would reply soon 🙂

evlo, 2022-11-14:
@oleo65 do you still have some that are already in EU and with firmware on available for shipping to Czechia?

@31670253 I would not really recommend zigbee because of it's pairing issues etc. But obviously, everyone has their own experience, I do have much better results with bluetooth + esp32 BLE-wifi gateways.

oleo65, 2022-11-17:
@evlo Yes I do. Please contact me via E-Mail on my public profile.

jona-fun, 2023-03-26:
[image]
I finally have 10 parasites set up in my garden 5/BThome and 5/zigbee
They are two versions of the high air flow case one in transparent petg and a combination of green and transparent petg.

Judman, 2023-07-11:
First time setting up some parasites!

Old and new airflow case version.

[image]
[image]

ryanrolds, 2023-10-19:
Left to right: hand-soldered with regular chip, front of JLPCB soldered with X chip, back of two-layer placement by JLPCB, with inner case and antenna mount.

Next, I will print an outer case that provides protection from watering. 

[image]

ryanrolds, 2023-11-15:
[image]

Now with a new case.

acamilo, 2024-03-04:
I made a streached version to monitor temperature and moisture in my hay bale garden.
[image]

drspangle, 2024-04-03:
@ryanrolds Nice outer case design! Might I suggest a PR for including it in the repo as an alternative case? We have a few different case designs at present in there and this seems like a potentially useful alternative.

Pferdebockwurst, 2024-04-26:
> ![20240403_133607](https://private-user-images.githubusercontent.com/343979/319352737-15458522-a349-4ae2-a386-9ce272628ed7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxNjA4MzgsIm5iZiI6MTcxNDE2MDUzOCwicGF0aCI6Ii8zNDM5NzkvMzE5MzUyNzM3LTE1NDU4NTIyLWEzNDktNGFlMi1hMzg2LTljZTI3MjYyOGVkNy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNlQxOTQyMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05YTdhNmJmYmM0Y2UwZWRmNTU1N2FhYjE0N2Y1ZmUwMWQ4N2I2NWRkN2Q0Mzk0MDlhZDJmNWFmNTg2MGYzYzNlJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.jLOl2dH4nb_nJqovc4tESVxqxW7dRsGG1VkUPnbeKSQ) ![20240403_135708](https://private-user-images.githubusercontent.com/343979/319352934-f2d91d2e-f179-4706-a8b2-7fc1cd36f2c4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQxNjA4MzgsIm5iZiI6MTcxNDE2MDUzOCwicGF0aCI6Ii8zNDM5NzkvMzE5MzUyOTM0LWYyZDkxZDJlLWYxNzktNDcwNi1hOGIyLTdmYzFjZDM2ZjJjNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNDI2JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyNlQxOTQyMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wNTNmMzRmNDQ0ZThiMGVmYzI4YmU4Mjc0MmNhOTIyODVjNmMyOThmZmZkZGM2NjRhYmI5ZWM2OTk4MTNjYWVkJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.enXPwdj9s__5jC__x29s8cNOxk1qxZ-OSHQJtcYE87s)

Hey @ryanrolds, I really like the design of your case and would be very happy if you could share the model files with us.

Pferdebockwurst, 2024-06-01:
> ![20240403_133607](https://private-user-images.githubusercontent.com/343979/319352737-15458522-a349-4ae2-a386-9ce272628ed7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVk [...]

Hey @ryanrolds, please share your design files with us. This looks so good!

mrred2k, 2024-06-01:
> Hey @ryanrolds, please share your design files with us. This looks so good!

If you have an idea what the clear Part could be, I will copy that Thing quickly.

Pferdebockwurst, 2024-06-01:
> If you have an idea what the clear Part could be, I will copy that Thing quickly.

Oh really? That would be so great. Unfortunately I have no idea for the clear part on top. It surely is made this way for the luminance sensor to work. But I don't have a 3D printer so I have no clue how this part could be printed.

mrred2k, 2024-06-04:
> > > > ![20240403_133607](https://private-user-images.githubusercontent.com/343979/319352737-15458522-a349-4ae2-a386-9ce272628ed7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyMzYwNDAsIm5iZiI6MTcxNzIzNTc0MCwicGF0aCI6Ii8zNDM5NzkvMzE5MzUyNzM3LTE1NDU4NTIyLWEzNDktNGFlMi1hMzg2LTljZTI3MjYyOGVkNy5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwMVQwOTU1NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT03ODgxYjk3YmY1NTE3ZTNjMTU0ZmY0NGUxYTA0YzcyZjg3ZDEzYjAzOTVkODY2NzFkYWMwZThjYTgwZTZjZjJhJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.NTOqFvJQGHlI0WP40XpPV8Cn7gAWSByoRkFbADdCHB8) ![20240403_135708](https://private-user-images.githubusercontent.com/343979/319352934-f2d91d2e-f179-4706-a8b2-7fc1cd36f2c4.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTcyMzYwNDAsIm5iZiI6MTcxNzIzNTc0MCwicGF0aCI6Ii8zNDM5NzkvMzE5MzUyOTM0LWYyZDkxZDJlLWYxNzktNDcwNi1hOGIyLTdmYzFjZDM2ZjJjNC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNjAxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDYwMVQwOTU1NDBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kZDdlOWJiYzE5M2EyYTY3NzdmNGRkMzQ3YTY5OTNkMWI0OGM0ZDU0YWI4MzE0ODRjZWY4Mjc1ZmU5MDdhZDQ1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.EL-Bd_hkVuNOu3jNI6_Vq_Ks21ytSos-KgyxVPbLW3I)
> > > 
> > > 
> > > Hey @ryanrolds, please share your design files with us. This looks so good!
> > 
> > 
> > If you have an idea what the clear Part could be, I will copy that Thing quickly.
> 
> Oh really? That would be so great. Unfortunately I have no idea for the clear part on top. It surely is made this way for the luminance sensor to work. But I don't have any experience with 3D printing so I have no clue how this part could be printed.

https://www.printables.com/de/model/901220-waterproof-case-for-b-parasite-soil-moisture-air-s

Pferdebockwurst, 2024-06-04:
> https://www.printables.com/de/model/901220-waterproof-case-for-b-parasite-soil-moisture-air-s

Awesome. This looks so good. Thank you very much!

ryanrolds, 2024-06-04:
The clear part on the top is a glass table top bumper with the nub cut off. I'm glad people appreciate the case, but it's not something I'm willing to share. On my list of today is to replace the bumper with a 3d printed light pipe. I also want to move the photo sensor to be top center on the PCB.

Pferdebockwurst, 2024-06-04:
> The clear part on the top is a glass table top bumper with the nub cut off. I'm glad people appreciate the case, but it's not something I'm willing to share. On my list of todos is to replace the bumper with a 3d printed light pipe. I also want to move the photo sensor to be top center on the PCB.

Cool story bro!

dingausmwald, 2024-08-29:
> I made a streached version to monitor temperature and moisture in my hay bale garden. [...]

those look awesome. Can you tell something about them? How long? Are they reliable? I see you just copied the water measurement part x3. I saw your fork is 20 commits behind and there is no info on the modifications.