rbaron/b-parasite #20: programming - low voltage detected at the target

SchmidL, 2021-11-12:
hi, 

I try to flash the firmware with my J-Link Edu. but unfortunately I get following error? As it's the first time flashing with swd, I cannot figure out what the issue is. After a successful compile, it would like to flash the softdevice part and encounter the following error: 
```
% SDK_ROOT=~/Documents/b-parasite-main/code/nRF5_SDK_17.1.0_ddde560 PLATFORM=E73_2G4M08S1E make flash_softdevice
Flashing: s140_nrf52_7.2.0_softdevice.hex
nrfjprog -f nrf52 --program /Users/lorenz/Documents/b-parasite-main/code/nRF5_SDK_17.1.0_ddde560/components/softdevice/s140/hex/s140_nrf52_7.2.0_softdevice.hex --sectorerase
ERROR: Unable to connect to a debugger.
ERROR: Low voltage detected at the target. Please make sure the device is
ERROR: properly supplied.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
make: *** [flash_softdevice] Error 43
```

same for just make flash, the Vcc is directly connected to the jlink. no coincell involved.
Any ideas? Thx

rbaron, 2021-11-13:
Hi @SchmidL,

Have you taken a look at the [How to Program on the Wiki](https://github.com/rbaron/b-parasite/wiki/How-to-Program)?

Can you show us how you're connecting your J-Link Edu to the target? Think to check:
1. You need to connect 4 wires from the J-Link to the b-parasite. Make sure you're also connecting the GND and Vcc
2. **Take off the battery** if you're connecting the Vcc from the J-Link to the b-parasite.

Please let me know if that helps.

SchmidL, 2021-11-14:
Hi @rbaron, 

yes I looked at it but in the end i followed the REAMDE from the /code https://github.com/rbaron/b-parasite/tree/main/code/b-parasite, as there where several issues/errors for me (`make flash-softdevice` is wrong, -> `make flash_softdevice`, `make flash-device` does not exist), but no problem.

Then I connected my J-Link Edu via USB to a USB-C hub to my Mac.
The sensor i connected like that:
- Pin 1 (VTref) to VCC
- Pin 4 (GND) to GND
- Pin 7 (SWDIO) to CLK
- Pin 9 (SWDCLK) to IO

I do not use any coin cell at this stage (to avoid any voltage uncertainty)

Any idea where I can start checking?

rbaron, 2021-11-16:
I just corrected the typos in the Wiki page.

I believe the J-Link EDU cannot power target boards. It means that you need to provide external power to the b-parasite, and also connect VTref to it. One idea is to then leave the battery in or supply 3.3V to it with an external power supply while flashing.

Meroje, 2022-01-14:
I'm using an EDU mini too, it had to have the reset line connected to move past this.

Unfortunately, in my case (with E73_2G4M08S1E modules) it's then looping on `ERROR: [SeggerBackend] - JLinkARM.dll reported "-1", "An unknown error.".` for nrfjprog, or a combination of `ConfigTargetSettings() / InitTarget()` loops with JLinkExe

rbaron, 2022-01-16:
Hi @Meroje. Did you manage to fix it?

I've had my fair share of these frustrating errors. It's hard to pinpoint the root cause from a distance, but from experience it's usually the following, in order of likelihood:

* Double check the GND, SWCLK, SWIO, 3V (provided from an external source if using the J-Link EDU);
* Faulty USB cable. This used to be the last thing I suspect and now it's the first. Consider trying different cables;
* Faulty solder joints/bridging/excess of solder paste. For manually soldered E73 modules, it's a real possibility. Consider ruling this out by trying to program a second b-parasite if you have access to one;
* Faulty programmers - unlikely the case if you're using an original J-Link EDU, but possible for those cheap clones;
* Driver issues. Make sure you have the most up-to-date J-Link drivers installed;

Please let me know if you manage to find the root cause.

Meroje, 2022-01-16:
> * Double check the GND, SWCLK, SWIO, 3V (provided from an external source if using the J-Link EDU);

Probe reports `VTarget = 3.223V`, using an external source (not battery), reseated cables multiple times and checked swio/clk
> * Faulty USB cable. This used to be the last thing I suspect and now it's the first. Consider trying different cables;

tried a few, seems unlikely
> * Faulty solder joints/bridging/excess of solder paste. For manually soldered E73 modules, it's a real possibility. Consider ruling this out by trying to program a second b-parasite if you have access to one;

I hand soldered 2 parasites so far, using rossman levels of flux. I have good confidence joints are ok on both but no way to be sure 100%
> * Faulty programmers - unlikely the case if you're using an original J-Link EDU, but possible for those cheap clones;

bought from adafruit. not sure what other chips I have to check on
> * Driver issues. Make sure you have the most up-to-date J-Link drivers installed;

Tried linux and windows so far
> Please let me know if you manage to find the root cause.

still no luck :/

rbaron, 2022-01-16:
@Meroje can you confirm you get a different error if you don't power the b-parasite?

You can also try issuing the `nrfjprog --reset --log` command. Maybe it will give you a hint of what's happening.

If you want to share a picture of your setup, I can take a look and see if there's anything visibly wrong.

Meroje, 2022-01-16:
removing external power gives the expected "low voltage" error

[image]
[jlink.log](https://github.com/rbaron/b-parasite/files/7877289/jlink.log)
[nrfjprog.log](https://github.com/rbaron/b-parasite/files/7877290/nrfjprog.log)
```
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link EDU Mini V1 compiled Dec  7 2021 08:38:51
 - Device "NRF52833_XXAA" selected.
 - Target interface speed: 4000 kHz (Fixed)
 - VTarget = 3.167V
 - InitTarget() start
 - InitTarget() end
 - InitTarget() start
 - InitTarget() end
 - ERROR: Failed to connect.
Could not establish a connection to target.
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link EDU Mini V1 compiled Dec  7 2021 08:38:51
 - Device "NRF52833_XXAA" selected.
 - InitTarget() start
 - InitTarget() end
 - InitTarget() start
 - InitTarget() end
 - Target interface speed: 200 kHz (Auto)
 - VTarget = 3.169V
 - InitTarget() start
 - InitTarget() end
 - InitTarget() start
 - InitTarget() end
 - ERROR: Failed to connect.
Could not establish a connection to target.
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link EDU Mini V1 compiled Dec  7 2021 08:38:51
 - Device "NRF52833_XXAA" selected.
 - InitTarget() start
 - InitTarget() end
 - InitTarget() start
 - InitTarget() end
 - RESET (pin 15) high, but should be low. Please check target hardware.
 - Target interface speed: 200 kHz (Auto)
 - VTarget = 0.000V
 - ERROR: Failed to connect.
Could not establish a connection to target.
Connecting ...
 - Connecting via USB to probe/ programmer device 0
 - Probe/ Programmer firmware: J-Link EDU Mini V1 compiled Dec  7 2021 08:38:51
 - Device "NRF52833_XXAA" selected.
 - InitTarget() start
 - InitTarget() end
 - InitTarget() start
 - InitTarget() end
 - Target interface speed: 200 kHz (Auto)
 - VTarget = 3.182V
 - InitTarget() start
 - InitTarget() end
 - InitTarget() start
 - InitTarget() end
 - ERROR: Failed to connect.
Could not establish a connection to target.
```

rbaron, 2022-01-16:
@Meroje, another thing to try. You mentioned you connected the RST pad on the b-parasite to the J-Link. In my workflow, I do not connect the RST pin to the programmer. I manually briefly tie it to GND when I want to reset the b-parasite.

Try the following:
- Completely disconnect the b-parasite RST pad;
- Connect the b-parasite RST pad to GND;
- Run `nrfjprog --reset --log` - it should hang;
- Disconnect RST from GND;
- The `nrfprog` command should unblock and continue to execute;

Please let me know if that works.

Meroje, 2022-01-16:
Without RST the jlink bails immediately with low voltage

rbaron, 2022-01-16:
Did you try the sequence of actions I suggested? What happened? Did it initially hang?

If it shows a low voltage warning, I would try a different external power source for 3.3 V. And make sure to remove the battery then.

Meroje, 2022-01-16:
Turns out the swd ribbon was backward (pretty sure I did try at some point): it has to face "inward" on the jlink-edu board

rbaron, 2022-01-16:
Happy to know it's working now.

Meroje, 2022-01-16:
looks like it succeeded, will check the packets
```
nrfjprog -f nrf52 --program s140_nrf52_7.2.0_softdevice.hex --sectorerase
Parsing image file.
WARNING: A programming operation has been performed without --verify.
WARNING: Programming can fail without error.
nrfjprog -f nrf52 --program nrf52833_xxaa.hex --sectorerase
Parsing image file.
WARNING: A programming operation has been performed without --verify.
WARNING: Programming can fail without error.

nrfjprog -f nrf52 --reset
Applying system reset.
Run.
```
anyway, thanks for the tips and sorry for the confusion

a83r, 2022-04-28:
For future reference, on my j-link edu mini board the cable needed to plugged in like that:
[image]

closed 2022-09-09 by rbaron