rbaron/b-parasite #21: Poll from battery-powered client

chris-b-tanner, 2021-11-21:
Hi - this looks like a fantastic device. We run a solar powered client (central control box) that currently uses wired sensors and I would like to change to wireless. However, the main unit sleeps most of the time, waking every 30 minutes. From what I know about BLE, it's possible to poll a sensor and still maintain a good sensor battery life. Is polling something you have thought of including in the b-parasite rather than broadcasting?

rbaron, 2021-11-21:
Hi @chris-b-tanner, thanks for the kind words.

I think it's totally possible to set up a polling scheme. We can define a BLE service & a characteristic that returns the sensor values when read. The question is how low we can keep the power consumption. At the very least, we need to turn the radio on for advertisements, and it has to be often enough that, when the central wakes up, it reads the broadcasts and is able to connect to the b-parasite to read data.

[This Xiaomi BLE thermometer](https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc) works like this. It broadcasts its values like we do here, but also exposes BLE services for configuration. We'd need to do some investigation on how to keep the power consumption low, probably by tweaking the advertisement interval & TX power.

On the other hand, I recently set up a [solar-powered, always-on BLE device](https://twitter.com/rbaron_/status/1457046900859981828) and I was thinking about its power requirements. Here's a back-of-the-envelope calculation: I think it's reasonable to expect a battery of let's say > 2000 mAh, which could power a BLE central chip (that consumes let's say 10 mA) for 200 h, even if there's no sun. So I don't think it's far fetched  to have an always-listening solar powered BLE central.

chris-b-tanner, 2021-11-21:
Hi @rbaron - that's interesting. Our central is ESP32 powered (lots in the field already, can't change that) with a 2900mAh battery. It's notoriously power hungry as you know, so we deep sleep. I'll look into whether we can have it always (or nearly always) listening for broadcast at a lower power ... but indications from early research are that it's still ~100mA. 

I briefly investigated an ESP32 sensor and built a functional link which synchronised both sensor and central wake up times ... so both could sleep. Central woke 5 seconds before sensors to allow the sensor to (1) send data packet then (2) receive updated time then (3) sleep again. But it's still too power hungry on the sensor end for an ESP32. More research led me to your w-parasite and then I ended up here.

Central can stay awake with radio listening for up to 30 seconds every half hour without too much impact on battery power, but no more (especially in winter). Our challenge is working with the control units already in the field, and not making any changes on that board.

I considered using the [Xiaomi / generic BLE soil sensors](https://www.amazon.com/Wanfei-Intelligent-Bluetooth-Fertility-Temperature/dp/B07ZH7FQJ7) which can be made to work with ESP32 (I read) - but we need something IP67 so will need to develop our own potting/enclosure. If there's a way to get the b-parasite to be polled in a similar way, we'd be happy to contribute tooling designs for an IP67 version.

rbaron, 2021-11-22:
@chris-b-tanner, got it. I think implementing a read BLE service for polling is a reasonable way forward. Unfortunately I can't work on that right now, but I'm happy to provide guidance for you or someone else who wants to work on it. I would only push that this is a config-enabled, off-by-default change.

Here's another thought. Let's say we have that service so that clients can connect to b-parasite and read from it. In order to connect, clients need to wait for broadcast from b-parasite. But, if they receive the broadcast (as is), they already have the data, and won't need to connect. In reality we would probably optimize the advertisement packet for size/less energy, but it seems like a good temporary compromise to just decrease the deep sleep (let's say to 20 seconds) & experiment with TX power. This can be experimented with with a two line code change ([here](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/config/prst_config.h#L17) and [here](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/config/prst_config.h#L41). The more time consuming work would be to profile the power consumption consumption, but the code change is tiny.