rbaron/b-parasite #22: Bad availability of nRF52 module - alternatives?

sefranz, 2021-12-02:
Hello there, lately I stumbled upon this awesome project and was totally set on building a dozen of those.
Unfortunately, the nRF52 Bluetooth module seems to be more than a little elusive right now, in my region starting only at €15 per piece. 

Therefore, my question(s) did you ever consider any alternative Bluetooth modules? Or if not, could you elaborate on the specific features used by the nRF52 module, e.g. the requirements for other modules?
Then I could try to adapt the board accordingly.

rbaron, 2021-12-02:
Hi @sefranz,

Indeed these nRF52 modules got hit hard with the current semiconductor shortage both in price and availability. The original project uses a nRF52840 module (E73-2G4M08S1C). A while ago we added support for an (almost) pin-compatible nRF52833 module, the E73-2G4M08S1E (details in rbaron/b-parasite#13). This was done exactly because of the shortage.

These nRF52833 chips are currently a little easier to find. I can see them for around 5 USD on AliExpress, which is similar to the original price of the nRF52840 module, and they work just as well. This would be my recommendation.

If you'd like to use another SoC from a different vendor, it would have to have a few of ADC channels for sampling the sensors (soil, photoresitor, battery monitoring), be able to generate a PWM signal, be powered by a CR2032 in terms of voltage range and max current it draws and be able to deep sleep in the low microamp range. The software would also have to be rewritten to use a different SDK.

janpfo, 2022-01-08:
Hello @rbaron!
Is E73-2G4M08S1E using the same PCB as E73-2G4M08S1C? Is there any reason for not buying the E-model?

rbaron, 2022-01-08:
Hi @janpfo,

For b-parasites, they work pretty much the same. If you use the E73-2G4M08S1E one, you just have to specify `PLATFORM=E73_2G4M08S1E` when compile - [docs here].

closed 2022-01-08 by rbaron