rbaron/b-parasite #29: Negative Temperature Values

jhbruhn, 2022-03-13:
Wanting to use one left-over b-parasite to verify the temperature of my fridge, I noticed that the device does not transmit negative flawlessly (at least not to the ble_monitor solution). Is it a problem with the encoded packets or with the decoding?

As a plant-logging use-case, it might come in handy for plants which are on a balcony, or in the garden, although that specifically will not happen for my plants right now.

rbaron, 2022-03-13:
It seems like we:
1. Read the value out of the temp sensor as a signed 16 bit int [here](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/prst/shtc3.c#L61) (the [`temp_millicelcius`](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/prst/shtc3.h#L19) field is an `int16_t`)
2. But we [pass it to the `prst_ble_update_adv_data`](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/main.c#L80) that [expects an `uint16_t`](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/prst/ble.c#L163) and does some [bit-shifting](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/prst/ble.c#L173)
3. Then on the ESPHome bridge for example, I [handled it as an `uint16_t](https://github.com/esphome/esphome/blob/95a6715b2b3ebc30d31430b24beb888583a1f913/esphome/components/b_parasite/b_parasite.cpp#L61).

So I would say this is a bug. I believe a good solution would be to treat it as `int16_t` everywhere.

rbaron, 2022-03-13:
I did a quick test and it seems a quick workaround is just changing the decoding side to read the value into a signed `int16_t` instead:

```C
int main(void) {
  int16_t n_sig = -100;
  uint16_t n_unsig = n_sig;

  uint8_t encoded[2];
  encoded[0] = n_unsig >> 8;
  encoded[1] = n_unsig & 0xff;
  
  int16_t decoded = encoded[0] << 8 | encoded[1];
  
  // Prints -100.
  printf("%d\n", decoded);
}
```

But that invites another limitation. By using a signed 16-bit integer, we can only represent values in [−32768, +32767], which would map to °Celcius in +- 32. 

The SHTC3 sensor is only accurate to 0.2 degrees Celcius, so encoding the temp in °Celcius * 1000 is unnecessary. We could instead encode is in °Celcius * 100, which would give us +- 327 °C range (this is exactly what I ended up doing [in another project](https://github.com/rbaron/sugarboat/blob/main/code/sugarboat/lib/sugarboat/sugarboat/ble.cpp#L183)). 

I think this would warrant a bump in the [protocol version](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/config/prst_config.h#L31) and appropriate handling in the bridge code.

jhbruhn, 2022-03-13:
That sounds reasonable. I can make the appropriate PR to the ble_monitor HA component as soon as the new protocol version is defined.

On another note, while we are at it bumping the protocol version: There are BLE sensors which are using the same BLE-UUID to broadcast sensor information, specifically Xiaomi temperature sensors using the ATC firmware. The HA component currently differentiates between them using the length of the advertised message, which is obviously quite instable (see discussion https://github.com/custom-components/ble_monitor/pull/597#discussion_r776682917). One solution could be to add an identifier-byte somewhere into the message, perhaps to the end? Or we leave it as is and go for good luck... Alternatively we could switch to a new UUID as suggested in the comment thread of the referenced PR.

rbaron, 2022-03-13:
I think you answered the comment above very similarly to what I would. Initially I thought about encoding b-parasite's data into the "manufacturer specific data" section of the BLE advertisement packet. But for that I believe we'd need a registered [company ID](https://www.bluetooth.com/specifications/assigned-numbers/company-identifiers/). To avoid stepping on other people's toes, I instead opted for using a generic "environmental sensing UUID" (`0x181A `) and advertising the data in the service data section of the advertisement packet. In theory, we could have multiple service data sections.

The full BLE payload contains some useful info. By setting `PRST_BLE_DEBUG 1` (comments mine):
```
[ble] Encoded BLE adv packet
// flags section
[ble] 0x02  // length: 2 bytes
[ble] 0x01  // AD type 0x01; flags
[ble] 0x04
// Service data section
[ble] 0x15  // length: 0x15 bytes
[ble] 0x16  // AD type 0x16 - Service data
[ble] 0x1A  // This and next spell 0x181a (generic env. sensing 16-bit  UUID)
[ble] 0x18
[ble] 0x11  // First byte of our encoded data
[ble] 0x01
[ble] 0x0C
[ble] 0xF3
[ble] 0x5A
[ble] 0x6F
[ble] 0x3C
[ble] 0xAB
[ble] 0x00
[ble] 0x00
[ble] 0xF0  // Bytes 10-15 (inside the service data section) encode the MAC address itself (f0:ca:f0:ca:01:01)
[ble] 0xCA
[ble] 0xF0
[ble] 0xCA
[ble] 0x01
[ble] 0x01
[ble] 0x0A
[ble] 0xA9
// Name section.
[ble] 0x05  // length; 5 bytes
[ble] 0x09  // AD type 0x09 - name
[ble] 0x70  // This and next 4 bytes spell "prst"
[ble] 0x72
[ble] 0x73
[ble] 0x74
```

So we have a few possibilities:
1. Change from `0x181a` to something else. If we opt out of this generic env. sensing UUID, we'd probably have to use a 128 bit random UUID instead, which would make the whole adv. packet to exceed 31 bytes.
2. Use the extra info to differentiate b-parasite's advertisement from others. We could assert that the name in the adv. packet is equal to `"prst"` (this data is "outside" the service data section) and/or assert that bytes [10-15](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/prst/ble.c#L35) ("inside" the service data section) match the same MAC address as the the bridge sees[^1]

I think going for option 2 is a good compromise. LMK what you think.

[^1]: Having the MAC address inside the adv. packet seems redundant, but this is useful on macOS/iOS, which mostly don't give us the real MAC address of BLE peripherals

jhbruhn, 2022-03-13:
Checking the Mac bytes seems like a good idea as long as the Mac address matches the one provided. But at least for the HA plugin, that seems to be the case. I will implement that check as soon as the new format is finished.

PR #30 (2022-03-15): Implements BLE protocol v2
The only change from v1 is how the temperature is encoded. In v2, two
bytes at offset 4 and 5  represent a 16-bit signed int (in big-endian).
It contains the temperature in degrees Celsius *  100.

Fixes #29
commit 8fd4d722 (2022-03-15): Implements BLE protocol v2

The only change from v1 is how the temperature is encoded. In v2, two
bytes at offset 4 and 5  represent a 16-bit signed int (in big-endian).
It contains the temperature in degrees Celsius *  100.

Fixes #29
closed 2022-03-15 by rbaron