rbaron/b-parasite #33: Conformal coating / Case

a83r, 2022-03-29:
Hi, 
I want to use the sensor outside on my balcony. Would it be possible to coat the pcb and what parts would I need to spare?
Regarding the case, is there any issue with temperatures, is there any max temperature which shouldn't be exceeded?
Thanks a lot for providing the whole informations and the project itself!

Ianszh, 2022-03-29:
Dito

rbaron, 2022-03-29:
Hi @a83r,

I have only experimented with acrylic coating so far (details in this [twitter thread](https://twitter.com/rbaron_/status/1491467468996431873)). Nowadays I tend to coat only the bottom half of the PCB with this [PLASTIK 70 spray](https://www.farnell.com/datasheets/318249.pdf). I would love to test epoxy dipping, but I haven't had the chance yet.

As for the case, it depends on the type of filament/resin you use. I have an FDM printer and I use PETG filaments for prints that are exposed to the weather. It seems to hold up well.

a83r, 2022-03-29:
So I could coat the whole PCB? Sparing the battery holder and the 2 lanes going down?

rbaron, 2022-03-29:
I tend to coat only the bottom half of the board and use a case. If you coat the whole board, I would avoid the battery holder and the temp sensor.

a83r, 2022-03-29:
Thanks again! I will try to coat it. Funny rotisserie you build there.

rbaron, 2022-03-29:
Let us know how it goes! Hopefully I can get around to set up a wiki page to document all of our efforts with coating. Good luck!

closed 2022-03-29 by rbaron

NathanSweet, 2022-08-24:
Would an aluminum PCB work?