rbaron/b-parasite #34: Minimal Case Design

drspangle, 2022-04-28:
I've been working on getting a bulk order of these sensors together for a home garden sensing project, and I've noticed that it might be possible to drastically reduce the amount of material used in the case without compromising the overall design. Is there any chance that someone with access to a 3d printer might want to take a crack at prototyping? I'm afraid I'm using an outsourcing service for mine, as I haven't got a printer.

drspangle, 2022-04-28:
Also, could anyone provide the original design files for the case? I can't import the STEP file into Fusion 360.

rbaron, 2022-04-28:
@drspangle did something recently change in Fusion 360? I am able to import the STEP file. I'm on version 2.0.11894 on macOS.

drspangle, 2022-04-28:
I'm not sure. I'm using version 2.0.12888 on Windows 10. When I try to import the STEP file it just hangs. I'll give it a try again today and see if I can't make it work.

rbaron, 2022-04-28:
@drspangle unfortunately I haven't ran any comparative tests for the humidity sensor with and without case. It would be interesting to come up with a case design that allows for better airflow in case this is an issue, while still protecting the board well.

For the illuminance sensor, if you want to use a case, my best suggestion would be to print it in a transparent material.

drspangle, 2022-04-28:
I think I've managed to get a few useful features implemented in the design, so that it doesn't have to be printed in a transparent material (but conformal coating, even the cheapest that can be found, would be recommended). This way the sensor should hold up better to splashes or even possibly outdoor use, provided it's used in less damp climates. You can find this in #35. I've asked other users in #3 to see if they can test out the design's printability (it should be as good as the original).

If anyone manages to get a print of the updated case out there and has some pictures, I'd be very grateful if you could share them here. Unfortunately I don't have access to a printer so I can't iterate on the case design very much. In an ideal world I'd like to see if I could incorporate some TPU gaskets on some of the inner surfaces so that a truly water resistant design would be possible, with minimal conformal coating.

JoelWise, 2022-04-28:
You might want to take some design inspiration from [weather station shields](ttps://www.google.com/search?q=weather+station+shield) to maintain temp and humidity accuracy.

I've also seen prints with a 1.75mm hole so a clear piece filament pressed and glued into it as a light pipe. Maybe a good idea for the light sensor and led.

drspangle, 2022-04-28:
> You might want to take some design inspiration from [weather station shields](https://www.google.com/search?q=weather+station+shield) to maintain temp and humidity accuracy.

This is a nice idea, but it runs a bit counter to my "reducing the amount of material" goal. To be honest I think the current interference-fit clamshell case design is a good one, the main areas where water infiltration is an issue is the programming pads and the battery connector, the rest (including the photoresistor) can be conformally coated and become effectively waterproof cheaply and easily. Because there needs to be gas exchange with the humidity sensor, the top of the board has to be exposed to the outside environment no matter what. Obviously the humidity sensor shouldn't be coated but I don't worry much about corrosion there. 

I believe the programming pads are also less of an issue if they are solder-filled and coated on the top side, so they can be programmed from underneath on the uncoated side. A future hardware iteration might consider to put these as actual pads and only on the bottom, but this would require a fair bit of reworking and might not be worth doing. When conformally coated, the design of the case is mostly needed to prevent water from intruding on the bottom and into the battery clip, and that's where I think a TPU gasket could work to surround this portion of the board and keep moisture out. 

I sure wish I had a 3d printer... I'm going to need to start thinking about buying one.

JoelWise, 2022-04-28:
Sounds cool. I'm all for having an improved case design! I'd love to have a 'hat' case that just securely slides on from the top that copies the design of weather sensor covers. Combined conformal coating it could work really nice!

hallsny, 2022-04-29:
I designed a  single piece thin wall l cap out of black ASA  that slides on over the upper section of the sensor.  The goal was to protect against water ingress for the upper section  via no additional case seams ( these will be outside 24/7  with automatic watering )  The bottom is open to allow some airflow, but the temp swings from sunlight limit the effectiveness of the temp/humidity measurement during the day. Even with a clear material to enable the illuminance measurement , I don't think  you could get an accurate temp measurement without significant airflow through the case. .

I'm reprinting a  few in white filament to  see if the temp performance is any better.
[image]

drspangle, 2022-04-29:
The humidity levels are dependent on temperature since I think it measures relative humidity, so that's why my updated design has air holes to promote freer gas exchange. It would be great to be able to more scientifically test the benefits and drawbacks in terms of these different case designs against one another.

JoelWise, 2022-05-02:
> I designed a single piece thin wall l cap out of black ASA that slides on over the upper section of the sensor. The goal was to protect against water ingress for the upper section via no additional [...]
 
Can you share the design files? This would be a great base to build a pagoda-style cover that I suggested earlier.

hallsny, 2022-05-03:
I had a go at a higher airflow  printable cover. This should be printable upside-down as long as the printer is set up to bridge gaps and can print a 30 degree overhang.

[image]
[image]

JoelWise, 2022-05-03:
You beat me to it! Quite the upgrade! Printed in PETG. Good fit at default size but maybe be better at 101%.

[image]

[image]

drspangle, 2022-05-03:
Impressive work!

I take it that transparent PETG or similar material is required to make use of the photosensor? 
Do you see a noticeable difference in the temp/humidity readings with this case versus without, or versus other case designs?

JoelWise, 2022-05-03:
I used PETG because my printer was setup for it. I would use white PLA but I don't have any at the moment!

I have found the photosensor to be useful only when the probe is monitoring small nursery plants since other larger plants' leaves shade the sensor. Ambient temp readings seem to be unaffected. Humidity reading are 10-15% above ambient but that is to be expected from a sensor that sits just inches above wet soil. Hah.

rbaron, 2022-05-04:
@hallsny that case looks gorgeous! Would you consider submitting a PR with it? Thank you!

JoelWise, 2022-05-05:
@hallsny I'm still interested in getting a copy of the slim case if possible. Thanks!

hallsny, 2022-05-05:
> @hallsny that case looks gorgeous! Would you consider submitting a PR with it? Thank you!
I would like to make some design update first and then can submit a PR.

 In general the difference  between the  high airflow case and my standard cover has been minimal in a side by side comparison. 0 is high airflow, 1 is standard, and 3 is a sensor with a normal cover in a  different shaded spot (low sun control). I may try opening up the sides to get more airflow through the cover.
[image]

> @hallsny I'm still interested in getting a copy of the slim case if possible. Thanks!

Here you go!
[soil sensor cover v4.zip](https://github.com/rbaron/b-parasite/files/8628927/soil.sensor.cover.v4.zip)

JoelWise, 2022-05-05:
Thanks!

Maybe humidity readings benefit more? The airflow case is still pretty slim and probably needs to extend far further to give the needed shade. Even then, the sensors are right at ground level which doesn't seem ideal if you expect accurate ambient readings. I'll share some of my own comparisons in a few days.

drspangle, 2022-05-05:
I'm starting to come to the conclusion that there are fundamental trade-offs between getting good light measurements, and the sensor's ability to read soil moisture and air humidity. In an ideal world, the light sensor would be someplace away from the device, but that would make it a piecemeal solution (perhaps a separate sensor solely for light/temperature/humidity readings is worth investigating).

The important question that is raised for me is, does the accuracy of the capacitive measurements for soil moisture have any dependency on temperature, air humidity, or light readings? If not, these sensors are no longer "must haves" on this platform and perhaps the size and design of the PCB and therefore the case could be simplified in a more specialized variant. A separate variant design that abandons the capacitive sensor and only has the other sensors could do that job a whole lot better. After all, there's no reason for everything to happen all at once on one device unless it really needs to, right? Thoughts?

JoelWise, 2022-05-05:
I always have viewed the other sensors as fun-to-have. They cost next to nothing to add and implement if you don't have a need to extreme accuracy. Moisture sensing ease and accuracy is #1. If anyone starts to have big redesign requirements then it's time to fork and roll your own.

drspangle, 2022-05-05:
I think that totally makes sense and is the same philosophy I have arrived at. The reason why I decided to look into this project was the capacitive sensor, the comms, and the prospect of long battery life -- not the other sensors. 

Once I've brought up some of these boards and tested them out I'll probably take a crack at a fork for the sunlight and temperature measurements on a separate board, but it might prove unnecessary anyway. Another fork might be a dedicated actuator (irrigation, for example), built on the same foundation. Getting these to all work together is a software problem which is much simpler to solve and to version control than all this hardware waffling.

JoelWise, 2022-05-12:
Temp and humidity vs local weather service using the high airflow case. The probe is in a 5L pot and mostly shaded. Local weather is mostly cloudy, rainy, and cool.

[image]
[image]

jhbruhn, 2022-10-10:
Does someone have these (very nice!) case-files as an .stl file? I don't have access to Fusion 360. Preferably of course as a PR to this repository! :)

hubikj, 2022-12-05:
I recently played around with the high airflow case and it was too tight for me and the battery clip was getting in the way. So I made it slightly larger with bigger tolerances. You can check my results here: https://www.printables.com/model/306214-high-airflow-case-for-b-parasite

closed 2023-02-12 by rbaron