rbaron/b-parasite #38: Add conductivity measurement

jhbruhn, 2022-05-01:
It seems that soil conductivity is an indicator for nutrient-content of the soil. If my understanding is correct, it could be realized via a simple soil-resistance measurement, where the electrodes are spaced with a known distance. 

As exposing bare copper would lead to very easy corrosion there, the electrodes could be realised via (stainless steel) screws which are screwed into electrode-screw-holes?

This would obviously mean a rather large redesign, potentially influencing the current moisture measurement circuit (esp. the capacitor) as well (unless introduced as a second prong?).

drspangle, 2022-05-01:
You're still going to see galvanic breakdown of something in this circuit due to corrosion. Even though the stainless is unlikely to, this will more likely cause copper up the line to corrode. If you really really want to measure this way, using sacrificial anodes might be the way to go. I'm not sure that will ultimately be a workable solution for you though.

jhbruhn, 2022-05-01:
A low maintenance solution would be preferable of course. 
I wonder what the MiFlora sensor is using as an electrode then, it does look like some kind of metal rivet.

JoelWise, 2022-05-02:
The TTGO T-Higrow tries to do capacitance and conductivity in one sensor too. No clue of they are easy to implement or even reliable. A round dome switch is the only standard component that might work as a pick and place solution.

UntwistedApple, 2023-08-12:
I know this board is supposed to be quite standalone, but I am thinking about buying something like this [0] (Just a set of 2 electrodes) and trying to add a connection for it to the board plus a voltage divider. What do you think?
[0] https://de.aliexpress.com/item/1005005366099673.html