rbaron/b-parasite #39: Mosfet instead of Q2 and Q3 bipolar?

pit34, 2022-05-07:
Hello,

I ask myself why does Q2 and Q3 are bipolar tranistor and not mosfet ?
There is an explanation of operation here :
https://wemakethings.net/2012/09/26/capacitance_measurement/
But not why bipolar instead of mosfet to discharge the capacitor.

Thanks

rbaron, 2022-05-07:
Hi @pit34,

I think a MOSFET would work as well, possibly with fewer passives. But if I were to rework the electronics, I would probably go with an Op-Amp instead of rolling out my own amplifier with discrete transistors.

closed 2022-05-07 by rbaron

pit34, 2022-05-07:
Thanks @rbaron 
Can you tell me more about your Op-Amp solution?
I can't see why a mosfet after R1, acting like a simple closed switch when square wave is low, are not the simpliest and "best" solution ? Why using amplifier?
Thanks in advance

rbaron, 2022-05-07:
There are two things here:
- Q3 is already a p-channel MOSFET. It's used to protect the circuit from sticking the battery in the wrong way (reverse polarity protection);
- Q1 & Q2 (that's Q2 and Q3 in the article you linked), are used to improve the low-pass filter & integrator, so we get a bigger swing on the output (the article explains why)

So the circuit works without Q1 and Q2, but it has small dynamic range. One idea is to use an op-amp on the output to improve this dynamic range. This has some advantages (e.g.: more linear, less power, simpler circuit) and some disadvantages (we also amplify the noise). I'm not 100% sure this is better, but it's something that I would like to experiment with. I hope this answers your question.

pit34, 2022-05-07:
My apologies, I haven't seen the differences between the articles and your schematic... I though it was a fork. Sorry.
I understand your Op-Amp solution, and why Q1 and Q2 are used to improve dynamic range.
But I still not understand why you too use bipolar for your Q1 and Q2 instead of a p-mosfet. Sorry if this is a stupid question..

Another question, why using a "FAST_DISCH_EN" signal instead of just connecting this to "PWM" signal ?

rbaron, 2022-05-07:
@pit34 you can find many resources on BJTs vs MOSFET on the web. In this particular case like I said, it doesn't matter too much - either should work I believe.

The FAST_DISCH_EN signal is essentially 3.3V, but it's software enabled so it doesn't drain the battery during deep sleep. The signal on the base of Q1 (Q3 in the article) is still the PWM signal.

pit34, 2022-05-07:
Thanks for your reply. I understand this part now ;-)