rbaron/b-parasite #42: Orientation of Q4 (Photoresistor) on v1.2.0 - JLCPCB is asking

tispokes, 2022-05-18:
Hi, got this mail from JLCPCB:
[image]

I used the latest `parasite.csv` as BOM and `parasite-top-pos.csv` for positioning.
As I'm not so good in reading schematics, I Prefer to ask.

[b]My opinion: It's wrong.[/b]
The Emitter (PHOTO_OUT) has to be connected to Pin 7 and not 8. For the collector the same just vis-à-vis.

@rbaron is the pos csv maybe wrong here?

rbaron, 2022-05-18:
tl;dr: Q4's collector goes on the left. The original position is correct and JLC's suggestion is not.

Often engineers from JLCPCB will suggest this change (see #3). I just made two changes to help with this confusion:
1. Made the footprint dot for Q4 (which represents the collector, sometimes annotated with a `+`, with a `|` or with a `C` by JLC ) more visible (it's partially cropped in JLCPCB's render)
2. Added instructions about Q4's orientation in the [How to order](https://github.com/rbaron/b-parasite/wiki/How-to-order:-PCB-fabrication-and-SMT-assembly) Wiki page.

Hopefully this helps.

closed 2022-05-18 by rbaron

1technophile, 2022-08-13:
Note that with the files from [cc8dcf9](https://github.com/rbaron/b-parasite/tree/cc8dcf92403dc3343cd0ad5bbc1a4e40520965fd), the photodiode is per default oriented in what seems to be the wrong direction, when importing the file with JLCPCB. Before any JLCPCB engineer checks.

[image]

[image]

rbaron, 2022-08-13:
Thanks @1technophile. I have a feeling the base internal orientation unfortunately changes at JLCPBC from time to time, whenever they update the component model. It was correct when I first set it up and also in the original comment above. If somebody has a better explanation, please let us know. 

Please always double check the orientation [with the wiki](https://github.com/rbaron/b-parasite/wiki/How-to-order:-PCB-fabrication-and-SMT-assembly#components-orientations) whenever placing a new order. There you can see the orientation unambiguously, independent of the internal, supposedly unstable representation JLCPCB uses.