rbaron/b-parasite #48: Major battery life improvement

rbaron, 2022-08-08:
It turns out the SoftDevice lets us put our application in deep sleep while it keeps advertising.

In the current firmware, we have [this piece of code](https://github.com/rbaron/b-parasite/blob/8fd4d7222af9b58073561325c0e873f301d20f42/code/b-parasite/src/main.c#L83):
```C
  prst_adv_start();
  nrf_delay_ms(PRST_BLE_ADV_TIME_IN_MS);
  prst_adv_stop();
  symbolic_go_to_deep_sleep_for(PRST_DEEP_SLEEP_IN_SECONDS);
```
We can change it into something like this:
```C
prst_adv_start();
symbolic_go_to_deep_sleep_for(PRST_BLE_ADV_TIME_IN_MS);
prst_adv_stop();
symbolic_go_to_deep_sleep_for(PRST_DEEP_SLEEP_IN_SECONDS);
```

### Impact in power consumption
To see the difference, I set:
```C
#define PRST_DEEP_SLEEP_IN_SECONDS 3
#define PRST_BLE_ADV_TIME_IN_MS 1000
```

Before:
[image]

After:
[image]

This trick brought the average current while awake from 7.48 mA to 1.63 mA. This drop has a meaningful impact on the oveal battery life.

In both cases the full deep sleep (no app + no advertising), hovers around 3.24 uA:

[image]

If we plug these new values in the [battery estimation spreadsheet](https://docs.google.com/spreadsheets/d/157JQiX20bGkTrlbvWbWRrs_WViL3MgVZffSCWRR7uAI/edit#gid=0), for a cycle of 10 minutes:

Before:
[image]
After:
[image]

These numbers are optimistic, but the relative increase is very real. I am looking forward to putting this to test in the real world.

JoelWise, 2022-08-08:
Very cool! Has this been completely implemented in sleep-adv branch yet? The code still seems to just delay during advertise rather than deep sleep.

prst_adv_start();
nrf_delay_ms(PRST_BLE_ADV_TIME_IN_MS);
prst_adv_stop();

rbaron, 2022-08-08:
@JoelWise yes, it's implemented in full there. The old delay has been removed - you can check the [`main.c`](https://github.com/rbaron/b-parasite/blob/5cfd69479a315078a2802253cea685ed24289380/code/b-parasite/src/main.c) in that branch.

It's a little different from what I illustrated in the issue here. The firmware wakes up from deep sleep after a timer. So when it wakes up we just need to check whether it's waking up from full deep sleep ([SLEEPING](https://github.com/rbaron/b-parasite/blob/sleep-adv/code/b-parasite/src/main.c#L21) state) or just from the "advertising sleep" (ADVERTISING state):
- If it wakes from the SLEEPING state, it runs the application, starts advertising and go into the ADVERTISING sleep state for `PRST_BLE_ADV_TIME_IN_S` (in other words: configures the timer to wake the system up after `PRST_BLE_ADV_TIME_IN_s` seconds)
- If waking up from ADVERTISING, it means we finished advertising, so we just stop advertising and go into the SLEEPING full deep sleep for `PRST_DEEP_SLEEP_IN_SECONDS` (in other words: configures the timer to wake the system up after `PRST_DEEP_SLEEP_IN_SECONDS` seconds)

JoelWise, 2022-08-08:
Got it. I flashed one of my probes and found similar battery life results. My Power Profiler suggests at a default 5 minute update frequency the change improves battery life by 3x!

[image]

[image]

[image]

rbaron, 2022-08-11:
I've reflashed all my b-parasites with this code and I'm happy to say they've been working well. I merged #49, but I'll leave this PR open for a while for visibility.

commit 7ebfe7e3 (2022-09-09): Update "Battery Life" section in README.md

Fixes #48.
closed 2022-09-09 by rbaron