rbaron/b-parasite #51: Flashing Issue

binaryhellstorm, 2022-08-16:
I've been trying to flash a couple of these PCB's and I am getting this error on one of them 

user@box:~/Desktop/b-parasite/b-parasite-main/code/b-parasite$ make flash_softdevice
Flashing: s140_nrf52_7.2.0_softdevice.hex
nrfjprog -f nrf52 --program /home/user/Desktop/b-parasite/nRF5_SDK_17.0.2_d674dde/components/softdevice/s140/hex/s140_nrf52_7.2.0_softdevice.hex --sectorerase
Failed to read device memories.
ERROR: JLinkARM DLL reported an error. Try again. If error condition
ERROR: persists, run the same command again with argument --log, contact Nordic
ERROR: Semiconductor and provide the generated log.log file to them.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
make: *** [Makefile:316: flash_softdevice] Error 33

Not sure if I maybe cooked my Bluetooth module when I used the hot air reflow station on it, have some more on order but wasn't sure if I was missing something silly in the meantime.

chucknorris101, 2022-08-16:
If it helps, I have been assembling with a hot air gun as well and have no issues. 

From my poking around with alternative programmers, and with the jlink on the DK working now, if you are getting this message, it at least knows it is an NRF (not cooked) - not being able to read the memories may be due to the chip being locked. Try a nrfjprog -f NRF52 --recover first then the make commands as this should remove the chip lock

binaryhellstorm, 2022-08-16:
Ok, gave that a shot and get this

user@box:~/Desktop/b-parasite/b-parasite-main/code/b-parasite$ nrfjprog -f NRF52 --recover
Recovering device. This operation might take 30s.
ERROR: Recover failed. Please make sure that the correct device family is given
ERROR: and try again.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.

chucknorris101, 2022-08-16:
hm - if you run recover cmd with the -log or --log it should give you a log file to poke at to see what is hanging it up, or same thing with the original command. 

I still dont think its cooked as it would say that its not an NRF in the earlier command, i would think. I have one board that is still giving me issues in recovery as well - i think its an issue with the soldering, so my plan is to reflow it at some point and retry. I may also poke around with the ESP32 SWD flasher to see if it recognizes it as NRF as well which could be another way to confirm its not cooked.

binaryhellstorm, 2022-08-16:
I think you're right that it's a software issue. When I run with the --log flag I get

[error]: [SeggerBackend] - JLinkARM.dll reported "-1", "An unknown error.".

Which is weird as I'm on Ubuntu so there shouldn't be DLL's being used at all.

schejbity, 2022-08-16:
I had same issue.
Solved with "make flash_loop"
Look in to the wiki...

> $ make flash_loop
> At this point, there's gonna be a lot of output in the terminal. That's normal.
> On the b-parasite board, briefly connect and disconnect the RST pad to GND. This will
> cause the b-parasite to RST, and the flash-loop command will catch it before it goes
> to sleep.

binaryhellstorm, 2022-08-16:
I did see the note in the wiki but it seemed to indicate that was for previously flashed boards, so I ignored it.

I gave that a shot though and I get this output:
nrfjprog -f nrf52 --program _build/nrf52840_xxaa.hex --sectorerase
ERROR: Unable to connect to a debugger.
ERROR: JLinkARM DLL reported an error. Try again. If error condition
ERROR: persists, run the same command again with argument --log, contact Nordic
ERROR: Semiconductor and provide the generated log.log file to them.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
make[1]: *** [Makefile:310: flash] Error 33
make[1]: Leaving directory '/home/user/Desktop/b-parasite/b-parasite-main/code/b-parasite'
make[1]: Entering directory '/home/user/Desktop/b-parasite/b-parasite-main/code/b-parasite'
DONE nrf52840_xxaa
Flashing: _build/nrf52840_xxaa.hex
nrfjprog -f nrf52 --program _build/nrf52840_xxaa.hex --sectorerase
^Cmake[1]: *** [Makefile:310: flash] Interrupt
make: *** [Makefile:329: flash_loop] Interrupt

When I pull the RST line to ground the output stops until it's disconnected. Tried it a few times, started the command, pulled RST to ground for a second or two, sadly no change.

binaryhellstorm, 2022-08-17:
Got it to work on at least one unit. I think my crap soldering was to blame, in fairness this is my first time doing any sort of SMT soldering that wasn't more in the warranty violating arena. 

Thanks everyone for the suggestions and putting up with my dumb questions.

closed 2022-08-17 by binaryhellstorm