rbaron/b-parasite #54: Consider switching to nRF52840 based module like Ebyte E73-2G4M08S1C instead to also allow Zigbee and Thread firmware?

Hedda, 2022-09-05:
Please consider switching to nRF52840-based module like Ebyte E73-2G4M08S1C instead to also allow Zigbee and Thread firmware?

https://www.ebyte.com/en/product-view-news.aspx?id=445

https://community.element14.com/members-area/personalblogs/b/blog/posts/trying-out-the-chengdu-ebyte-nrf52840-module

Alternatively could use nRF52840 based radio modules from RFstar like RF-BM-ND05 or RF-BM-ND05I :

https://www.szrfstar.com/product/list-182-en.html

https://www.szrfstar.com/product/155-en.html

https://www.szrfstar.com/product/219-en.html

nRF52840 can support Bluetooth, Zigbee, and Thread so could make a universal hardware design using different optional firmware.

https://devzone.nordicsemi.com/f/nordic-q-a/37602/zigbee-thread-support-for-nrf52832

jhbruhn, 2022-09-06:
This project is using exactly that module, or optionally the E Variant with an nrf52833 MCU.

Hedda, 2022-09-07:
Ah, cool, I thought it used the E73-2G4M04S1B which is based on nRF52832 so only supports Bluetooth and not Zigbee or Thread.

https://www.ebyte.com/en/product-view-news.html?id=243

So you think it would be possible to provide optionally Zigbee and/or Thread firmware images as alternative to Bluetooth? See:

https://community.home-assistant.io/t/zigbee-flower-sensors-for-soil-moisture-light-temperature-humidity/272803/97

and whole thread discussion here about Zigbee flower sensors for soil moisture:

https://community.home-assistant.io/t/zigbee-flower-sensors-for-soil-moisture-light-temperature-humidity/272803

rbaron, 2022-09-09:
Yes I believe it would be possible.

closed 2022-09-09 by rbaron