rbaron/b-parasite #55: Consider redesign board for CR2477 or CR2450 battery instead of CR2032?

Hedda, 2022-09-08:
Suggest re-design board with battery-clip/battery-holder/battery-retainer for CR2450 or CR2477 batteries instead of CR2032 batteries?

* CR2477 is 24 x 7.7 mm in size and usually has about 1000 mAh of capacity on average.
* CR2450 is 24 x 5.0 mm in size and usually has about 550 mAh of capacity on average.
* CR2032 is only 20 x 3.2 mm (coin cell size) but only has 200 mAh average capacity.

Alternative universal solder pads supporting fitting battery-clips/battery-holders/battery-retainers for either CR2477, CR2450, or CR2032?

Example:

MYOUNG MY-2477-02 (Mfr. Part # MY-2477-02 LCSC Part # C2979175)
https://lcsc.com/product-detail/Battery-Connectors_MYOUNG-MY-2477-02_C2979175.html

MYOUNG MY-2477-01 (Mfr. Part # MY-2477-01 LCSC Part # C2979173)
https://lcsc.com/product-detail/Battery-Connectors_MYOUNG-MY-2477-01_C2979173.html

MYOUNG MY-2450-01 (Mfr. Part # MY-2450-01 LCSC Part # C964845)
https://lcsc.com/product-detail/Battery-Connectors_MYOUNG-MY-2450-01_C964845.html

Reference:

https://jeelabs.org/2015/07/29/could-a-coin-cell-be-enough/index.html

rbaron, 2022-09-09:
I'm fairly happy with the CR2032. They're cheap, small, available everywhere and can be made to theoretically last a over couple of years.

You are of course free (under CC BY-SA) to redesign the board to better suit your needs.

closed 2022-09-09 by rbaron

Hedda, 2022-09-09:
OK, but FYI, CR2450 and CR2477 are also very inexpensive and commonly available almost everywhere today, so a 10-pack of either CR2450 or CR2477 costs less than $10 US-dollar now so please do not let price per battery play a major role in your decision, see example: 

https://www.google.com/search?q=CR2450&tbm=shop

https://www.google.com/search?q=CR2477&tbm=shop

https://www.amazon.com/s?k=CR2450

https://www.amazon.com/s?k=CR2477

https://www.amazon.de/s?k=CR2450

https://www.amazon.de/s?k=CR2477

nielsfaber, 2022-11-21:
I agree with the proposal of @Hedda 

CR2032 clip footprint (the one that is recommended):
[image]

CR2450 clip footprint (random one from aliexpress):
[image]

By increasing the spacing between the solder pads for the sides to approx. 25mm, and leaving some clearance next to the pads for the bigger clip, it would allow both versions to be soldered.

I would like to see this updated in the project.
Users can choose which version they want to make, only the case would be different (CR2450 battery is 24.5mm wide, so will be wider than the PCB). 

(I don't have+know KiCad otherwise I could submit this change myself)