rbaron/b-parasite #61: Advice/help to Build the Code firmware

Metus88, 2022-09-27:
I have a problem following the [Wiki](https://github.com/rbaron/b-parasite/wiki/How-to-Program#configure-build--flash-it) because I'm not usual to compile/build/make firmware... (my fault not of the wiki)
I'm using windows10 64bit with visual studio code and I can install all extension you want/need.

Following the wiki steps:

- [Download and Extract SDK ](https://github.com/rbaron/b-parasite/wiki/How-to-Program#get-the-nrf5-sdk-from-nordic) Easy but first question: When I extract the folder "DeviceDownload" inside there are all zip files SKD and s112nrf52720 s113nrf52720 s122nrf52800 s140nrf52720 ... I have to extract all of them or only the  "nRF5_SDK_17.1.0_ddde560" zip-folder? The command `make flash_softdevice` will expect a zip file or a simply folder?
- [Configuration](https://github.com/rbaron/b-parasite/wiki/How-to-Program#configuring) Easy editing files.
- [Build the Code](https://github.com/rbaron/b-parasite/wiki/How-to-Program#build-the-code) This is my big huge problem... I open with Visual studio code for windows the folder that I download from github `code/b-parasite`. I tried using windows terminal inside Visual studio code to set variable SDK_ROOT (on linux is `export SDK_ROOT=C:\Users\andre\Downloads\nRF5_SDK_17.1.0_ddde560`  ) but I found on google that on windows is `set `command `set SDK_ROOT=C:\Users\andre\Downloads\nRF5_SDK_17.1.0_ddde560` . But now I'm not able to do `make `command. The error is like "make is not a known command... command not found". (You can see below)
<details>
  <summary>Screenshot: sorry it is in italian</summary>
  [image]
</details>

Is it possible use make command on windows? which visual studio code's extension I have to use? (For example for build marlin firmware for 3d printer there is a specific extension stand alone.) On google there are lot of different methods to use `make` installing software that I don't know: MinGW or WSL or Chocolatey or Winget. Can I ask you which you use? There is a better way to Build the code on windows? Do you have link tutorials or guide? [this tutorial by SparkFun](https://learn.sparkfun.com/tutorials/arm-programming/all) will help me to flash the firmware with my new J-Link EDU Mini but for now I'm stuck... (It will be a second problem)
I have also raspberry and a linux pc but if it is possible I prefer to use windows (also because I download sdk for windows.)
Thanks

rbaron, 2022-09-27:
Hey @Metus88,

Unfortunately I have never build/flashed the firmware from a Windows computer, but I will try my best to give some pointers/ideas.

**Compiling the firmware**
I don't know what the nRF52 toolchain for Windows looks like, but I imagine our Makefile would have to be adjusted if you want to compile on Windows.
Compiling should be easier with WSL/MinGW (although I have never used it), Docker (see [example Dockerfile for b-parasite](https://github.com/rbaron/b-parasite/blob/main/.github/actions/build/Dockerfile) - I'm happy to elaborate if needed), macOS (what I use) or a Linux virtual machine. 

**Flashing the code**
To flash the SoftDevice (pre-build binary shipped with the SDK) and the compiled firmware, our Makefile calls [nrfjprog](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/Makefile#L314), a utility from Nordic. There is a [Windows version](https://www.nordicsemi.com/Products/Development-tools/nrf-command-line-tools). I would take a look at the `flash_softdevice` and `flash` rules and reproduce them with the Windows binary.

It should also be easier to use a Linux virtual machine, and you can just pass the JLink through from the host to the guest ([docs from SEGGER](https://wiki.segger.com/J-Link_on_VirtualBox), and just use our Makefile as specified in the docs.

drspangle, 2022-09-27:
@Metus88 I strongly recommend using WSL if you can. Doing this with a Windows toolchain will be painful and in a VM you might run into issues with USB support (rare, but it's happened to me in a way that results in unexpected behavior when flashing other devices that are similar to the b-parasite).

Metus88, 2022-09-27:
> @Metus88 I strongly recommend using WSL if you can. Doing this with a Windows toolchain will be painful and in a VM you might run into issues with USB support (rare, but it's happened to me in a way that results in unexpected behavior when flashing other devices that are similar to the b-parasite).

Thanks for your support. I decided to use linux (raspberry). Before I gave up with linux for a strange error but now thanks your dockerfile I understood to install gcc-arm-none-eabi. So I'm trying...

closed 2022-09-27 by Metus88