rbaron/b-parasite #63: Open source the rotisserie / drying jig

rbaron, 2022-09-29:
A while ago I built this extremely questionable spinning jig to try and get the coating layer to dry more uniformly:

[image]

I wrote about it in [a twitter thread](https://twitter.com/rbaron_/status/1491467468996431873) and to my surprise more people wanted to join in on the fun. Fair warning: don't take it too seriously.

closed 2022-09-29 by rbaron

jhbruhn, 2022-11-08:
How in-sync are the stl-files with the device on the picture you posted? I am trying to build a rotissierie but my moter seems to have a smaller radius than the one you used, could be AliExpress randomness-factor though. But it seems that in the picture, the case has an additional box between the cases body and the gears which could lead to a better fitting motor. Or is the motor held in place by more than just a friction fit?

Also, it seems that the .step file only contains the lid.

rbaron, 2022-11-08:
I think they are pretty much in sync. Did you check the [gear_motor.stl](https://github.com/rbaron/b-parasite/blob/main/experimental/rotisserie/gear_motor.stl) (which is different from [gear.stl](https://github.com/rbaron/b-parasite/blob/main/experimental/rotisserie/gear.stl))? The `gear_motor.stl` has a notch and is a tight fit with the motor axis.

I just measured my motor and the axis is 4mm, the same as [this on Aliexpress](https://www.aliexpress.com/item/32987942205.html).

jhbruhn, 2022-11-08:
Yes, I meant the body of the motor itself, not the motors axis.

rbaron, 2022-11-08:
[image]
[image]

Do these help?

jhbruhn, 2022-11-08:
Yes, thanks. Perhaps I am missing something, but is that connector between the motor and the other axes missing from the repo?

rbaron, 2022-11-08:
You're right! I just added it to the repo as `base.stl`.

gniezen, 2023-03-10:
I finally have all the parts and I'm in the process of building it, but have some questions:

- All the holes on `base.stl` are the same size. Should I drill out the two holes for the motor and its screw myself?
- There's no hole in the lid for the power supply jack. Should I drill this myself?
- There's no hole for the speed controller knob in the case. Should I drill this myself?

rbaron, 2023-03-11:
@gniezen As far as I can remember I think I drilled these holes myself. It's been a while 😰

gniezen, 2023-03-17:
It's working! :tada: 

[image]

PR #64 (2022-09-29): Open sources the rotisserie / spinning drying jig
Fixes #63
commit 0c7222d1 (2022-09-29): Open sources the rotisserie / spinning drying jig

Fixes #63
commit baec1c2a (2022-11-08): Adds missing base.stl to experimental/rotisserie

Fixes #63