rbaron/b-parasite #66: How to solder nRF52

Ulrar, 2022-10-12:
Hi,

I realize it's not really about the b parasite as such, but I've done what's suggested in the doc and ordered the board and controller separately. But I've never had to solder something that doesn't have pins before and after trying, it's insanely hard, I just made a huge mess.

Any tips, tricks or links to good tutorial on how to assemble this ?
Also is there a list of the "pins" (pads ?) actually used, looks like some of them can be ignored but which ones ?

Thanks !

rbaron, 2022-10-12:
Hey @Ulrar, nice to see you here. Thanks for all the work in ESPHome.

Here's some notes on how I solder the nRF52 module on my b-parasites.

## Method 1 - Soldering Iron + Flux
### Step 1. Cleaning
I use a cotton swab and isopropyl alcohol (IPA) both on the PCB's and on the module's pads.
### Step 2. Aligning
This is by far the most crucial step. To align the module with the footprint, I start with a small solder blob on a single pad:
[image]
Then I place the module and temporarily set it in place:

https://user-images.githubusercontent.com/1573409/195399660-aa318ff2-e647-4001-b22e-d1ae2c04f39d.mp4

I do the fine alignment by melting the single solder joint and gently moving the module around until the all the pads on all three sides are aligned. I take my time and repeat this as many times as needed, until the alignment is as good as possible:
[image]

**Important:** Be careful not to twist or lift the module when its being held by this single joint - any small force will rip out the pad from the PCB. That's why I use a non-connected pad for this step. If I accidentally ruin it, there's still hope. It hasn't happened so far.
### Step 3. Flux & Solder
I apply flux and briefly place the soldering iron + a tiny bit of solder on each pad, one side at a time:

https://user-images.githubusercontent.com/1573409/195400588-d95648d6-6361-481d-9242-e93d65403a01.mp4

I find that less is more. I use just a tiny bit of solder for each pad.

### Step 4. Cleanup
I again use a cotton swab and isopropyl alcohol to get rid of the flux.

### Step 5. Visual Inspection & Rework
After cleaning up the flux, I can spot two of the most common defects on each solder joint:
1. Gap between the module and the PCB pads. Likely happens if the pad is not heated up enough while soldering, so all the solder only sticks to the module. Placing the soldering iron (without extra solder) again for 1-2 seconds normally solves the issue
2. Bridges between two pads. If there's too much solder, I go in with a wick wire and get rid of it. Most of the time just reflowing the solder with the soldering iron is already enough to fix it 

## Method 2 - Hot Air Gun & Solder Paste
Steps 1 and 2 are the same as for Method 1. Clean and align with a single solder joint.
### Step 3. Applying Solder Paste
It's usual to apply the solder paste before placing the component that's being soldered, but since alignment is so important, I apply the solder paste after aligning, on top of the pads.
### Step 4. Blasting Hot Air
I set the hot air station to 360-400 °C and let it work its magic:

https://user-images.githubusercontent.com/1573409/195403067-20e43b71-866b-4962-900c-71886e8394bd.mp4

I then cleanup, visually inspect and rework bad joints.

Ulrar, 2022-10-13:
Hey @rbaron,

Thank you very much for the very detailed answer, as usual very much appreciate what you do !

I'll give method 1 a try today or tomorrow and hopefully that'll be that, but if not I'll order a heat gun and try method 2 next week.
I'll report back if and when I succeed, thanks again

jhbruhn, 2022-10-15:
Chipping in on this, I previously used method 1 to solder b-parasites, but soldering more than two or three, using method to simplified the process by orders of magnitude and lead to a significantly less frustrating experience, at least in my case. Good luck!

jhbruhn, 2022-10-19:
I also agree with that. I already thought about increasing their size for my next order to simply soldering, if there is the space.

Ulrar, 2022-10-24:
Hey,

Update on this, I finally got around to trying it out and it seems to have gone better than expected. First time using a heat gun so it did feel a bit like magic, although I probably need to use way more than I originally did.

I was able to then flash it on the first try, surprisingly, but HA isn't picking it up.
Not sure if it's something to do with bluetooth_proxy or if I've got something wrong somewhere else, I need to look into it but since the debugger is seeing the chip I think the soldering part worked out, at least for four pins !

agners, 2022-10-24:
> I was able to then flash it on the first try, surprisingly, but HA isn't picking it up.

@Ulrar if you want it to pick up automatically you need to use the BTHome protocol, afaik.

Ulrar, 2022-10-24:
I have changed the define to use it yes, I think there's some kind of issue before that.
The LED never turns on even though I've enabled it, and with all the debug defines enabled that's all it outputs : 
```
<info> app: Log inited.
<info> app: GPIO pins inited.
<info> app: GPIO pins inited.
Transfer rate: 0 Bytes/s Data written: 90 Bytes
```
Then it just stays stuck on that forever it seems, until I reset it by force. Looks like it might be crashing ?
I've lowered the deep sleep to 1 to confirm and it never seems to wake up again after. Not sure if I damaged something or if I'm using some weird unsupported version of one of the tools involved in flashing, maybe

oleo65, 2022-10-25:
Did you flash the softdevice?

I flashed a bthome one the other day and it got autodiscovered by Home Assistant almost instantly. I used a bluetooth proxy via esphome on an ESP32.

Ulrar, 2022-10-25:
Yes, I've flashed the soft device and the b-parasite code multiple times with no errors. The LED never flashed, and it consistently hangs on the log message above, never waking up until I reset it myself.

Not sure if that means there's an issue with the firmware / chip itself, or if a soldering issue could cause this. I'll have to try it again with a different board and chip

Ulrar, 2022-10-26:
That was it, placed the soldering iron on both for a couple of seconds and it's now working perfectly, spot on.

Once again thanks for your work and your help, very much appreciated!

closed 2022-10-26 by Ulrar