rbaron/b-parasite #67: How to program

kenny00111, 2022-10-19:
Hi,
Can anyone help me how to program them? I already assembled a few, but I have no idea how to compile the firmware. The readme not much of a help. I will use the E73 2g4m08s1e version chip, and I have Arduino ide.
Thanks in advance

oleo65, 2022-10-22:
I recently went down the road to compile and flash the parasites. There probably are many ways which lead to a successfully flashed device and depend on your knowledge of C toolchains, `make` and your debugging probe.

The various readme files and the wiki has basically all needed information but scattered around and in my experience sometimes hard to understand, if you are not quite familiar with the whole toolchain.

The whole process can be broken down into three main parts:

## Setup

You need four different components for successful building:

* The _legacy_ nRF5 SDK mentioned in the Wiki. I used the most current version `17.1.0` successfully.
* The S140 softdevice  as a prebuild `*.hex` file downloadable from the same link as above.
* An `ARM-GNU-Toolchain` for your platform, which basically consists of a `C` compiler, assembler, linker and all other tools to successfully cross-compile the source code for the ARM-Cortex M4 processor on the nRF52840 chip. I downloaded this directly from the ARM website and used the most current version `11.3.rel1`
* And of course, the source code in this repository.

## Compiling

I compiled the firmware in WSL on Windows 11 since I usually find the toolchains to be slightly easier to use in a linux environment. But you should be able to compile on any platform as long as you use the right `ARM Toolchain` for your platform.

The basic steps were as I recall from the docs and my personal error resolving:

* Set the SDK_ROOT environment variable to point to the nRF5 SDK folder as described in the wiki. In my case: 
```
export SDK_ROOT=/opt/nRF5_SDK_17.1.0_ddde560
```
* In the SDK, I edited the `components/toolchain/gcc/Makefile.posix` to match the downloaded `ARM toolchain`, which I extraced in the `/opt` folder. 
```
GNU_INSTALL_ROOT ?= /opt/arm-gnu-toolchain-11.3.rel1-x86_64-arm-none-eabi/bin/
GNU_VERSION ?= 11.3.1
GNU_PREFIX ?= arm-none-eabi
```
* Edit all other files for the b-parasite as described in the wiki, namely the `prst_config.h`
* Simply invoking `make` in the folder `code/b-parasite/` should now build the firmware, which can be flashed later.
* I had some compiler warnings but the firmware build successfully anyways.
* In your `_build` folder you will find the `*.hex` file for flashing.

## Flashing

The instructions will probably differ alot depending on your debugging probe and hardware setup. Long story short, I tried the whole `nRF command line` path with different variations and failed to get it running but ended up installing gigabytes of softwaretools. In the end I purged it all and went down an in my opinion much cleaner path and used a python based tool for flashing microcontrollers `pyocd`.

The setup is fairly straightforward and documented in the `pyocd` docs.

I used an inexpensive debugging probe from Aliexpress called nanoDAP from Muse Labs which is less than 10 € and readily available. The J-LINKs are hard to get and quite expensive. In my understanding basically all `CMSIS-DAP`-compatible probes which are recognized by `pyocd` should work.

After connecting my debugging probe via usb and connecting the b-parasite as described in the wiki, flashing was easy and fast.

First flash the softdevice from Nordic. 
```
pyocd load .\sdk\s140nrf52720\s140_nrf52_7.2.0_softdevice.hex
```
Then flash the previously build firmware: 
```
pyocd load .\code\b-parasite\_build\nrf52840_xxaa.hex
```

That's it. :wink:

kenny00111, 2022-10-25:
Hello, thanks for the detailed answer. I actually find an other way to compile and program before I read your comment.

I will share maybe it will be a help for someone. I used Win 10 for the entire process

- Step 1:
Download & install MinGW-arm-eabi
- Step 2:
Download & install GNU Arm Embedded Toolchain: 9-2020-q2-update
- Step3: Add to the system enviroment variables: 
  - MinGW-arm-eabi-11.1.0\bin
  - SDK_ROOT
  - Platform
- Step4:  Go to the directory where the parasite makefile is, then open the windows powershell and type make. now it should compile the firmware without any issue.
- Step5: Download & install nrf connect for programming, it is quite straightforward to use.

Side note: In the readme it is mentioned that you either use the programmer power source or the battery. However if you use J-link EDU mini it can't power the board and you will have to use battery while the VDD is connected ( Vref for the programmer)

kenny00111, 2022-10-25:
Yes I have almost forgotten:
Does anyone has a problem with the esp32 ble tracker? For me it can handle max 3/4 parasites before the esp stops transmitting the mqtt messages.

closed 2022-10-25 by kenny00111

letshin, 2022-11-04:
@oleo65 Hi - I tried your suggestion of using pyocd and am getting this error. Do you know what this might be please

"
```
pyocd load s140_nrf52_7.2.0_softdevice.hex --target nrf52833
0001139 C STLink error (9): Get IDCODE error [__main__]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/subcommands/load_cmd.py", line 96, in invoke
    with session:
  File "/usr/local/lib/python3.8/dist-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/core/soc_target.py", line 147, in init
    seq.invoke()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/coresight/dap.py", line 439, in _connect
    probe_conn.connect(self._protocol)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/coresight/dap.py", line 172, in connect
    self._connect_probe(protocol)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/coresight/dap.py", line 189, in _connect_probe
    self._probe.connect(protocol)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/probe/stlink_probe.py", line 159, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/probe/stlink/stlink.py", line 327, in enter_debug
    self._check_status(response)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/probe/stlink/stlink.py", line 365, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error
```

"

oleo65, 2022-11-04:
There seems to be some kind of issue with your debugging probe or wiring. Since I am not an expert there I don't have a solution for you. 

The issue to me does not seem to be related to the b-parasite in general. 🤷

letshin, 2022-11-04:
No problem. Thanks for replying. I'm getting a CMSIS-DAP to see if this works instead of the Jtag.