rbaron/b-parasite #70: Main branch advertising while in deep sleep

1technophile, 2022-11-11:
Hello,

Thanks for this project. Your PCB and firmware worked well after the soldering of the nRF52 and the flashing step! The most difficult part was the flashing for me.

Once done got directly the data from [OpenMQTTGateway](https://docs.openmqttgateway.com/) and [TheengsGateway](https://gateway.theengs.io/):
```json
{"name": "prst", "id": "F0:CA:F0:CA:01:01", "rssi": -78, "brand": "rbaron", "model": "b-parasite", "model_id": "BPv1.0-1.2", "tempc": 24.828, "tempf": 76.6904, "hum": 58.59311818, "moi": 3.578240635, "lux": 20, "volt": 3.301}
```
Also, the auto-discovery created the sensor automatically into HA, great !

I flashed first your main branch with https://github.com/rbaron/b-parasite/commit/baec1c2a808d42872d471069482ab190abd21729 and was surprised to see an advertisement coming to my gateways every 5 minutes only.
After some investigations, I found https://github.com/rbaron/b-parasite/tree/sleep-adv , flashed it, and was happy to see that with this branch, the sensor also advertises when in deep sleep. Making it easily discoverable by non-continuous scanners.
With this, I can get the data also in the [Theengs App](https://app.theengs.io/) 🎉

Where I need your help is for understanding if I'm missing something. The `sleep-adv` branch has been merged here https://github.com/rbaron/b-parasite/pull/49 but doesn't seem to be the default behavior.
I like it better because it is a similar experience compared to the other BLE sensors. Am I missing something to have advertisements when sleeping with the main branch ?

rbaron, 2022-11-11:
Happy to hear things worked.

There are [two states](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/src/main.c#L21) the app can be in:
1. Advertising [while in "sleep"]: it means the application is not running, but the radio is on, broadcasting adv packets every [30 ms](PRST_BLE_ADV_INTERVAL_IN_MS). The application is in this state for [PRST_BLE_ADV_TIME_IN_S](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/config/prst_config.h#L53) seconds (defaults 1 s) and then goes to state 2.
2. Full sleep, when even the radio is turned off. It lasts for [PRST_DEEP_SLEEP_IN_SECONDS](https://github.com/rbaron/b-parasite/blob/main/code/b-parasite/config/prst_config.h#L19) (5 minutes) before going into state 1.

1technophile, 2022-11-25:
Thanks for this information. I got the advertisements when in "sleep" with the main branch.

If we wanted to have a behavior like the Mi Flora or other environmental sensor in terms of BLE advertisement with a battery that last months (or years :-)), what do you think could be the adequate parameters?
An example could be an advertisement in sleep every 30ms, a measurement of the sensors every 10 minutes, but no deep sleep as this removes the usage of broadcasted data with non-continuous scanners.

For instance, the Theengs App scans when you request it. This works great with other sensors but with the default parameters of the b-parasite, we are missing the advertisements if we keep the default parameters.

rbaron, 2022-11-25:
That's hard to say. It's all tradeoffs between power, range, discoverability.  
For continuous advertisement, you can try lowering the advertisement interval and tx power. Maybe do "continuous" for a couple of minutes and then sleep for some. 

My mental model is that continuous advertising makes sense when the device is connectable and lets humans interact with it. b-parasite  is much simpler, and just broadcasts some info once in a while to be passively picked up by automated scanners. This simplicity lets us get away with aggressive sleeping and much longer battery life. I find it hard to justify having the radio on just to transmit values that don't really change from one minute to the other.

1technophile, 2022-11-26:
I get your point, it makes sense as a default behavior for automated scanners.
I will play with the parameters and see what can be done to have it picked up easily by the app, while keeping power consumption low.

closed 2022-11-26 by rbaron