rbaron/b-parasite #73: Wall moisture capacitance measurement?

evlo, 2022-11-14:
This is not an issue, but I wonder if I place the sensor on the wall, could it measure wall or at least wall surface moisture content?

I know there multe wall humidity / moisture measurement method and devices, the basic ones are 
- resistance - with two prongs that is pretty much useless
- **capacitance** - with ball at the end 
- microwave etc. I don't care really - super expensive devices, probably could be done somehow by mmwave sensors diy, but would require lot of research

So when/if my test batch of assembled pcs arrive and i place them on the wall in different heights is it possible that they would give some reliable moisture measurement even if just one side is in contact with the wall?

Even the air RH seems to be more accurate than the two prong meters.

_At first I wanted to order these sensors from JLPCB as the first made to HW experience, as the commercial sensing smart home solutions are unreliable and only ones working, at least for me, are hacked xiaomi BT TH sensors. But now I may have real use for this as I want to get rid of the moisture in walls of the 110+ years old house I'm attempting to fix._

letshin, 2023-01-10:
I don't want to be a wet blanket but would it not be more cost-efficient to just get a couple of these:

https://www.amazon.co.uk/gp/product/B08DLHFKT3/

I presume you have Home Assistant already - HA detects these easily and readily.

evlo, 2023-01-10:
No, that measiures air RH And i already have about 20 od those :)

letshin, 2023-01-10:
Ah, you want to directly measure the wall humidity? But water droplets/condensate would totally saturate the capacitance right? Why not just get a small computer fan and combine them with the mini sensors and place as close as possible to the wall? That should give better empirical data.

evlo, 2023-01-13:
Ok O found some capacitance transceivers on ali, so seems easier to try that way - run to bolts into the wall and measure capacitance, hopefully it will give some reasonable output.

closed 2023-02-12 by rbaron