rbaron/b-parasite #74: 4 layer PCB version as alternative for coating

nielsfaber, 2022-11-21:
I'm very interested in this project.
I have a couple of MiFlora's, but they have not proven to be reliable for me, so I'd like to try a different approach.

I was wondering if any considerations or attempts have been made to use a 4 layer PCB instead.
This would allow the copper pads for moisture measurement to be placed on the inner layers of the PCB, sandwiched between (bare) FR4.
To me it seems more elegant and reliable option compared to using a coating spray, by mitigating the risk for corrosion.
Of course the PCB will become a bit more expensive, but I think the benefits outweigh this.

One concern I have is whether the measurement would be impacted by a change in the PCB layout.
The capacitive coupling to the soil would become slightly different.
Would there be any way to (re)calibrate such PCB layout to obtain the same performance as the original design?

Note: a similar approach can be found here: https://github.com/diyruz/flower

nielsfaber, 2022-12-03:
I decided to proceed by making a 4 layer PCB layout myself.
Summary of my changes:
* Added 2 inner layers, inner 1 is almost completely empty (only few traces) to limit parasitic coupling of the traces+pads on the top layer for the capacitive sensing. Inner 2 has ground plane + sensing probe
* Increased the footprint for the battery holder to be compatible with both CR2450 and CR2032 (see https://github.com/rbaron/b-parasite/issues/55)
* Added a push button on a GPIO (for zigbee pairing?)
* Made changes to the footprint of the nRF chip to remove conflicts with bottom pads (see https://github.com/rbaron/b-parasite/issues/75) and easier manual soldering
* Had to shift (and shuffle the order of) the testpads for programming a bit to simplify routing
* Applied some rounded edges on the bottom side of the board
 
Here's a kicad render of the top and bottom sides:
[image]

I just ordered the board at JLC. 
They are able to assemble all parts except the nRF chip (no stock) + battery holder.
Note that I had to do a little trick: JLC was initially charging almost 20 euros extra since the board doesn't fit in a 100x100mm area.
So I rotated the board 45 degrees in Kicad before generating gerber files such that it fits diagonally within a 100x100mm square, worked like a charm.

I will post back here when I have done some testing with the boards.
Will definitely have to re-perform the calibration.
Some changes will be needed for the 3d print case as well.

In case someone else wants to build this, I created a fork with my changes in: https://github.com/nielsfaber/b-parasite
If @rbaron would be open for it, I can eventually make a PR as well. But I understand the changes are quite significant.
Big disclaimer: this is my first experience with Kicad before this project, so I might have screwed things up.

rbaron, 2022-12-03:
Hey @nielsfaber tks for sharing! I am happy link your variant in the wiki, but I would suggest we keep your fork separate for now. Otherwise it gets tricky for us to maintain all of them with different calibrations / peripherals (we also have a mini variant in #69 for example).

A few thoughts:
- I noticed that the sides of the uncoated board tend to be specially vulnerable. They are exposed, since no solder mask nor copper, and water creeps in between the layers faster than from the front. I would still suggest coating the boards.
- Shifting the programming pads unfortunately causes the clamp to interfere with the battery clip on the bottom (if you use a clamp)
- I would suggest changing the 1.2.0 version on the silkscreen layer to something specific to your fork.
- I think the new battery clip might require redesigning the cases, worth checking (I highly recommend using one!).

I've started working on a v2.0.0 hardware revision with some improvements (revised capacitance sensing circuitry, a couple of buttons, better footprints for hand soldering the RF module). I also fixed the bottom footprint issue you raised in #75. It's a WIP in the `hardware-v2` branch if you want to take a look.

Let us know how it goes!

nielsfaber, 2022-12-03:
@rbaron 

> I am happy link your variant in the wiki, but I would suggest we keep your fork separate for now. Otherwise it gets tricky for us to maintain all of them with different calibrations / peripherals 

I agree. Perhaps (in time) it would be nice to add some sort of table with all the flavours such that users can choose.
As I don't have my version working yet I think it's too soon to recommend it to anyone (it might need a redesign).

> I noticed that the sides of the uncoated board tend to be specially vulnerable. They are exposed, since no solder mask nor copper, and water creeps in between the layers faster than from the front. I would still suggest coating the boards.

I work in a company that manufactures electronics boards, we apply conformal coating to boards which need IP67+ rating, but the sides are always left uncoated.  I have never heard about cases where moisture creeps in between the layers. 
But then again, these boards are not sticked in the soil.
I'm not planning to do the coating, so I guess time will tell 🙂 
Note that for now my use-cases are only interior plant pots.

> Shifting the programming pads unfortunately causes the clamp to interfere with the battery clip on the bottom (if you use a clamp)

Good point, I didn't consider this at all.
I did buy this clamp for the project: https://nl.aliexpress.com/item/1005004258921308.html
It doesn't seem to allow opening more than ~0.5cm so I guess this will not fit.
For now I see some workarounds:
* Just do the programming step before soldering the battery clip (hopefully we can do OTA updates afterwards)
* Make a modified 3d printed bottom part for this clamp which allows wider opening
* Solder wires to the pads (last resort)

> I would suggest changing the 1.2.0 version on the silkscreen layer to something specific to your fork.

Fair point.

> I think the new battery clip might require redesigning the cases, worth checking (I highly recommend using one!).

Definitely true, I was planning to make an adapted version (also for reaching the button) when the boards are here.

> I've started working on a v2.0.0 hardware revision with some improvements (revised capacitance sensing circuitry, a couple of buttons, better footprints for hand soldering the RF module). I also [...]

I will take a look into the revised sensing circuit, thanks for the heads-up.
Good to hear you're covering the other adaptations as well.

I was wondering whether the Cpara cap needs to be assembled or not (in the current HW revision). Any thoughts on this?

rbaron, 2022-12-03:
> I work in a company that manufactures electronics boards, we apply conformal coating to boards
That's cool! Can I ask you what kind of coating do you use? I only have real experience with acrylic-based ones, but I think there's better stuff out there.

> I was wondering whether the Cpara cap needs to be assembled or not
No, it does not. I put the Cpara para there to make probing with the scope easier, but the pads are just connected to the parasitic capacitor in the PCB. It's not a very good idea, and at least I should have marked it as "do not populate" in KiCad. In `hardware-v2` I got rid of it altogether in favor of real test pads.

nielsfaber, 2022-12-03:
> That's cool! Can I ask you what kind of coating do you use? I only have real experience with acrylic-based ones, but I think there's better stuff out there.

One of the coatings we use is Peters Elpeguard SL 1307 FLZ ([link](https://www.gluespec.com/Materials/SpecSheet/a3d71ce6-56ee-4df1-9855-fc123843c855)). 
I'm not sure whether this is available or suitable for hobbyists. It is applied automatically by means of dispensing + UV curing machines.
As far as I can see this material is also acrylic. I don't have personal experience (I rather make sure the electronics is in a sealed enclosure instead), but I can tell you the smell is not pleasant.

> No, it does not. I put the Cpara para there to make probing with the scope easier, but the pads are just connected to the parasitic capacitor in the PCB. It's not a very good idea, and at least I [...]

Good to know! It was not that evident to me, especially since it was linked with a LCSC part and indeed not marked as not assembled part. Will just remove it when the boards arrive, no problem.

Steiniken, 2023-01-03:
Hi together,
first of all thank you @rbaron for initiating and pushing this great project forward!

@nielsfaber 
Your idea seems very promising to me, and I would like to build my first sensors using your layout. 
I know it has only been ~1 month since you posted your idea, but perhaps you already have some experience with your sensors? If so, I would really appreciate it if you could share your experience here.

Thanks! 
Bernd

nielsfaber, 2023-01-05:
@Steiniken 
I have received the PCBs (all looks fine) and the parts to solder manually, as well as the programmer and probe clamp.
The next steps:
1) Complete the soldering
2) Firmware programming *
3) Capacitive measurement calibration (+re-programming)
4) design+3D print enclosure
5) Use it

\* I still have to decide whether to use the BLE or zigbee firmware version.

I haven't found any time yet for any of these.
I am hoping that I have some free saturday somewhere this month to at least reach step 3.

Will follow-up when there is progress on this (I'm also planning to update my forked repo with the changes).

Steiniken, 2023-01-06:
Hi @nielsfaber 
thanks for your update!
Will be very interesting to see what effect this extra layer of FR4 will have and if it's possible to compensate it with calibration.
Looking forward to your next update!

nielsfaber, 2023-02-22:
@Steiniken I put the project on hold since I couldn't get the debugging / programming working.
As I don't have a JLink debugger I was using the [BluePill method](https://github.com/rbaron/b-parasite/issues/43#issuecomment-1336617818) instead.
In gdb I kept getting errors that the target voltage was too low (around 1.5V), although I could measure a stable 3.0V at the supply pin with a multimeter.
Since I don't know how to circumvent this (wiring or PCB doesn't seem the issue), I got stuck here.

Steiniken, 2023-03-05:
Hm, do you think a J-Link helps with troubleshooting?
I'll hopefully get one soon and can lend it to you if you want.
Just let me know.

seekriver, 2023-04-16:
According to@nielsfaber 
>  the sides are always left uncoated. I have never heard about cases where moisture creeps in between the layers.

But judging from the picture of the oxidation situation in the wiki, the oxidation starts from the edge.
Maybe it's because the copper clad area is too close to the edge. When the manufacturer was producing, he accidentally touched the edge of the copper clad copper when cutting the shape, causing the copper to be exposed to the air.
You can take a look at another project https://github.com/diyruz/flower to adjust the copper shape of the sensor to avoid copper exposure to the air during production.