rbaron/b-parasite #75: Clarification about E73 module PCB footprint

nielsfaber, 2022-11-23:
When opening the PCB file of this project in Kicad some conflicts show up regarding the pads underneath the E73-2G4M08S1C module:
[image]

It seems the inner pins (shown in gold) are not used and left unconnected in this design.
However they should be avoided by other traces as this creates short circuits.

I'm using the footprint from the library provided in https://github.com/rbaron/b-parasite/tree/main/kicad/lib/nrfmicro

Could you please comment on the conflicts? 
Are the pins deliberately/manually deleted from the footprint assuming the pads will not touch the PCB (or solder mask provides sufficient isolation?
Thanks!

rbaron, 2022-11-23:
Good observation. In practice I think the solder mask should provide enough insulation. In the worst case, if that fails, these bottom pins (except 24 which is ground) are not used and left in a high impedance/disconnected state, which I _think_ should also be okay. I removed the bottom pads from the footprint to make routing on a 2-layer board feasible - the big battery pad on the back copper layer makes it extra tricky).

That being said, this is extremely inelegant at best and there's room for improvement, specially for the vias placement.

nielsfaber, 2022-11-24:
Thanks for the explanation. 
I understand your design choices, it's in line with what I expected, and I can live with them 🙂

closed 2022-11-24 by nielsfaber