rbaron/b-parasite #78: Bluetooth module footprint

BeckmaR, 2022-12-05:
I found the footprint of the module to be quite small. Most of the pads is covered by the module on top, and I was unable to solder it down by hand. The only solution was to use solder paste and a hot plate we have in the office. Is that the recommended way for assembly? I think it would be nice if the pads were enlarged 1-2 mm to the outside. IMO there is no reason not to do it...

rbaron, 2022-12-05:
I've started working on a new hardware review and I updated the footprint there. I will merge it once it's ready. It's on a branch 'hardware-v2' for now.

closed 2022-12-05 by rbaron

mill1000, 2023-01-30:
If I was going to order some b-parasites soon could I use the v2 design or should I stick with the current main?

rbaron, 2023-01-31:
@mill1000 I have recently put the first order of v2 boards for testing and developing, but I don't know if there are some hardware bugs yet. They still need software support, recalibrating and testing, since the sensing circuit changed. Unfortunately I can't give you an exact timeframe.

mill1000, 2023-01-31:
Ok thanks for the update. I will hold off on an order for now.