rbaron/b-parasite #79: Conformal Coating - Call for Contributors

rbaron, 2022-12-06:
I started collecting information about options of conformal coating into the new [wiki page](https://github.com/rbaron/b-parasite/wiki/Conformal-Coating). It's very minimal but hopefully we can make it richer soon.

I added the brand of acrylic coating I use, and I would love if more people shared what worked for them. Please consider sharing your successes and failures on this issue and I'll compile them into the wiki.

Tks!

drspangle, 2022-12-06:
I'm currently experimenting with MG-branded urethane conformal coating (4223), which I got off [Amazon here](https://www.amazon.com/gp/product/B008OA7AWE/) but seemingly is no longer available. It's a bit thin and doesn't seem to be designed to withstand the rigors of gritty soil and plant material rubbing up against it. It's optimized for spreadability and waterproofing. It doesn't work all that well to protect the sides of the PCB.

I think a more viscous coating would be better, and I'd be very curious to know if anyone has tried using UV resin for SLA 3d printing -- I imagine that "tough resin" could be built up in layers by dipping the PCB and curing the dipped area. This would result in a long-lasting and very durable conformal coating that you could add more layers to as needed for the environment you plan to put the b-parasite in.

JoelWise, 2022-12-16:
I've tried MG Chemicals - 419D acrylic conformal coating a few times. My first try was a single brush on coat using the included brushes which was a mess and the smell is unbearable. The coating on 9 of the 10 sensors eventually failed and I peeled them off to try again. My second try was three coat dip following the datasheet's instructions. The results looked great but after months of use they all show bubbled de-lamination with only one complete failure. There isn't much sign of pcb corrosion but I'm not confident the coating is truly waterproof and instead is just slowing down water destruction. I wouldn't be surprised if the probes go dead before the batteries.

This is a photo of the triple acrylic dipped probes about 6 months of use:
[image]

letshin, 2023-01-10:
I've tried nail varnish - nope doesn't work well, and have also MG Chemicals 419D-55ML Premium Acrylic Conformal Coating. Sounds the same as @JoelWise's. Brushed thrice instead of dipping. I found that with the 419D, it was a hit and miss with the sensing. On watering the potted plants the sensor would frequently saturate, and in many cases I would need to manually 'reset' the sensor by reseating it in the soil. I think that there is a layer of water that sticks to the soil/sensor for this one. I would need to run more tests to check this hypothesis of course but haven't had the time recently to do so.

chucknorris101, 2023-01-16:
So far I have tried a few. a consistent trend id recommend with any coating approach is allowing a long cure time - a given coating is 'dry' in 4 hours, but may take months to fully polymerize/waterproof.

Varathane Oil Poly - single vertical dip then dry horizontal, LONG air cure ~3 weeks - gets a decent amount of resistance, but seeing some pitting already after a month or so - 75/25 on still good vs 'leaking' after a month or so - will see if this is a 'crib death' type deal and those 75% will continue to be good. Also looking at doing double dip after a 1-2 week cure. Short cure is pretty bad pretty quick. A double dip i think will be enough to make it rather indestructible but again - long lead time.

Varathane triple thick water poly - epic fail, though short cure.

some brand - Spar Varnish - about the same as Oil Poly - though have not tried longer cure here. theoretically might be better since its designed for marine/salt water applications

Flexseal rubber - seemingly impervious, but way too thick with a simple in-out dip to make a useful sensor. May try to thin it out a bit when applying. I use this now on the top half of sensors before burying them and the top halves always come back up flawless when pulling back up

PaulSzymanski, 2023-01-26:
"Gelcoat" used to paint fibreglass boat hulls might be worth a try.

dlavey, 2023-02-08:
If my understanding is correct, it's really the edges of the PCB that need the protection to prevent the water infiltrating the board, right? I picked up some UV light curing solder mask ink and I'm going to coat the edges of the PCB with that and see how it goes.

rbaron, 2023-02-09:
@dlavey if you have the option, I would be on the safe side and coat the whole part that touches the ground. I've seen the solder mask peeling off of uncoated boards in the past.

chucknorris101, 2023-02-09:
Even in some ive coated with various poly/rubbers, the water can and will
seep through around the edges of the coating if there isnt something
covering the first coating/sealing the edges contacting the board

letshin, 2023-03-10:
Has anyone tried rubber heat shrink. Can the ends be sealed off with higher heat and would these be too thick. How thick can the coating afford to be before it negatively affects readings?

dlavey, 2023-03-10:
To me, that seems like it'd be too thick and you'd have to find some way to seal the end, heat shrink isn't designed to close down to zero, regardless of the heat you apply. If you happen to have some heat shrink, you can always give it a try, test the probe with nothing on it, put it in some water and see what it reads, then dry it off and let it go back to whatever it's value was before you put it in water. Then put the heat shrink on it, I'd seal up the end with some sort of tape just for the test, put it back in water and see what it reads. I'm sure there are more scientific ways, but this is just what comes to my mind.

chucknorris101, 2023-03-23:
An update - I have had some good success with coating the 'distal' pointy end in epoxy - ive been using jbweld marine epoxy, not the cleanest or most consistent application process yet, but as long as its entirely encasing the soil/moisture contact area its been great. The epoxy is pretty well inert so far and no water ingress. I have a bigger batch of epoxy thats non-marine/more 'crafty' style to try to reduce cost, still tbd on this durability, but i dont expect it to be much different. I have been using a 'custom' calibration process with this for each sensor for each plant depending on the soil mixture but not sure if this is entirely necessary, but a different than standard calibration I would recommend as the coating causes it to read much lower than standard (as expected)

My application is burying the sensor - so im also coating over the proximal/nrf end, and in order to allow battery changes/re-flashing, ive been using a liquid rubber (flexseal). Ive tried the canned version/liquid and the spray can version. as these need much less time to cure than poly and are theoretically more waterproof/less porous

In my experience the spray can version has caused issues due to the nature of the thinning agents for the rubber - it is so thin it can interfere with the battery contact. The spray also does not deposit much of a layer - much less durable without more coats. The spray also is messy as hell. **do not recommend.** (for proximal end, or for the distal end). 

The canned liquid though, as I think Ive mentioned earlier, is nigh impervious, like the epoxy. The drawback with it for the distal end is the thickness preventing measurements, but on the proximal end thats not an issue. Its much easier to cut through than epoxy for reflashing or battery changeouts. It also doesnt seem to have the same issue as the spray version seeping under the battery.

I do still have some oil-poly + cured sensors running well as well, so its not a poor solution either as long as youre willing to wait for the cure or better yet apply multiple coats (and cure).

I have used the heat shrink approach for the proximal ends on some, but quickly abandoned it due to the issues listed above - the open edge was hard to re-seal with glue or rubber, etc and removal/modification of it was a PITA

my current challenges are surrounding the coating process steps and initial calibration. previous iterations did not impact the calibration too much, so was able to coat the bottom then the top after calibration/battery loading, the last few ive done have had issues after coating the proximal end, which may be due to the epoxy not coming up far enough. this may be due to the mini-version im using, the battery is much closer to the parasitic loop, i may look at moving the battery farther up for that fork (and over a bit to prevent flashing pin interference with the standard battery holder)

1technophile, 2023-03-23:
Could you share some photo of how does it look like with the canned liquid?

chucknorris101, 2023-03-23:
You can see it on the nrf side of this one - note I very specifically covered the flashing pins during coating so that they would remain exposed / i plan to re dip or otherwise coat that section once flashed

[image]
[image]

chucknorris101, 2023-04-06:
Having just gotten my first 3d printer, im now wondering if there isnt a good way to have the 3d printer make a coating for us, im not super familiar with the gcode side, but at least for the part of the parasite that is flat before attaching the battery holder, could have a print where the printer lays down the bottom side of the coating along with a small support square to support the nrf end to keep it flat....add a pause to insert the parasite board, then continue the extrusion over the top. maybe there would be issues with water tightness or adhesion to the board? though maybe another through hole or utilizing the 2 drilled holes would help 'secure it'.

Based on the thickness of the epoxy coating ive been using, a PETG or even  PLA coating over the board shouldnt affect capability too much and should be much more consistent across boards

dlavey, 2023-04-06:
> Having just gotten my first 3d printer, im now wondering if there isnt a good way to have the 3d printer make a coating for us, im not super familiar with the gcode side, but at least for the part of the parasite that is flat before attaching the battery holder, could have a print where the printer lays down the bottom side of the coating along with a small support square to support the nrf end to keep it flat....add a pause to insert the parasite board, then continue the extrusion over the top. maybe there would be issues with water tightness or adhesion to the board? though maybe another through hole or utilizing the 2 drilled holes would help 'secure it'.
> 
> Based on the thickness of the epoxy coating ive been using, a PETG or even PLA coating over the board shouldnt affect capability too much and should be much more consistent across boards
> 
> edit: now that I think about it, PLA might swell over time which would defeat the purpose/exposure moisture internally

I think this would only end in sadness. You'd have to either print something super thin that's tight fitting to the PCB which would be difficult, potentially doable with TPE, but I've found TPE to be a PITA to work with. The only other option would be to print directly on the PCB. I think I'd end up pulling out of my hair out before I could get anything printed on there, the shape isn't conducive to laying flat on the bed and I'm not sure the PLA would even adhere to it. Plus printing on the edges, which I believe is where the real moisture problem is would be next to impossible.

chucknorris101, 2023-04-06:
> I think this would only end in sadness. You'd have to either print something super thin that's tight fitting to the PCB which would be difficult, potentially doable with TPE, but I've found TPE to [...]

I think thats fair, and maybe this is more challenging than expected, but maybe im not elaborating as well as I could...in my mind the print would start with a little rowboat of PETG/PP or other waterproof filament to go up over the edges and missing the boat stern where the board would sit, and with the model having a hole in it that would otherwise be empty/support. I dont think it needs to be -that- thin to maintain device capability, at least from what ive seen on layer heights. its probably best maintained at only 1 layer, but with some overextrusion, maybe it coalesces/is waterproof? 

In trials with my flexseal liquid tub, the thickness of the coating is at least as thick as whats in the picture above over the resistors and the like, the sensor still could 'measure' in a sense but it was highly restricted and there was maybe a 10% reading difference between air and water. I think youd be able to get 2x 0.2mm layers maybe 3 on each side before you have an impact to usability in a practical sense (maybe you wont be able to track down to the nanoliter % change anymore...). If you can get lower layer heights even better.

I think it could be tight enough on the sides to friction fit (or interference fit even better?) or be close enough, but yea specific adhesion to the board, idk, maybe a throughhole so its really melting to the printed material itself over the top. you really only need to specifically 'seal' the board edges at the proximal/nrf end of the board with this type of solution, you could make a wider 'coating' across the sensor..like wings... to get more self adhesion on either side of the print. Maybe you still would need a rubber type coating over the top end of the print interface, but 

Definitely doesnt seem like a simple case, but the 'overmolding' type aspect doesnt seem like it should be that complicated with the right model design.  

Spitballing now you could maybe even make a few at once with a vertical print, maybe not the most structurally sound, but if you made 2-3 sensors at once, you could just make it almost like a popsicle tray and insert the parasite after instead of needing it to pause. This would have overall less adhesion chance/interface security on its own and require a top edge rubber seal, but maybe overall more feasible?

foreverimagining, 2023-04-06:
> I think thats fair, and maybe this is more challenging than expected, but maybe im not elaborating as well as I could...in my mind the print would start with a little rowboat of PETG/PP or other [...]

I don't want to say that I think this idea is impossible, but I do think that it would be the biggest royal pain to work out and has a lot of potential pitfalls. Sorry, this will be long.

First, it would probably have to be a single-piece print. Any time you have a multi-part print, you have a seam that will at some point let water in unless otherwise sealed. Because of the layered nature of 3D prints, they will never have a completely smooth wall and the two parts will never meet perfectly. The only way I could see doing a multi-part print is if you glued them together with something like 3D Gloop, which is a solvent and would actually meld the parts together. Testing would need to be done to make sure that it wouldn't mess up the PCBs, but now that I've mentioned it, I am actually rather curious.

FDM 3D prints can hold water with as little as 1 layer/wall thickness. I like making things in vase mode when possible, so I know that it can work. I've been printing planters/seed pots in vase mode for the past week and have a couple of watering bulbs that I made previously, only one of which had a leak to seal. The problem is going to be that not every print is perfect and you may end up with unexpected/microscopic holes that would have to be otherwise sealed. For the bottom layer, it's going to be a matter of bed adhesion (dial in your z-offset and squish that first layer) and possibly a little over extrusion if you find it's still holey. One layer may or may not be enough; two should be plenty with the right settings and no unexpected surprises. The walls are going to be more of a problem because it only takes a small slip up on layer adhesion to create a leak. Over extrusion here may also help, but you might then run into the issue of internal dimensions where you would place the PCB. I also don't know how well thin wall prints hold up to keeping water out over time, so it may not be enough in the long run and more walls may be better. 

As for the actual printing technique, it is totally possible and often very interesting to print on other materials or to encapsulate an object. I'm thinking about printing on fabric and encapsulating NFC tags, specifically. In this case it would be a matter of printing a frame first that would support and hold the position of the PCB flat in the correct spot in reference to the shell being printed (movement would be bad). Then it would simply require a pause at the right layer of the print (after the one below the first layer of the upper wall) that would allow you to position the frame over the print and insert the PCB in the middle acting in place of the printed supports you would normally have to have. Then extrude some filament to prime the nozzle and continue printing. A potential pitfall here, though, would be the thickness of any components already installed on the PCB because they would either require a greater distance from the build plate or they could be hit by and interfere with the nozzle depending on which side they're facing. Some of that risk can be mitigated by adjusting travel settings in your slicer and reviewing the G-Code for risky movements, but Murphy will get us all eventually.

For adhesion between the filament and the PCB, I expect that it would be a bit mixed. The bottom layers and the side walls would be cooled by the time you reach the pause to insert the PCB, so they would not adhere at all. The first layer that goes on top of it might adhere because it may melt onto the PCB or melt the material of the PCB slightly. I don't know that I expect that adhesion to hold, though, because once the filament cools, it will contract slightly and very likely separate. So my expectation is that water could get in from the top if you aren't careful and don't seal that seam, but the print may not slip off so long as it is well fitted. I do wonder if the slight air gap would have an affect on readings, though.

I think the vertical printing idea would also have some problems, unfortunately, some of which could be worked out with effort and experimentation. You could take the time to dial in the model measurements and print settings to have it fit the PCB, but I foresee issues with inserting the PCB without a path for the air to escape. It may be that the gaps created by the layer effect would be enough of a passage, though, as they are in the right direction. You would also have to be careful of drooping because long, thin vertical walls sometimes don't support themselves very well as they get taller and start to lean slightly creating a warp. But the wall thickness would be determined by the nozzle thickness this way rather than the layer height, so it also might be harder to get thin enough walls to avoid too much distortion to the measurements. Why can't 3D printing be easy? Still, would be easier as long as they can actually slide on.

I would also be concerned about the potential for heat damage from prolonged exposure to high temperatures coming from the print bed, extruded filament, and proximity to the nozzle. It may not transfer enough heat or for long enough to cause problems, but it could be something to take note of.

So, like I said at the beginning, possibly not impossible. Do I want to be the one to work it out? No. Do I think it might be better than resinous coatings? Ehh, I can't say for sure.

letshin, 2023-05-27:
Has anyone tried something like this: https://www.paint-direct.co.uk/p/sandtex-dirt-repellant

Judman, 2023-06-29:
For those that have tried the MG Chemicals - 419D, did anyone prep the solder mask with sanding or anything?

Also, what about simple exterior spray paints or something?

Judman, 2023-07-16:
So I got some MG 419D-55ML because it seemed to be the most common option. I tried some like 1000 grit wet sanding to see if it would increase adhesion in the long run. 

However, in the short term it seems to have really changed the sensing capabilities of the moisture sensors. All the coated boards have multiple layers of the conformal coating with sanding in between. I am guessing the multiple layers is part of the problem and that the coating just lowers the moisture sensing capabilities in general.

I am beginning to wonder if it would be better to make a two part design with a "disposable" probe section and a board section. So that after the probe part corrodes you could just solder on a new one without redoing the entire board.

[image]

[image]
[image]

rbaron, 2023-07-17:
Tks for sharing @Judman. My first impression is that using a thicker, multi-layer coating may change the capacitance of the sensor enough to warrant a recalibration, but I imagine they might still work perfectly fine once it's done. 

To calibrate we need to collect points and fit [these two polynomials](https://github.com/rbaron/b-parasite/blob/95ec660cb6bdbe4c4dfd8ebc5f98a1141811ec9d/code/nrf-connect/prstlib/boards/arm/bparasite_nrf52840/bparasite_nrf52840_2_0_0.overlay#L9), used [here](https://github.com/rbaron/b-parasite/blob/main/code/nrf-connect/prstlib/src/adc.c#L96):
- f(voltage)_dry -> sensor_voltage [sensor out in the air]
- f(voltage)_wet -> sensor_voltage [sensor in a glass of water, up to the PCB marking]

For varying power supply voltages (around 3 V - 2 V). The [soil_read_loop sample](https://github.com/rbaron/b-parasite/tree/main/code/nrf-connect/samples/soil_read_loop) spits out pairs of (voltage, sensor_read), which I have used for computing the default coefficients.

Judman, 2023-07-24:
@rbaron I appreciate the guidance! I was finally able to get the soil read loop flashed and read using a cheap amazon debugger and the combo of J-link Flash lite and timing the release of the reset button with clicking the program button in the software.

I was then able to use the J-link RTT viewer to see the adc sensor values. Below is what I measured from one of the coated sensors.

I used my bench power supply to vary the voltage for the readings.
xxx | Dry | Water to line
3.3 | 545 | 190
3.0 | 536 | 194
2.8 | 530 | 200
2.6 | 525 | 200

Below 2.6v the sensor would drop out.

I have figure out a lot to get where I am now, but looking at the code sections you linked I am back in over my head as to how to make the adjustments for this calibration.  I was kind of expecting the numbers from the soil read to the like 6 digits to match the dry wet numbers in the calibration coeffs. Like I said I am lost.

I also somehow fried a board so, down to 3 of 5 orders haha Be careful out there tinkerers!

Judman, 2023-07-25:
My bad, the numbers above are from the bare board, here are the numbers from the coated board.

xxixx | Dry | Water to line
3.30 | 545 | 424
3.15 | 540 | 418
3.00 | 535 | 416
2.80 | 530 | 416
2.70 | 527 | 406
2.60 | 524 | 404

Judman, 2023-07-29:
So something is funky.
Below is my spreadsheet and the data logged from the soil test sketch.

[Google Doc](https://docs.google.com/spreadsheets/d/1d_42K0qu3Gp8OTpRAFzVyzU1p_m-XuC_G8FH6hterzM/edit?usp=sharing)

To make it short here are the values I changed in the soil coefficients.

```
	soil_calibration_coeffs: soil_calibration_coeffs {
		compatible = "soil-calibration-coeffs";
		dry = <374700 84600 (-10100)>;
		wet = <295000 44800 (-4500)>;
```

With those values in a new build flashed to the board it is now reading zero moisture in a cup full of water.

I re-validated my wet adc results and they are the same as in the sheet.

As for the dropping on low voltage, who knows. All three of my working boards have the same behavior.

Also, the MG 419D-55ML on the board I have had in soil for two weeks is starting to exhibit the same bubbles as JoelWise's boards.

rbaron, 2023-07-30:
@Judman I don't have access to your spreadsheet, but I think you're close to get it to work. If you enable debug logs, you should see [these logs](https://github.com/rbaron/b-parasite/blob/85ca1a67bc0e82ef829a8080c06b555a047b39b3/code/nrf-connect/prstlib/src/adc.c#L99) that should help identifying what's going on. The evaluated dry and wet values should be close to the ones in your collected spreadsheet for the corresponding nearest batt value.

Judman, 2023-08-09:
@rbaron Well I have given up on my previous coating method with with multiple coating of MG 419D-55ML. I could not get the calibration to work correctly. I am not trying to just edge coat the pcb with MG 419D-P-WH, a pen based pcb overcoating. It seems meant to repair pcb, maybe it will be moisture resistant enough.

One other question I did have is the effect of only coating the outer parts of the board and leaving the middle section uncoated. As seen in the image below, if only the blue area is coated and the middle section is masked off, would that still interfere with the measurements?
 I admit I do not fully understand how the capacitance measuring works, so my idea might be a moot point. 
[image]

rbaron, 2023-08-10:
@Judman maybe it's worth trying, but I'd feel safer with a full coat. I'd try a single coat of MG 419D-55ML. I have a few 1.0.0 boards still running since two years on a single coat of acrylic, but of course ymmv.

Can I ask you why did you give up on the calibration? Did you manage to get those logs? I think you were quite close to get it to work. Even if with decreased dynamic range, it's probably good enough for our use cases.

Judman, 2023-08-10:
@rbaron Thanks I may give the single coating another shot.

I gave up because the coating had already started to fail. There were clear bubbles forming under the coating very similar to Joelwise's boards, but to a lesser degree.

BenRoe, 2023-12-15:
Was there already a two piece design, with an interchangeable probe like @Judman suggested?
If yes why was it changed? It seems the easiest solution and also cheap.

chucknorris101, 2024-01-02:
Having taken a step back from this effort for a bit, wanted to follow up on the above attempts. 

My dataset isnt as big as I had hoped for the long term assessment, but for what sensors remain in plants (and not pending re-flashing or otherwise were moisturized) - the epoxy approach seems **very stable** - the marine grade stuff I was putzing with above does not seem necessary, just regular craft grade stuff is enough so far for ~6-8 months.

The trick is coating consistency, which is hard for essentially hand dipped or 'painted' on epoxy. (Perhaps 3d printing could be used to make some peel away molds to ensure uniform thickness of the epoxy - though the peel away process might cause imperfections again? maybe more of a 3d squeegee to dip and then pass through to remove excess for drying in the open....tbd)

The thickness of the epoxy definitely impacts the characteristics of the board - while thin films allow use of the standard/board based calibration profiles provided above, none of those really worked for my epoxy boards - everything showed as 100% all the time -  though they seemed to fall within a close enough range for my purposes, i wouldnt call them 'calibrated'. The thicker coating also impacts sensitivity, but unless you need like sub 1-2% measurement accuracy it should be fine.

My struggle with firm application and calibration practice has been that my use case (hidden/buried) requires basically the full board to be waterproofed. Ive been dealing with this so far with an epoxy lower over the capacitive area, then flexseal over the upper. very durable, but the application of the upper coating over the mainboard/etc seems to alter the calibration as well - in addition to causing issues with battery connections. Trying to keep these more re-usable Ive been seeing if there is a good balance somewhere in the prepwork (maybe a different battery holder design or pre-coating the battery connection points to prevent ingress of the 'overmold' material)

TL;DR - Craft Epoxy dip/paint should work indefinitely

alexhk, 2024-01-06:
I really like your project (zigbee!) and started to consider ordering some PCBs for it.

I have some experience with various 3M potting compounds (ie. 3M Scotch-Weld Epoxy Potting Compound DP270 and others) to seal PCBs and connectors, also Devcon Flexane (a rosin, but we also call it "liquid rubber"). Also, very limited experience with 3M Novec coatings (for harsh environments).

How thick could a layer of coating be and still give meaningful feedback from the RC circuit?

One thing that was not touched in this thread is mechanical wear and tear when the sensors are used in a busy environment (ie. rotated to different pots several times a year). I'd also like to try with other "hard" substrates such as Lechuza PON or Seramis, which will scratch the boards.
 
My next problem is the length of the sensor. For example, I'd like the sensor (the RC circuit actually) to be close to the bottom of of larger pots, which currently is not possible.

So, having said all that, and having experience of the cost and downfalls (time, money, failures, etc) of coating and potting, would it be a good idea to think about separating the RC circuit from the actual device? 

There is an open issue where somebody suggest a Lite version for almost the same reasons:
https://github.com/rbaron/b-parasite/issues/161

He links to a sensor PCB for Arduino that cots $0.35 on AliExpress (for his link to work you have to add .html). 

If b-parasite could separate the RC circuit from the actual device and a dump sensor PCBs could be produced for a few cents (I assume that would be PCB and 2 soldered wires or connector only), it might actually solve a few problems:

- forget about coating the whole thing
- replace sensor PCBs when not reliable anymore
- Use longer wires to "submerge" sensor PCBs deeper into the substrate (in this case a few drops of epoxy or similar to seal where the wires attach should do)

If I finally go ahead (still busy till mid February) and order some b-parasite PCBs, I can of course attempt some coating and maybe provide calibration feedback, but I can't make promises at this point.

pedro-jorge-pereira, 2024-01-25:
I came across another type of probe sensor on AliExpress. It's relatively inexpensive and advertised as corrosion-resistant. Here's the image. Are they a good alternative to the capacitive sensor? Perhaps this could fit the idea of an external probe, potentially eliminating the need to coat the boards.

[image]

> I really like your project (zigbee!) and started to consider ordering some PCBs for it.
> 
> I have some experience with various 3M potting compounds (ie. 3M Scotch-Weld Epoxy Potting Compound DP270 and others) to seal PCBs and connectors, also Devcon Flexane (a rosin, but we also call it "liquid rubber"). Also, very limited experience with 3M Novec coatings (for harsh environments).
> 
> How thick could a layer of coating be and still give meaningful feedback from the RC circuit?
> 
> One thing that was not touched in this thread is mechanical wear and tear when the sensors are used in a busy environment (ie. rotated to different pots several times a year). I'd also like to try with other "hard" substrates such as Lechuza PON (lava) or Seramis (clay), which will scratch the boards.
> 
> My next problem is the length of the sensor. For example, I'd like the sensor (the RC circuit actually) to be close to the bottom of of larger pots, which currently is not possible.
> 
> So, having said all that, and having experience of the cost and downfalls (time, money, failures, etc) of coating and potting, would it be a good idea to think about separating the RC circuit from the actual device?
> 
> There is an open issue where somebody suggest a Lite version for almost the same reasons: #161
> 
> He links to a sensor PCB for Arduino that cots $0.35 on AliExpress (for his link to work you have to add .html).
> 
> If b-parasite could separate the RC circuit from the actual device and a dumb sensor PCB could be produced for a few cents (I assume that would be PCB and 2 soldered wires or connector only), it might actually solve a few problems:
> 
> * forget about coating the whole thing
> * replace sensor PCBs when not reliable anymore
> * Option to produce longer RC circuit PCBs for larger pots
> * Use longer wires to "submerge" sensor PCBs deeper into the substrate (in this case a few drops of epoxy or similar to seal where the wires attach should do)
> 
> If I finally go ahead (still busy till mid February) and order some b-parasite PCBs, I can of course attempt some coating (good old dip into varnish?) and maybe provide calibration feedback, but I can't make promises at this point.

mrred2k, 2024-04-29:
I have added UV-curing resin to a couple of boards and used 2 of them for 8 monath now. They are not living with very moist soil but they held up very nicely. 
I used the uv curable resin used in my anycubic 3d printer. This stuff is cheap and very low viscousity. (Anycubic Photon Standard Resin - 35€/1.5l)  This means that you will get an even waterproof coating in a repeatable form. I hang them pointing straight down, outside for 1 day and everything is superhard. Thanks Sun!

I am currently working on the coeffs for this and I have gotten my first values and the coating is effecting the wet ( submersed) values. I get raw adc at 3.0V change from 110 to 250 losing a little bit of range. The change should in my opinion not be voltage dependend, but I will (as suggested) do some more tests.

deimi, 2024-06-01:
Has actually anyone tried using a vacuum sealer (https://www.amazon.com/VS-12-Deluxe-Starter-Viewing-Compact/dp/B01KCK9W1K/) to seal the whole sensor with it?
I guess the moisture measurement should still work and it would definitely be protected against corrosion. The air humidity measurement will not work anymore and the temperature sensor will probably be slower react to changes.

SchmidL, 2024-08-01:
hi - while reading the comments here, and not having the right equipment at home, i long avoided to tackle the calibration.
But finally i pulled the trigger on an J-link and an nrf PPK2  (power profiler 2) and started to make the calibration process as smooth as possible as facing at least 10+ boards with the hope of having even more. (I have still my original v1.2 boards unflashed since 2022... :/

so if someone is interested all the code and documentation is over at https://github.com/SchmidL/b-parasite_autocali

rbaron, 2024-08-01:
@SchmidL that's incredible, thanks for sharing.

I would be curious to hear your experience regarding the variance in calibration across different boards from the same batch. I've always wondered how much of a difference it makes to use the same coefficients for all identical boards versus calibrating each individually. That, and if we can provide better defaults for when it's good enough, based on averaging the calibration of many boards.

SchmidL, 2024-08-05:
@rbaron - thank you, and hope it is useful.

I did the first batch, and the result is here https://github.com/SchmidL/b-parasite_autocali?tab=readme-ov-file#consistency-of-the-calibration-curve-for-multiple-sensors 
The results are for me not fully conclusive, as we have 1 "outlier" and also the rest of the sensors are not super close together, or what do you think?

kzyapkov, 2024-10-19:
My method was to use some basic epoxy and carefully apply around the edges. I finally got around to take a look at their condition after almost two years of service.

Specimen A had epoxy applied around the edges of the PCB and shows no visible signs of damage:

[image]
[image]

The surprising result is, however, specimen B. I must have forgotten to treat it, so it went in to a pot with Ficus without any treatment, and also shows no visible signs of damage:

[image]
[image]

Devices come from a bulk order organized by @oleo65 . I have a few  from a fresh batch from him, and I am tempted to try using them without any treatment.

tteggelit, 2025-07-17:
Has anyone tried [Electrolube AFA200 Aerosol](https://electrolube.com/product/afa-aromatic-free-acrylic-conformal-coating/)?