rbaron/b-parasite #83: Visual studio code - FATAL ERROR: Unable to parse input tree

JoelWise, 2022-12-14:
I'm running into this build error. Help would be appreciated.

-- west build: generating a build system
Including boilerplate (Zephyr base (cached)): C:/<removed>/ncs/v1.7.1/zephyr/cmake/app/boilerplate.cmake
-- Application: C:/<removed>/GitHub/b-parasite/code/nrf-connect/samples/ble
-- Zephyr version: 2.6.99 (C:/<removed>/ncs/v1.7.1/zephyr), build: v2.6.99-ncs1-1
-- Found west (found suitable version "0.11.1", minimum required is "0.7.1")
-- Board: bparasite_nrf52833, Revision: 1.2.0
-- Cache files will be written to: C:/<removed>/ncs/v1.7.1/zephyr/.cache
-- Found dtc: C:/<removed>/ncs/v1.7.1/toolchain/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6")
-- Found toolchain: gnuarmemb (c:/<removed>/ncs/v1.7.1/toolchain/opt)
-- Found BOARD.dts: C:/<removed>/GitHub/b-parasite/code/nrf-connect/prstlib/boards/arm/bparasite_nrf52833/bparasite_nrf52833.dts
-- Found devicetree overlay: C:/<removed>/GitHub/b-parasite/code/nrf-connect/prstlib/boards/arm/bparasite_nrf52833/bparasite_nrf52833_1_2_0.overlay
Error: bparasite_nrf52833.dts.pre.tmp:371.13-14 syntax error
FATAL ERROR: Unable to parse input tree
CMake Error at C:/<removed>/ncs/v1.7.1/zephyr/cmake/dts.cmake:219 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/<removed>/ncs/v1.7.1/zephyr/cmake/app/boilerplate.cmake:553 (include)
  C:/<removed>/ncs/v1.7.1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  C:/<removed>/ncs/v1.7.1/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:40 (include_boilerplate)
  CMakeLists.txt:7 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'c:\<removed>\ncs\v1.7.1\toolchain\opt\bin\cmake.EXE' '-DWEST_PYTHON=c:\<removed>\ncs\v1.7.1\toolchain\opt\bin\python.exe' '-Bc:\<removed>\Documents\GitHub\b-parasite\code\nrf-connect\samples\ble\build' '-Sc:\<removed>\Documents\GitHub\b-parasite\code\nrf-connect\samples\ble' -GNinja

 *  The terminal process terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

rbaron, 2022-12-15:
It seems you're using an older ncs version. I would try updating it 2.1+. I'd also make sure the vscode extension is up to date.

JoelWise, 2022-12-15:
OK. I updated ncs and reinstalled the vscode extension. Seems to have worked.

I've added these lines to prj.conf to set a custom mac, saved, built, and flashed but the probe isn't set with this address. It's still picking a random one. Am I doing this correctly?

CONFIG_PRST_BLE_HAS_USER_DEFINED_RANDOM_STATIC_ADDR=y
CONFIG_PRST_BLE_USER_DEFINED_RANDOM_STATIC_ADDR="f5:a0:80:fa:ed:2f"

rbaron, 2022-12-15:
That should have worked I think. Maybe try doing a pristine build?

JoelWise, 2022-12-15:
I set it via Kconfig too and that resulted in the same ouput into prj.conf. There is no option for a pristine build.

[image]

[image]

rbaron, 2022-12-16:
Here's how to do a pristine build from VSCode:
[image]

I would also refrain from editing configs in the Kconfig editors while debugging, since the settings [are lost once you clean the build](https://docs.zephyrproject.org/3.2.0/build/kconfig/setting.html#setting-kconfig-configuration-values). Just add the configs like you did to `prj.conf`. Make sure it's a clean git checkout before that too -- I see some extra stuff in the screenshot you posted.

I just tested your settings myself and it seems to work as expected:

Relevant changes to `prj.conf`:
```
CONFIG_PRST_SLEEP_DURATION_SEC=1 # Just for quick debugging
CONFIG_PRST_BLE_HAS_USER_DEFINED_RANDOM_STATIC_ADDR=y
CONFIG_PRST_BLE_USER_DEFINED_RANDOM_STATIC_ADDR="f5:a0:80:fa:ed:2f"
```
Output from the RTT console:
```
*** Booting Zephyr OS build v3.1.99-ncs1-1  ***
[00:00:00.000,335] <inf> sdc_hci_driver: SoftDevice Controller build revision: 
                                         29 5c 92 f1 36 81 92 d1  b7 a9 f0 f1 99 e9 4c 19 |)\..6... ......L.
                                         1f 23 83 4a                                      |.#.J             
[00:00:00.003,143] <inf> ble: MAC Address: f5:a0:80:fa:ed:2f
```
I set up an ESP32 to scan BLE advertisements and I can see the expected user-supplied address:
```
[08:51:43][VV][esp32_ble_tracker:578]: Parse Result:
[08:51:43][VV][esp32_ble_tracker:595]:   Address: F5:A0:80:FA:ED:2F (RANDOM)
[08:51:43][VV][esp32_ble_tracker:597]:   RSSI: -76
[08:51:43][VV][esp32_ble_tracker:598]:   Name: 'prst'
[08:51:43][VV][esp32_ble_tracker:606]:   Ad Flag: 6
[08:51:43][VV][esp32_ble_tracker:623]:   Service data:
[08:51:43][VV][esp32_ble_tracker:624]:     UUID: 0xFCD2
[08:51:43][VV][esp32_ble_tracker:625]:     Data: 40.02.8E.06.03.7E.0F.05.9C.FF.63.0C.7D.0A.14.6F.02 (17)
```

JoelWise, 2022-12-16:
I tried a pristine build, fresh github clone, opening closing VSC. The mac config lines remain in prj.conf but still not working. Maybe I'll revisit this sometime later but for now I just used the legacy branch to get the power saving ble advertising changes onto my devices. Thanks for helping, though.

rbaron, 2022-12-16:
If you revisit it later, check out the output of the RTT console like the one I sent above. 

You can also try using Docker to build and debug your setup. 

Feel free to reopen the issue if/when you revisit it.

closed 2022-12-16 by rbaron