rbaron/b-parasite #87: Feature request of a Matter standard over Thread (Project CHIP) firmware image for b-parasite

Hedda, 2022-12-18:
I like to request an optional Matter firmware image for b-parasite which could be used as an alternative to Bluetooth and Zigbee FW.

https://csa-iot.org/all-solutions/matter/

To add and control Matter over Thread devices, users will need a Thread Border Router (e.g. USB radio dongle) and a Matter controller.

https://www.theverge.com/22832127/matter-smart-home-products-thread-wifi-explainer

https://en.wikipedia.org/wiki/Matter_(standard)

https://en.wikipedia.org/wiki/Thread_(network_protocol)

FYI, since the new Matter standard (including Matter over Thread protocol) also relies on the same ZCL (Zigbee Cluster Library specification) for device clusters and attributes as the very latest revision of the Zigbee 3.0 specification, it, in theory, should mean that so as long as try to strictly adhere to cluster and attribute standard with the official Zigbee Cluster Library (R8) and Zigbee PRO 2017 (R22) Protocol Specification then should, again in theory, also be compatibility to build a matching firmware for the Matter standard over Thread protocol:

* [Zigbee PRO 2017 (R22) Protocol Specification](https://zigbeealliance.org/wp-content/uploads/2019/11/docs-05-3474-21-0csg-zigbee-specification.pdf)
* [Zigbee Cluster Library (R8)](https://zigbeealliance.org/wp-content/uploads/2021/10/07-5123-08-Zigbee-Cluster-Library.pdf)
* [Zigbee Base Device Behavior Specification (V1.0)](https://zigbeealliance.org/wp-content/uploads/zip/zigbee-base-device-behavior-bdb-v1-0.zip)
* [Zigbee Lighting & Occupancy Device Specification (V1.0)](https://zigbeealliance.org/wp-content/uploads/2019/11/docs-15-0014-05-0plo-Lighting-OccupancyDevice-Specification-V1.0.pdf)
* [Zigbee Primer](https://docs.smartthings.com/en/latest/device-type-developers-guide/zigbee-primer.html)

Nordic Semiconductor nRF52840 b-parasite use already supports Thread protocol and now also applications for the new Matter standard:

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/samples_matter.html

https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/nrfconnect

https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-minimal-app/nrfconnect

For reference also check out:

https://github.com/project-chip/zap

https://github.com/project-chip/connectedhomeip/tree/master/examples/ota-requestor-app/ota-requestor-common

https://github.com/project-chip/connectedhomeip/tree/master/examples/common/QRCode

https://github.com/project-chip/connectedhomeip/tree/master/examples/common/screen-framework

PS: Home Assistant now support Matter over Thread via either their new Home Assistant SkyConnect USB stick or Home Assistant Yellow:

https://www.home-assistant.io/blog/2022/12/07/release-202212/

https://www.home-assistant.io/integrations/matter/

https://www.home-assistant.io/skyconnect/

https://community.home-assistant.io/t/home-assistant-skyconnect-usb-stick-announced-will-be-compatible-with-both-zigbee-and-thread-including-matter-chip-over-thread/433594

https://www.home-assistant.io/blog/2021/09/13/home-assistant-yellow/

https://yellow.home-assistant.io/

PPS: Others that will soon get support for Matter over Thread is the upcoming hubs from Athom Homey and Samsung SmartThings, as well as integrated Thread radios inside some Google Nest Hub / speakers (for Google Home)  and Amazon Echo (for Amazon Alexa).

Hedda, 2022-12-18:
btw, nRF offers possibility for concurrent/simultaneous multiprotocol support for Thread and Zigbee in parallel at the same time:

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/1.4.2/nrf/ug_multiprotocol_support.html

Another option there is to support both Thread and Zigbee in the same firmware image and allow the end user to choose which of those protocols they want to use by commissioning the device to either a Thread network or a Zigbee network.

rbaron, 2022-12-18:
I've been looking at Thread recently and it seems both interesting and feasible to put together a proof of concept based on the [nRF Connect Thread Samples](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/samples_thread.html). The goal in separating the common core in a `prstlib` was exactly to make these tings easier.
What it's currently not clear to me is how the [Thread licensing/certification](https://www.threadgroup.org/What-is-Thread/Certification) plays with an open source projects. Ideally I would want to create a clearly uncertified, experimental, DIY-style, dev-board-style firmware sample with Thread support, but I'm not 100% clear whether it's possible without becoming a member. I would love to understand how this works better.

Hedda, 2022-12-19:
> What it's currently not clear to me is how the [Thread licensing/certification](https://www.threadgroup.org/What-is-Thread/Certification) plays with an open source projects. Ideally I would want to [...]

As I understand it, the Matter standard still allows for uncertified products for DIY and development, etc. on a local Matter fabric network as long as the user's Matter controller enables support for adding uncertified products if the Matter controller support it (which open-source home automation solutions like Home Assistant should obviously support sooner or later if they do not already).

FYI, while the [node-matter project by mfucci](https://github.com/mfucci/node-matter) (which depends on project-chip's [matter.js](https://github.com/project-chip/matter.js) library) is still in early development it has already proven that it is possible to add virtual DIY node.js devices via the Matter to latest Apple iOS platform (which was the first commercial smart home platform to release support for Matter). That node-matter project does not use Thread or even an MCU radio but it is still a nice proof of concept that it is possible to add DIY devices to a Matter controller:

https://github.com/mfucci/node-matter

Home Assistant / Nabu Casa has also got a DIY workshop on how to build your own Matter device (for Wi-Fi) with an Espressif ESP32-C3 devkit, and while the Matter Integration installation instruction documentation for Home Assistant in that workshop is a little bit outdated (since Home Assistant 2022.12 has now been released with Matter support so no need to external development repositories for that6 part), the part about building your own Matter device using an Espressif ESP32-C3 devkit development board with their example applications and firmware should still be valid (again their workshop is currently only for Wi-Fi as just proof-of-concept).

https://www.home-assistant.io/blog/2022/05/29/matter-in-home-assistant-workshop-announcement/

https://www.youtube.com/watch?v=9fOHBl5w0_k&ab_channel=HomeAssistant

https://docs.google.com/document/d/1MldgA337wHTEx-vPkoxlCqKHrhVm5QIcHV32iLsv_2k/

https://nabucasa.github.io/matter-example-apps/

https://github.com/NabuCasa/matter-example-apps

Espressif has as well relevantly recently made available a new "ESP32-H2" series of SoC and radio module (which replace the WiFi radio part of the ESP32-C3) with an 802.15.4 radio supporting Zigbee 3.0, Thread, and Bluetooth (BLE 5.2), so I would expect there to come more examples soon on how to build DIY devices that use Matter over Thread based on it.

https://www.espressif.com/en/news/ESP32_H2

https://docs.espressif.com/projects/esp-matter/en/main/esp32/

https://github.com/espressif/esp-matter

Anyway, the Thread protocol is only the network layer part which does not require licensing or certification, however, the separate Matter standard is part of the application layer and all commercial Matter devices need to be officially tested and certified by the CSA (Connectivity Standards Alliance) to get into their official database listing of CSA certified products as a signed "Certified End Product", which means commercial companies need to get their products certified and signed for marketing purposes if nothing less as certification is meant to help customers and companies to show that the device and firmware have been tested as a "Certified End Product" (which device signatures I believe is stored in a decentralized blockchain ledger that it meant to get regularly updated on all certified Matter controllers).

PS: Regarding interoperability for the Matter standard, it should be noted that even if a device is an official "Certified End Product" it is still up to each Matter controller to approve which devices it will allow an end user to add. So do not expect brand-centric manufacturers example the latest IKEA Trådfri "Dirigera" Gateway to allow users to add non-IKEA devices even though it could technically support it, and similar goes for other commercial hubs, such as Aqara and Tuya, etc. as they will probably all only allow you to add devices that have their own brand.

rbaron, 2022-12-19:
> Anyway, the Thread protocol is only the network layer part which does not require licensing or certification

Can you help me find a link to back this claim?

Hedda, 2022-12-19:
Licensing of code/binaries is one thing which is a little more black and white so that part should be easier to get what is approved.

Regardless I think for certification you have to read between the lines because I believe that the documentation about certification is really targeting companies that make and sell commercial products and as such do not cover the scenario of making DIY products as those are normally not certified as made du individuals and not companies.  However, it is still nice that the SDK used has been pre-certified as that testing demonstrates compliance, and it will help if any company would ever like to get a specific product certified based on the code from this project. 

https://csa-iot.org/all-solutions/matter/matter-faq/

https://www.threadgroup.org/What-is-Thread/Certification

https://devzone.nordicsemi.com/f/nordic-q-a/52448/thread-certification-needed

https://www.nordicsemi.com/products/matter

  "_Nordic is actively involved and one of the main contributors the development of Matter. We are the first company to receive a Thread 1.3 certification badge. This badge can be inherited by our customers for their Thread products and is a prerequisite for Matter over Thread._"

As I understand it, you should only need a certification if you as a company make commercial products if plan on selling them a certified product in order to use their certificate badging (which then is per device) in order to be legally allowed to use the Matter badge for public marketing. 

Also note that Matter device certification is separate from Thread device certification, and certification for both is actually needed if want to sell a device that is certified for Matter over Thread. 

So if you are planning on selling commercial products as certified products then know that using the nRF Connect SDK or nRF5 SDK then the OpenThread pre-built binary libraries have already been pre-certified by the Thread Group for various Nordic Semiconductor devices so you can apply for a product-specific "certification by inheritance".  What that means is that a device can be quickly certified via paperwork alone as no more testing is needed as long as you use their SDK as designed and meet the requirements for "certification by inheritance".

As such, certification should not be needed for DIY devices because they will by definition not get certified as no one is likely to pay in order to go through the certification process, and even if they did the certification would likely only be legally binding to the legal entity that submitted the device for certification. 

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_thread_certification.html

https://www.nordicsemi.com/News/2017/09/Thread-certification

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_tz_v4.2.0%2Fthread_certification.html

https://infocenter.nordicsemi.com/index.jsp?topic=%2Fsdk_tz_v3.2.0%2Fthread_openthread_certification.html

Disclaimer: I am not a lawyer.

Hedda, 2022-12-19:
FYI, nRF Connect SDK has an example application for a "Matter weather station" as well as a Matter template + clusters examples:

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/applications/matter_weather_station/README.html#matter-weather-station-app

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/samples_matter.html

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/matter/template/README.html

https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/ug_matter_gs_adding_clusters.html#ug-matter-creating-accessory

https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/nrfconnect

https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-minimal-app/nrfconnect

daniel-nicolai-hansten, 2023-03-20:
Hi

I am using Zephyr with Openthread on the b-parasite hardware. I have not implemented Matter yet.
The part of the code that posts to my backend would not fit with the rest of your project, but I could create a Zephyr project with boardfiles for the b-parasite. That could be a good startingpoint for people trying to get Matter support with nrf Connect.

Where in this repo (IE where in the folderstructure) should an eventual PR put the firmwarefiles for Zehyr/ nrf Connect?

rbaron, 2023-03-21:
Hi @daniel-nicolai-hansten,

That's exciting, tks for joining in.

We extracted common board definitions and utilities into `prstlib`. I would be pretty picky that new firmwares must use the common devicetrees/Kconfig for all board revisions, as well as prstlib utils for initialization, io, logging, sensor sampling etc. So in a sense we already have the board definitions you're proposing to add.

As for the thread sample, as you noted, the application layer is pretty flexible and setup-dependent, in that we would have to define our own protocol on top of it. I have on my TODO list to get one of the [nrf-connect sdk samples](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/samples/thread.html#openthread-samples) to b-parasite using the common `prstlib`, so hopefully I can get that to work soon.

I think a good contribution would be to link your standalone project here so we can take a look and learn from it. Tks!