rbaron/b-parasite #88: jlcpcb question on components Q4 polarity

fasavard, 2022-12-21:
Hi, 
I did all the step on the PCB procedure for jlcpcb, and they had a question regarding the schematic. 
Here it is (please dont hesitate if i should not post this here)

`
Sorry to bother you, but there is one thing that we want to confirm with you about your SMT order before proceeding.

Since we are not so sure about the polarities of the components Q4 are correct or not. Could you please kindly check if the polarities and placements of the components are correct in the below picture? Is it okay to proceed with production?

`
[image]

1technophile, 2022-12-21:
See https://github.com/rbaron/b-parasite/issues/42

fasavard, 2022-12-21:
Great thanks for the reference
They also did ask this question

As shown below, for the indicated pad at arrow, it's not exposed in solder mask layer, shall we add opening for it in solder mask layer and cover them with tin?
[image]

And accepted the their suggestion to add opening for it in solder mask layer and cover them with tin.
Was that a good choice ? 
Should i tried to add a common question from JPLPLC in the wiki trought a pull request ?

rbaron, 2022-12-21:
@fasavard the solder mask is there for two reasons: prevent the water from shorting the parasitic capacitor pads and hopefully offering extra protection. Since it's strongly suggested that you [coat the board](https://github.com/rbaron/b-parasite/wiki/Conformal-Coating) anyway, it's possible that it's still fine either way. I'd still default to solder mask on though. I'm also not clear on how removing the solder mask will affect the calibration (as it was done with solder mask).

closed 2022-12-21 by rbaron