rbaron/b-parasite #90: zigbee-herdsman converter for Zigbee sample to work out-of-the-box in Zigbee2MQTT and IoBroker

Hedda, 2022-12-27:
zigbee-herdsman converter for Zigbee sample for b-parasite Zigbee devices to work out-of-the-box in Zigbee2MQTT and IoBroker:

https://github.com/rbaron/b-parasite/tree/main/code/nrf-connect/samples/zigbee

As per the initial Zigbee sample pull request:

https://github.com/rbaron/b-parasite/pull/81

https://github.com/rbaron/b-parasite/tree/main/code/nrf-connect/samples/zigbee

> The `b-parasite.js` file in this PR contains a converter that can be installed to Zigbee2MQTT to support this sample. See "Support new devices" for instructions.

https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html

Will anyone submit that `b-parasite.js` converter file as a pull request to zigbee-herdsman-converters repo for inclusion by default?

https://github.com/Koenkk/zigbee-herdsman-converters

ginkel, 2023-05-27:
If that's ok for @rbaron, I can submit a PR to [zigbee-herdsman-converters](https://github.com/Koenkk/zigbee-herdsman-converters).

rbaron, 2023-05-28:
Thanks @ginkel, that would be great. Could you please test [our converter](https://github.com/rbaron/b-parasite/blob/main/code/nrf-connect/samples/zigbee/b-parasite.js) against the latest HA just to be sure before submitting it? We also need to [submit a photo](https://www.zigbee2mqtt.io/advanced/support-new-devices/01_support_new_devices.html#_4-add-device-picture-to-zigbee2mqtt-io-documentation). I would suggest [this one](https://github.com/rbaron/b-parasite/blob/main/img/resized/b-parasite-2.0.0.jpg).

Thanks!

ginkel, 2023-05-28:
Excellent! I am using the b-parasite with z2m and latest HA 2023.5.4, so I can confirm that everything is working as expected. I just noticed in the converter that battery voltage reporting is commented out. Do you remember the reason for this?

Regarding the photo: I do have one with a more neutral background. The PCB has slightly rounded edges, though, but is unmodified otherwise: https://photos.app.goo.gl/smRuBSAahcgkgPyx8

rbaron, 2023-05-28:
> Do you remember the reason for this?

IIRC I thought the `batteryVoltage` attribute was not reportable, only the percentage. But I see [some devices in the converts repo](https://github.com/search?q=repo%3AKoenkk%2Fzigbee-herdsman-converters%20batteryVoltage&type=code) that do configure it for reporting, so maybe I was wrong 😬. It would be great to give this a test, but we can also remove the commented code and just work with the percentage for now. Up to you! 

> Regarding the photo

I kinda like the one in our repo because it gives a sense of scale with the coin and also shows the battery clip on the back. Unless you feel strongly about it, I'd vote for that one.

Tks again!

rbaron, 2023-05-28:
@ginkel I think I remember the reason now. In the [Zigbee Cluster Library specs](https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf) (section 3.3.2.2.3):

[image]

Only the `BatteryPercentageRemaining` is marked as reportable, and `BatteryVoltage` is not. But it's possible that different stacks also handle reporting on it.

ginkel, 2023-05-28:
Interesting. I have a couple of ZigBee devices that report the battery voltage, so I'll see if I can get this working.

ginkel, 2023-05-28:
> I kinda like the one in our repo because it gives a sense of scale with the coin and also shows the battery clip on the back. Unless you feel strongly about it, I'd vote for that one.

I think we need a photo with a white background, at least that's what is usually requested when adding a new device.

rbaron, 2023-05-28:
@ginkel Sounds good, if that's a requirement please feel free to use the one you took. Thanks for pushing this forward.

ginkel, 2023-05-30:
The [PR](https://github.com/Koenkk/zigbee-herdsman-converters/pull/5810) has been merged, I expect the support to be part of the next zigbee2mqtt release.

rbaron, 2023-05-30:
That's great, thanks once again @ginkel.

closed 2023-05-30 by rbaron