rbaron/b-parasite #94: Zigbee sample fails to build

andresp, 2023-01-06:
While trying to build for an nrf 52840-1.2 with nrf connect sdk 2.2.0, I'm getting the following error from CMake:

```
warning: IEEE802154_NRF5_FCS_IN_LENGTH (defined at drivers/ieee802154/Kconfig.nrf5:65) has direct dependencies IEEE802154_NRF5 && IEEE802154 with value n, but is currently being y-selected by the following symbols:
 - NET_L2_ZIGBEE (defined at C:/ncs/v2.2.0/nrf\subsys\zigbee/Kconfig:29), with value y, direct dependencies NETWORKING && ZIGBEE (value: y), and select condition NETWORKING && ZIGBEE (value: y)

error: Aborting due to Kconfig warnings

CMake Error at C:/ncs/v2.2.0/zephyr/cmake/modules/kconfig.cmake:328 (message):
```

Unfortunately, I'm unfamiliar with CMake and Kconfig. Any help would be appreciated.

kenny00111, 2023-01-07:
I had the same issue with the 2.2.0 SDK, it works with the 2.1.2

andresp, 2023-01-08:
Can confirm, worked with 2.1.2.

RobertLukan, 2023-01-09:
Same for me.

MJDSys, 2023-01-19:
It appears SDK 2.2.0 changed such that the device tree needs to enable 802.15.4 support.  I have a patch locally, but I'm waiting for hardware to arrive to try flashing and confirm it's working.  I can try submitting a PR once I know my changes are good.

PR #101 (2023-01-23): Update Zigbee example to work with SDK 2.2.0
The new SDK requires 802.15.4 to be enabled in the board DTS file.

Also disable IPv6 explicitly to avoid compile errors related to its network stack, since it is not used currently.

Closes #94
commit af47848b (2023-01-23): Update Zigbee example to work with SDK 2.2.0

The new SDK requires 802.15.4 to be enabled in the board DTS file.

Also disable IPv6 explicitly to avoid compile errors related to its
network stack, since it is not used currently.

Closes #94
closed 2023-01-23 by rbaron