rbaron/b-parasite #99: Connect to home assistant

sam632, 2023-01-19:
Hi, I downloaded this repo and flashed it to a b-parasite using the ESP32 method and all went well. I get the flashing red light when I put the battery in and after some time (overnight) it appeared in home assistant. I pressed configure but there doesn't seem to be any entities. This is all I see, 
[image]

How do I configure it to send the data to Home Assistant?

jhbruhn, 2023-01-25:
I believe this is related to #98

rbaron, 2023-01-25:
@sam632 are you using a proxy or talking to HA directly? How far away is it?

sam632, 2023-01-26:
> @sam632 are you using a proxy or talking to HA directly? How far away is it?

talking directly to HA. It is currently next to the RPI4 but I will buy a bluetooth antenna for the RPI as the range is terrible.

RobertLukan, 2023-01-31:
Which nRF chip are you using ? I have replaced ceramic antenna with 1mm thick wire and there is a tremendous improvement. There is an open issue about this topic. You might want to consider the same if you are facing signal issues.

sam632, 2023-02-01:
I'm using the E732G4M08S1C. It has a ceramic antenna so I will look into that, thanks!

rbaron, 2023-03-21:
Closing due to inactivity. Note that #98 may have fixed the issue. Please feel free to report back.

closed 2023-03-21 by rbaron