rbaron/b-parasite #100: Moisture Fluctuation

chucknorris101, 2023-01-21:
I have a sensor seemingly with bipolar disorder - but its not the sensor's fault! or so it seems. I have confirmed this weirdness with a second sensor known good. Would this be something with the sensor contacting the plant roots? Or has anyone seen something like this?

[image]

andrelung, 2023-01-22:
I've seen the same behaviour with mifloras. However I haven't figured out what's causing it.

rbaron, 2023-02-12:
That's a curious pattern that suggests hardware issues to me, specially if other sensors from the batch are working. I would double check the solder nRF52 module points and double check you have the correct module (nRF52840 vs. nRF52833) configured in the SDK.

rbaron, 2023-03-21:
Closing due to inactivity. Please let us know if you managed to identify the issue.

closed 2023-03-21 by rbaron