rbaron/b-parasite #104: Values change each time I move the sensor

Ulrar, 2023-02-01:
Hi,

I've had one board in a plant for a few months, and it's been reporting around 90% pretty reliably since, never really going down.

A few weeks ago I tried moving it to another plant, and it reported 90%, but started slowly trending down (with jumps by one or two percent either way on each reading, but trending down overall). It was down to around 85% the other day when I decided to take it out, and update the firmware with the new ble samples format.

I went with BTHome v2, and since it's been very spotty : 
[image]
Because I went from BTHome 1 to 2 the statistics weren't compatible so I dropped all the old data, this is what I got since.

Going unavailable and unpredictable reporting periods aren't great, but my real issue here is the numbers it's reporting. It went from ~85% before I flashed it down to 30, moved it and it went to 0, moved it again and it's now up to around 10, trending up. I'm only moving the board by a few centimeters each time.

Am I doing this wrong, and how do I go about placing these correctly ? I have no idea what the real number is.

I did coat the board, although it might not be done well.
Thanks

rbaron, 2023-02-01:
Hi @Ulrar, 

We've been looking into the BTHomeV2 data gaps issue in #98, so I'll focus on the numbers you're seeing. It's somewhat usual to see some interesting patterns (e.g.: #50, #2, [this comment](https://github.com/rbaron/b-parasite/issues/19#issuecomment-1261122074)), specially during the first couple of watering cycles. The jumps you're seeing when you move the sensor are likely due to:
1. Moving the sensor creates some air pockets between the sensor and the ground. These capacitive sensors must be in close contact with the soil for a reliable reading. You can try a quick experiment by holding the pads tightly with your hands and then removing some pressure. You'll see some significant difference in readings.
2. My mental model of the soil is close to a sponge. For example, when watering it, it takes naturally some time for the moisture to become close to uniform in the pot and the readings to converge. I speculate that there's a local disturbance when you move it, and it takes some time to converge back.
3. It seems like there is some "capillarity" effect in the surface of the sensor, in a way that it first "saturates" with water before setting into a more stable mode.

Whenever I place/replace a sensor, I try to stick it in a new place to avoid 1 and I make sure to compact the soil around it. If I'm testing something I will put some water around it to help it accommodate itself in the soil too.

Ulrar, 2023-02-02:
Thanks for the answer, that makes sense !

I hadn't actually realized there's a Zigbee firmware now, I do have a lot of Zigbee devices so I just flashed that. Looks good so far, but I'll keep an eye on it.

I took your advice hopefully correctly and compated the soil a bit, put a little bit of water and planted the sensor in moist compacted soil. It immediately started reporting 84%, which at least sounds about right.
I'll monitor that over the next few days and see how it goes, thanks for the advice !

rbaron, 2023-02-09:
@Ulrar, we're tracking the issue with HA marking the sensor as unavailable in #98. It seems like there was a bug with HA - see https://github.com/rbaron/b-parasite/issues/98#issuecomment-1424645478.

I'll close the issue for now, tks!

closed 2023-02-09 by rbaron