rbaron/b-parasite #113: Zigbee Home Assistant zha no battery state

electron23, 2023-03-21:
Hi,

I flashed my devices with the zigbee firmware. With the first device I have no problems with the first device and the states are reported correctly in home assistant (zha). However the second device does not report the battery state. 

[image]

The signature is:
`{
  "node_descriptor": "NodeDescriptor(logical_type=<LogicalType.EndDevice: 2>, complex_descriptor_available=0, user_descriptor_available=0, reserved=0, aps_flags=0, frequency_band=<FrequencyBand.Freq2400MHz: 8>, mac_capability_flags=<MACCapabilityFlags.AllocateAddress|MainsPowered: 132>, manufacturer_code=4660, maximum_buffer_size=108, maximum_incoming_transfer_size=1617, server_mask=11264, maximum_outgoing_transfer_size=1617, descriptor_capability_field=<DescriptorCapability.NONE: 0>, *allocate_address=True, *is_alternate_pan_coordinator=False, *is_coordinator=False, *is_end_device=True, *is_full_function_device=False, *is_mains_powered=True, *is_receiver_on_when_idle=False, *is_router=False, *is_security_capable=False)",
  "endpoints": {
    "10": {
      "profile_id": 260,
      "device_type": "0x0008",
      "in_clusters": [
        "0x0000",
        "0x0001",
        "0x0003",
        "0x0400",
        "0x0402",
        "0x0405",
        "0x0408"
      ],
      "out_clusters": []
    }
  },
  "manufacturer": "b-parasite",
  "model": "b-parasite",
  "class": "zigpy.device.Device"
}`

When I read it manually, it shows up the value: 
[image]

Do you know whats going wrong? 

Thanks

rbaron, 2023-03-21:
I think it's possible that there was an issue during pairing and ZHA didn't pick up the _reporting_ attribute for the battery. Since one of your devices worked, I would try to remove and re-pair the second device.

Let us know how it goes.

electron23, 2023-03-22:
Many thanks for your response. 

I tried pairing again both devices. Now the before working device isn´t recognized and what was previously faulty device is now working with the battery state. I will check it further and come back with the results.

oleo65, 2023-03-27:
Nice to see that you have an Aloe Vera monitored as well. Same here. 😉 

I experienced the same issue with my zigbee devices but I was not able to reproduce the issue consistently.

What is your zigbee coordinator device (zigbee stick?). I use a ConBee II via ZHA in HomeAssistant.

What helped me was:

1. Remove the device from home assistant (delete on device page). The LED on the parasite should start flashing shortly after indicating the network leave.
2. Factory reset the parasite by double power cycle.
3. Readd the parasite to home assistant.

Usually the reporting of all clusters was fixed by that. On one occasion however I ended up erasing and completely reflashing the parasite. I am still not convinced that that was the real fix for the issue, but it got resolved in the end. 🤷 

As a sidenote: I usually find the zigbee implementation quite reliable but there are still some rare issues with my devices, including the mentioned power cluster reporting. Some of my devices lose the network connection after working for weeks and will only reconnect after power cycling. As these issue are quite hard to debug, I tend to live with them for now but I suspect the zigbee nrf SDK implementation to be somewhat quirky.

BenRoe, 2023-12-22:
Can be closed. Tested Zigbee firmware and ZHA. Everything works as expected.

closed 2023-12-22 by rbaron