rbaron/b-parasite #114: Unable to reprogram an already programmed b-parasite

cloyd1815, 2023-03-22:
I followed along with discussion #105. I got my boards the other day and I'm just getting around to programming. I got two boards soldered up with battery clips. I programmed one with BLE to test out and one with zigbee. When I go to reflash the BLE one (or the zigee) I get an error from nrfjprog commands. Any command I run shows `[error] [SeggerBackend] - JLinkARM.dll reported "-1", "An unknown error.".`. Is it possible to reprogram these or are they one and done? Am I doing something wrong?

The commands I'm trying are 
`nrfjprog --family NRF52 --recover`
`nrfjprog -f nrf52 --program zephyr.hex --sectorerase`

RobertLukan, 2023-03-22:
I faced this issue also. Initially, I did not have any problems with flashing boards with RPI and OpenOCD, but I was unable to get RTT debug running. So I decided to purchase nrf dev kit. At this moment I started getting problems. I got this unknown error and I just could not understand it why for a couple of days. I decided to revert back to RPI and everything was ok again. In between, I decided to provide power from RPI and not for dev kit and have only Data and CLK connected to dev kit. Try to power your board by using another 3,3V source, but make sure to have GND connected to external source and dev kit.

cloyd1815, 2023-03-22:
I can try that, do you think trying to program while the battery is in instead would work too? 

I forgot to mention in my post that I am using the same programmer as the #105 post. Which is [this ](https://www.amazon.com/gp/product/B07V2CKG19/)one.

cloyd1815, 2023-03-22:
I successfully re-flashed one with the battery. I removed the pin from my 3.3v on my pogo pin clip and soldered on a battery to a new board and then flashed the zigbee hex to it. I set it up in HA then connected it again and ran `nrfjprog --family NRF52 --recover` without error and flashed it again with the zigbee hex without error.

rbaron, 2023-03-23:
Thanks for helping out @RobertLukan.  

@cloyd1815 since you managed to fix it, I'll close the issue. Please feel free to reopen.

closed 2023-03-23 by rbaron