rbaron/b-parasite #119: Q4 orientation in Hardware 2.0

Helmi, 2023-04-15:
I just ordered with JLCPCB this morning and similar to #42 and #106 their engineers are getting back to me regarding Q4 orientation.

They don't suggest to change it but ask to confirm it. Not being the hardware professional myself I'm unsure about that.

Can anyone help confirming the orientation with the collector down and emitter up in HW2?

[image]

rbaron, 2023-04-15:
Hi @Helmi,

Please refer to the [Wiki](https://github.com/rbaron/b-parasite/wiki/How-to-order:-PCB-fabrication-and-SMT-assembly#photodiode-q4).

closed 2023-04-15 by rbaron

Helmi, 2023-04-15:
Hmm but the wiki refers to Hardware 1.2 only, or did I miss something? If the footprint dot is still the valid reference I'd suggest updating the headline so this is clearly also valid for HW 2.0.

rbaron, 2023-04-15:
Yes, the dot is still the reference. Updated it.