rbaron/b-parasite #122: Visual Code - Problem with build

schejbity, 2023-04-19:
Can you tell me what I'm doing wrong?

I can't create a build in VS Code.
I tried updating the nRF Connect SDK via the toolchain manager.

Here is the error listing from the terminal.

Thanks.

[image]

rbaron, 2023-04-19:
@schejbity I just tried building the `ble` and `zigbee` samples using the 2.3.0 nrf-connect SDK and it seems to work fine for me. Did you get the same error before updating the SDK? 

I would try:
- Double checking your repository is clean
- Doing a pristine build (alternatively removing the `build/` directory inside the sample dir you're building)
- Trying to build some other built-in sample from the SDK (like [blinky](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/zephyr/samples/basic/blinky/README.html))

schejbity, 2023-04-21:
I have already tried several examples - ble and soil_read_loop. Without success.

Yes, the same errors were displayed even with the older version of the SDK before the update.

The repository was freshly downloaded, with no changes. I will try to download it again.

In the following days, I will try the blinky example.

Thanks.

schejbity, 2023-04-23:
Still not working :-(

I tried downloading the entire repository again, reinstalling the VSCode extension and still nothing.
Also delete all `build` directories.
Build succeeded only for the "blinky" example.

Is my build configuration correct?
[image]

Here is the output from the terminal, when building the example "ble":
```
[200/214] Linking C static library zephyr\kernel\libkernel.a
[201/214] Building C object prstlib/CMakeFiles/prstlib.dir/src/adc.c.obj
FAILED: prstlib/CMakeFiles/prstlib.dir/src/adc.c.obj 
C:\ncs\toolchains\v2.3.0\opt\zephyr-sdk\arm-zephyr-eabi\bin\arm-zephyr-eabi-gcc.exe -DKERNEL -DNRF52840_XXAA -DUSE_PARTITION_MANAGER=0 -D_ANSI_SOURCE -D_FORTIFY_SOURCE=1 -D__LINUX_ERRNO_EXTENSIONS__ -D__PROGRAM_START -D__ZEPHYR__=1 -ID:/b-parasite-main/code/nrf-connect/samples/ble/src -ID:/b-parasite-main/code/nrf-connect/prstlib/include -IC:/ncs/v2.3.0/nrf/drivers/mpsl/clock_control -IC:/ncs/v2.3.0/zephyr/include -Izephyr/include/generated -IC:/ncs/v2.3.0/zephyr/soc/arm/nordic_nrf/nrf52 -IC:/ncs/v2.3.0/zephyr/lib/libc/newlib/include -IC:/ncs/v2.3.0/zephyr/soc/arm/nordic_nrf/common/. -IC:/ncs/v2.3.0/zephyr/subsys/bluetooth -IC:/ncs/v2.3.0/nrf/include -IC:/ncs/v2.3.0/nrf/lib/multithreading_lock/. -IC:/ncs/v2.3.0/nrf/subsys/bluetooth/controller/. -IC:/ncs/v2.3.0/nrf/tests/include -IC:/ncs/v2.3.0/modules/hal/cmsis/CMSIS/Core/Include -IC:/ncs/v2.3.0/modules/hal/nordic/nrfx -IC:/ncs/v2.3.0/modules/hal/nordic/nrfx/drivers/include -IC:/ncs/v2.3.0/modules/hal/nordic/nrfx/mdk -IC:/ncs/v2.3.0/zephyr/modules/hal_nordic/nrfx/. -IC:/ncs/v2.3.0/modules/debug/segger/SEGGER -IC:/ncs/v2.3.0/modules/debug/segger/Config -IC:/ncs/v2.3.0/zephyr/modules/segger/. -IC:/ncs/v2.3.0/modules/crypto/tinycrypt/lib/include -IC:/ncs/v2.3.0/nrfxlib/mpsl/include -IC:/ncs/v2.3.0/nrfxlib/mpsl/include/protocol -IC:/ncs/v2.3.0/nrfxlib/softdevice_controller/include -isystem C:/ncs/v2.3.0/nrfxlib/crypto/nrf_cc310_platform/include -fno-strict-aliasing -Os -imacros D:/b-parasite-main/code/nrf-connect/samples/ble/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -gdwarf-4 -fdiagnostics-color=always -mcpu=cortex-m4 -mthumb -mabi=aapcs -mfp16-format=ieee --sysroot=C:/ncs/toolchains/v2.3.0/opt/zephyr-sdk/arm-zephyr-eabi/arm-zephyr-eabi -imacros C:/ncs/v2.3.0/zephyr/include/zephyr/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wexpansion-to-defined -Wno-unused-but-set-variable -Werror=implicit-int -fno-pic -fno-pie -fno-asynchronous-unwind-tables -fno-reorder-functions --param=min-pagesize=0 -fno-defer-pop -fmacro-prefix-map=D:/b-parasite-main/code/nrf-connect/samples/ble=CMAKE_SOURCE_DIR -fmacro-prefix-map=C:/ncs/v2.3.0/zephyr=ZEPHYR_BASE -fmacro-prefix-map=C:/ncs/v2.3.0=WEST_TOPDIR -ffunction-sections -fdata-sections -specs=nano.specs -std=c99 -MD -MT prstlib/CMakeFiles/prstlib.dir/src/adc.c.obj -MF prstlib\CMakeFiles\prstlib.dir\src\adc.c.obj.d -o prstlib/CMakeFiles/prstlib.dir/src/adc.c.obj -c D:/b-parasite-main/code/nrf-connect/prstlib/src/adc.c
In file included from D:\b-parasite-main\code\nrf-connect\prstlib\src\adc.c:4:
C:\ncs\v2.3.0\zephyr\include\zephyr\devicetree.h:190:36: error: 'DT_N_NODELABEL_soil_calibration_coeffs_P_dry' undeclared here (not in a function)
  190 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                                    ^~~~~~~~~~~~~~~
C:\ncs\v2.3.0\zephyr\include\zephyr\devicetree.h:3903:29: note: in definition of macro 'DT_CAT3'
 3903 | #define DT_CAT3(a1, a2, a3) a1 ## a2 ## a3
      |                             ^~
D:\b-parasite-main\code\nrf-connect\prstlib\src\adc.c:21:34: note: in expansion of macro 'DT_PROP'
   21 | static const int dry_coeffs[3] = DT_PROP(DT_NODELABEL(soil_calibration_coeffs), dry);
      |                                  ^~~~~~~
C:\ncs\v2.3.0\zephyr\include\zephyr\devicetree.h:190:29: note: in expansion of macro 'DT_CAT'
  190 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                             ^~~~~~
D:\b-parasite-main\code\nrf-connect\prstlib\src\adc.c:21:42: note: in expansion of macro 'DT_NODELABEL'
   21 | static const int dry_coeffs[3] = DT_PROP(DT_NODELABEL(soil_calibration_coeffs), dry);
      |                                          ^~~~~~~~~~~~
C:\ncs\v2.3.0\zephyr\include\zephyr\devicetree.h:190:36: error: 'DT_N_NODELABEL_soil_calibration_coeffs_P_wet' undeclared here (not in a function)
  190 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                                    ^~~~~~~~~~~~~~~
C:\ncs\v2.3.0\zephyr\include\zephyr\devicetree.h:3903:29: note: in definition of macro 'DT_CAT3'
 3903 | #define DT_CAT3(a1, a2, a3) a1 ## a2 ## a3
      |                             ^~
D:\b-parasite-main\code\nrf-connect\prstlib\src\adc.c:22:34: note: in expansion of macro 'DT_PROP'
   22 | static const int wet_coeffs[3] = DT_PROP(DT_NODELABEL(soil_calibration_coeffs), wet);
      |                                  ^~~~~~~
C:\ncs\v2.3.0\zephyr\include\zephyr\devicetree.h:190:29: note: in expansion of macro 'DT_CAT'
  190 | #define DT_NODELABEL(label) DT_CAT(DT_N_NODELABEL_, label)
      |                             ^~~~~~
D:\b-parasite-main\code\nrf-connect\prstlib\src\adc.c:22:42: note: in expansion of macro 'DT_NODELABEL'
   22 | static const int wet_coeffs[3] = DT_PROP(DT_NODELABEL(soil_calibration_coeffs), wet);
      |                                          ^~~~~~~~~~~~
[... cut ...]
D:\b-parasite-main\code\nrf-connect\prstlib\src\adc.c:25:22: note: in expansion of macro 'DT_NODELABEL'
   25 |     GPIO_DT_SPEC_GET(DT_NODELABEL(fast_disch), gpios);
      |                      ^~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'c:\ncs\toolchains\v2.3.0\opt\bin\cmake.EXE' --build 'd:\b-parasite-main\code\nrf-connect\samples\ble\build'

 *  The terminal process terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.
```

rbaron, 2023-04-23:
The issue is likely that you're not including `prstlib` as described in the [docs](https://github.com/rbaron/b-parasite/wiki/How-to-Build-the-Firmware-Samples#with-nrf-connect-for-desktop--visual-studio-code). I suggest you use the provided `.code-workspace` file for the sample you're trying to build.

schejbity, 2023-04-26:
I did the following and the problem persists:
Did I add prstlib correctly?
[image]

I also noticed that in the "source files" some items are red.
[image]

Here is the terminal output once again, this time in full.
```
Building ble
C:\WINDOWS\system32\cmd.exe /d /s /c "west build --build-dir d:\b-parasite-main\code\nrf-connect\samples\ble\build d:\b-parasite-main\code\nrf-connect\samples\ble --pristine --board nrf52840dk_nrf52840 -- -DNCS_TOOLCHAIN_VERSION:STRING="NONE" -DBOARD_ROOT:STRING="d:/b-parasite-main/code/nrf-connect/prstlib""

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: D:/b-parasite-main/code/nrf-connect/samples/ble
-- Found Python3: C:/ncs/toolchains/v2.3.0/opt/bin/python.exe (found suitable exact version "3.8.2") found components: Interpreter 
-- Cache files will be written to: C:/ncs/v2.3.0/zephyr/.cache
-- Zephyr version: 3.2.99 (C:/ncs/v2.3.0/zephyr)
-- Found west (found suitable version "0.14.0", minimum required is "0.7.1")
-- Board: nrf52840dk_nrf52840
-- Found host-tools: zephyr 0.15.2 (C:/ncs/toolchains/v2.3.0/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.15.2 (C:/ncs/toolchains/v2.3.0/opt/zephyr-sdk)
-- Found Dtc: C:/ncs/toolchains/v2.3.0/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6") 
-- Found BOARD.dts: C:/ncs/v2.3.0/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840.dts
-- Found devicetree overlay: D:/b-parasite-main/code/nrf-connect/samples/ble/nrf52840dk_nrf52840.overlay
-- Generated zephyr.dts: D:/b-parasite-main/code/nrf-connect/samples/ble/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: D:/b-parasite-main/code/nrf-connect/samples/ble/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: D:/b-parasite-main/code/nrf-connect/samples/ble/build/zephyr/dts.cmake
Parsing D:/b-parasite-main/code/nrf-connect/samples/ble/Kconfig
Loaded configuration 'C:/ncs/v2.3.0/zephyr/boards/arm/nrf52840dk_nrf52840/nrf52840dk_nrf52840_defconfig'
Merged configuration 'D:/b-parasite-main/code/nrf-connect/samples/ble/prj.conf'
Configuration saved to 'D:/b-parasite-main/code/nrf-connect/samples/ble/build/zephyr/.config'
Kconfig header saved to 'D:/b-parasite-main/code/nrf-connect/samples/ble/build/zephyr/include/generated/autoconf.h'

warning: UART_CONSOLE (defined at drivers/console/Kconfig:43) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: SERIAL (=n), SERIAL_HAS_DRIVER (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_UART_CONSOLE and/or look up UART_CONSOLE in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

-- The C compiler identification is GNU 12.1.0
-- The CXX compiler identification is GNU 12.1.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/ncs/toolchains/v2.3.0/opt/zephyr-sdk/arm-zephyr-eabi/bin/arm-zephyr-eabi-gcc.exe
CMake Warning at C:/ncs/v2.3.0/zephyr/CMakeLists.txt:824 (message):
  No SOURCES given to Zephyr library: drivers__console

  Excluding target from build.

Loading Zephyr default modules (Zephyr base (cached)).
-- Configuring done
-- Generating done
-- Build files have been written to: D:/b-parasite-main/code/nrf-connect/samples/ble/build
-- west build: building application
[1/214] Generating include/generated/version.h
-- Zephyr version: 3.2.99 (C:/ncs/v2.3.0/zephyr), build: v3.2.99-ncs2
[2/214] Generating misc/generated/syscalls.json, misc/generated/struct_tags.json
[3/214] Generating include/generated/syscall_dispatch.c, include/generated/syscall_list.h
[4/214] Generating include/generated/driver-validation.h
[5/214] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h, include/generated/otype-to-size.h
[6/214] Building C object zephyr/CMakeFiles/offsets.dir/arch/arm/core/offsets/offsets.c.obj
[7/214] Generating include/generated/offsets.h
[8/214] Building C object zephyr/CMakeFiles/zephyr.dir/C_/ncs/v2.3.0/modules/crypto/tinycrypt/lib/source/aes_decrypt.c.obj
[9/214] Building C object zephyr/CMakeFiles/zephyr.dir/C_/ncs/v2.3.0/nrfxlib/crypto/nrf_cc310_platform/src/nrf_cc3xx_platform_mutex_zephyr.c.obj
[10/214] Building C object zephyr/CMakeFiles/zephyr.dir/C_/ncs/v2.3.0/modules/crypto/tinycrypt/lib/source/utils.c.obj
[11/214] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/tracing/tracing_none.c.obj
[12/214] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/pm/state.c.obj
[13/214] Building C object zephyr/CMakeFiles/zephyr.dir/C_/ncs/v2.3.0/nrfxlib/crypto/nrf_cc310_platform/src/nrf_cc3xx_platform_abort_zephyr.c.obj
[14/214] Building C object zephyr/CMakeFiles/zephyr.dir/subsys/pm/device.c.obj
[15/214] Building C object zephyr/CMakeFiles/zephyr.dir/C_/ncs/v2.3.0/modules/crypto/tinycrypt/lib/source/aes_encrypt.c.obj
[... 257 lines cut ...]
      |         ^~~~~~~~~~~~~
C:\ncs\v2.3.0\zephyr\include\zephyr\drivers\gpio.h:317:24: note: in expansion of macro 'DT_GPIO_PIN_BY_IDX'
  317 |                 .pin = DT_GPIO_PIN_BY_IDX(node_id, prop, idx),                 \
      |                        ^~~~~~~~~~~~~~~~~~
C:\ncs\v2.3.0\zephyr\include\zephyr\drivers\gpio.h:352:9: note: in expansion of macro 'GPIO_DT_SPEC_GET_BY_IDX'
  352 |         GPIO_DT_SPEC_GET_BY_IDX(node_id, prop, 0)
      |         ^~~~~~~~~~~~~~~~~~~~~~~
[... cut ...]
D:\b-parasite-main\code\nrf-connect\prstlib\src\adc.c:25:22: note: in expansion of macro 'DT_NODELABEL'
   25 |     GPIO_DT_SPEC_GET(DT_NODELABEL(fast_disch), gpios);
      |                      ^~~~~~~~~~~~
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'c:\ncs\toolchains\v2.3.0\opt\bin\cmake.EXE' --build 'd:\b-parasite-main\code\nrf-connect\samples\ble\build'

 *  The terminal process terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

```

Thanks a lot.

rbaron, 2023-04-26:
@schejbity please take a look at the [docs I linked before](https://github.com/rbaron/b-parasite/wiki/How-to-Build-the-Firmware-Samples#with-nrf-connect-for-desktop--visual-studio-code). You're probably not selecting a b-parasite board in your build configuration, as you can see on the first line on your output: `--board nrf52840dk_nrf52840`. The b-parasite boards should be available now that you included the `prstlib` to your VSCode workspace.

Make sure you select the correct version for the nRF52 module you're using - example:

[image]

schejbity, 2023-04-27:
Oh, that's it.

Thanks a lot.

closed 2023-04-27 by schejbity