rbaron/b-parasite #124: VSCode - problem with include-files

Jones133D, 2023-05-03:
After I opened the workspace folder of the ble-sample with the `ble.code-workspace` script as described in the [wiki](https://github.com/rbaron/b-parasite/wiki/How-to-Build-the-Firmware-Samples#with-nrf-connect-for-desktop--visual-studio-code) I still have problems finding the include-paths.
[image]
[image]
Did I miss something? Do I have to additionally configure something else manually?

rbaron, 2023-05-03:
It looks like your `ble.code-workspace` file was changed from the one in the repo (`git status` should show that):

[image]

I'm not sure what happened there, but I would make sure the repository is clean (e.g.: `git clean -xfd`) and open the `ble.code-worskpace` file by either double clicking it or typing `code ble.code-workspace` in your terminal.

Jones133D, 2023-05-04:
Thank you for your help and your awesome project, I am really excited to get my parasites to work!

Unfortunately, I still have the same problem. Just to be sure, I just cloned the git again in an empty folder, opened the `ble.code-workspace` and clicked on the "Open Workspace"-Button. After that, the BLE (WORKSPACE) opened and startet compiling. I also noticed that automatically on line was added to the `ble.code-workspace`:
[image]

In the Explorer I can see the folder `prstlib` was successfully added, but in the source files all the include files still can't be found. 
Do I have to configure the IntelliSense-Settings (`c_cpp_properties.json`) to find missing headers like VSCode suggest in the popup here (sorry for having German selected here, I'm changing that right now)
[image]

rbaron, 2023-05-04:
The workspace looks correct, but I suspect you're not using the nRFConnect VSCode extension to build the code. [Here's how to do it (Nordic docs)](https://nrfconnect.github.io/vscode-nrf-connect/get_started/build_app_ncs.html#how-to-build-an-application-from-the-nrf-connect-sdk) (linked in the [Wiki](https://github.com/rbaron/b-parasite/wiki/How-to-Build-the-Firmware-Samples#with-nrf-connect-for-desktop--visual-studio-code) under the "... usual way to do it"). You can skip to step 8, since you're not creating a new application. Make sure to select the correct b-parasite board and revision as described in the Wiki

Let us know how it goes.

Jones133D, 2023-05-04:
Yes, in the screenshot above I forgot to add a build configuration and cmake startet to compile automatically without the nRFConnect VSCode extension. After added a Build Configuration again, it seems to build the hex-file correctly.
[image]

But, I still get errors about every include-file in the editor. Is this normal and I just have to ignore them or do I have to change some other settings?
Maybe in the Edit Build Configuration page of the nRFConnect VSCode extension? Is it correctly that it says `No fragments available` under `Kconfig fragments`? Do I have to choose `Use build system default` or `prj.conf` under Configuration? And is it right, that only this line is addet under `Extra CMake arguments`:
`-DBOARD_ROOT:STRING="c:/local_workspace/b-parasite_new/b-parasite/code/nrf-connect/prstlib"`

I also tried to add all the missing paths to the `c_cpp_properties.json`, which seemed to work fine first, but endet later by not finding only `alloca.h` (dependency of `rb.h` --> `sched_priq.h` --> `kernel_includes.h` --> `kernel.h` --> `led.h` and some other dependencys under led.h, everytime `alloca.h`). The funny thing is, this `alloca.h` really isn't on my entire PC disc.

rbaron, 2023-05-08:
Here's what my build edit tab looks like in a clean repository (never built):

[image]

Also in a clean directory (before building it), the extension also gives me include errors.

[image]

After clicking "Build configuration", the include errors are resolved:

[image]

And the build edit tab looks like this:

[image]

I suggest:
1. Double checking you're starting from a clean repository (e.g. `git clean -xfd`)
2. Creating the build config with default values (including build directory `build`)
3. Building the config and checking if the imports are resolved (and the build is successful)
4. Restarting VSCode

If that doesn't work, I would try building one of the built-in samples from Nordic as described in the [Nordic docs](https://nrfconnect.github.io/vscode-nrf-connect/get_started/build_app_ncs.html#how-to-build-an-application-from-the-nrf-connect-sdk) and see if that works for your environment.

rbaron, 2023-05-29:
Closing due to inactivity. @Jones133D Please feel free to reopen if you still need assistance.

closed 2023-05-29 by rbaron