rbaron/b-parasite #132: Non fuctional JLCPCB pcbs or other issues

azgael, 2023-06-06:
Dear All 
Does anyone else experienced the board not to even blink the blinky example after successful programing.
I have assembled one first board b-parasite version 2.0.0 and it worked still it behaves a bit strange, worked for a couple of days and then after I removed from the vase earth to check its assembly it stop sending data to home assistant. 
Meanwhile I have tried to assemble another 7 boards but I am facing a bigger problem, basically I can program successfully all of them still the led still does not blink (Blinky and Zigbee example), I have experience soldering and reviewed the board for shorts and cold solders and found none. now I start doubt from the radio modules with NRF52833, still I found very strange to have 7 failures in 8 radios. 
I just wonder if any one had similar problems?
Is it possible to have a mistaken passive and preventing the board to start, by looking at the schematics seams too simple fur such think to happen, anyway?
I have order the pcbs electrically tested but is it possible that they get damaged during soldering(hummmmm stupid question I am sure) or defect from the factory?

Sorry for the useless topics but I am totally desperate with 7 b-parasites not working and one that does when it feels like!
Thanks in advance,
BR
Azgael

rbaron, 2023-06-06:
Hi Azgael,

I would be very surprised if all 8 boards are defective due to a production error. It's more likely either an error in the submission (e.g.: wrong orientation of some component), manual soldering error (if you assembled the nRF module manually), or (hopefully!) some software config error.

Here's what I would do first:
1. Double check you're selecting both the correct board version _and_ revision when building the code in VSCode
2. Try to get some log out of it using the VSCode extension - [this is how I do it](https://nrfconnect.github.io/vscode-nrf-connect/reference/ui_sidebar_connected_devices.html#connected-devices-view)

azgael, 2023-06-06:
Dear rbaron 
First of all thanks a lot for sharing all this work, being available for the community is just impressive!!
Second thank you your quick reply, so I further explain, I have experience in soldering fine pitch electronica manually (down to coax cables on OV6948 chips :)) using our office professional equipment. With this in the back of my mind , I have ordered only the pcb's and fully populate them manually , including the HT Sensirion chip with the under path (done it before!). 
So I have manually fully solder the first semi working board, since it was working I have assembly additionally 6 boards which I have done by populate them manually excluding the battery contact and the rf module and re-flow them after in a company mini oven. After that I have check for shorts and cold solders, retouched and all looked clean. Them I  solder the RF module and the battery contact, So I have plug them to my segger mini edu and was able to program all of them(Zigbee example), still nothing waked up. Reviewed everything including orientation.
After I have solder one additional one more fully by hand extra carefully with components (thinking the re-flow may have damaged the board) funny is it programed and blinked for a period that the led become lower and off again, did not wake up again!!
Just this late afternoon I have reviewed the pcb layout and look more in detail about this underpad vias for the ground on pin 5 of the rf module, Hummm, (thought is possible that I have a week ground???) I pick one of the never working ones and soldered this pin directly to the ground pad and also a ground plane on the back of the pcb, funny is it started to blink and could catch it in the Zigbee2mqtt, after some minutes it lost communication, now has the same behavior as the first one, works when it feels like! 
I will spend some more time with it tomorrow, probably to reinforce all vias with air-wires I suspect that the fact that the vias are relatively thin for such a thick 2 layer pcb and the fact that they are under the soldering pad they may crack due to the iron heat applied on the radio module pads, this connection might be week for the full current drain of the module specially in the power up. So now my plan is to reinforce all the via connections. Hopefully I can avoid the hours of oscilloscope! 

As for the debug of the code I confess I have tried to connect the console but don't get anything out (not much time spent) additionally i have serious trouble on debug zephyr code using the nrf sdk with VScode, still I will try tomorrow to connect to get the log, still on my system I have no vcom only the segger edu mini RTT connection. 
Am I suppose to get anything on this RTT port, if so should I configure the baud-rate etc of it ?

The board Version I have selected is the 2.0.0 as my pcb's all looks fine on compiling (some errors and warnings on vscode but a clean built) !

Will keep you posted if this wiring works we could benefit from some review on the layout. 

Thank you once again!!
Azgael

rbaron, 2023-06-07:
> Am I suppose to get anything on this RTT port, if so should I configure the baud-rate etc of it ?

The defaults should work. I usually just click the RTT button in the extension. More rarely I also use the `JLinkRTTLogger` CLI from the terminal. (I also have no VCOM option, we don't define a UART logger)

azgael, 2023-06-09:
Hi rbaron 
Thanks for the feedback, I have check the logger via rtt and worked fine helping me to understand that on this misbehavior boards I had some issues with the HT sensor, again traced down to vias issues. 

Trying to debug the issue I have looked for shorts and bad soldering manually retouching all contacts on all boards but still did not recover the boards, after I have decided to assemble one board fully by hand (no reflow) and I was able to program it and it started to work normally. To further debug I have pick one of the previous boards and rewire all critical via pads (under-pad vias) and found out that the board begin to work properly. 
Concluding that there is an issue with the via connection between bottom and top layer, (in fact the via are 200um diameter not necessary small for breakage) following I have tested the electrical connections on non populated pcbs and the connection is present but it tends to break past the soldering process using standard re-flow temperature profile with peak temperature at 280C. 
Basically seams that there is thinner copper layer than expected the 18um on the vias, I would need xray to measure it so I just quit further investigation. Unfortunately I seams I have received pcbs that were miss processed by the manufacturer, I have presented a complain and they request me the return of the boards for further investigate, due to the cost of the pcb's the shipment would be more expensive than a new lot, so I have decided not to return them.

Probably when you order the complete assembled boards from the manufacturer this issues do not appear still seams to me that some review on the pcb design is recommendable.

So for my next order I will remake the layout by removing under-pad vias, probably increase via diameter (there is enough space!) and track wide, just in case!!

 Normally I use Altium but I will try to work with kicad, if I manage I will be happy to share the files just in case you want an updated version 2.0.1 :).

[image]
[image]

ps.: Ugly but it worked!!!

azgael, 2023-06-09:
Concluded that was a defect PCB lot, proposed some changes on pcb layout, I will review and reorder the pcbs.
Thank you a lot Rbaron for the debug instructions and further support!!
BR
Azgael

closed 2023-06-09 by azgael