rbaron/b-parasite #135: Try again...

LucidEye, 2023-06-08:
Really bad design, in fact, these look completely useless... they can not be waterproofed with the boot and reset buttons and that battery holder on the board like that.... and with the ESP32 integrated directly to the sensor, you can not completely bury the sensor because you would block the WiFI antenna under dirt. Also, why would anyone put an atmospheric temperature and humidity sensor on something that needs to be buried in the dirt and exposed directly to water?  You would never get any accurate atmospheric readings from it being buried in dirt rather than exposed to the air.  Are you expected to remove each sensor every time you water the plants?  A good soil moisture sensor needs to be able to be completely buried at least 7-8 inches under the soil surface to get any useful readings.  These will only test the soil on the surface, you need to test the soil down in the root zone.  Whoever "designed" this sensor wasn't really thinking... or isn't a real gardener...or both. Completely useless as anything other than a toy. 🙄

foreverimagining, 2023-06-08:
Some haters just want to hate.
Serious question though: What kind of gardener are you that you need to stick your hand 7-8 inches into the soil to see if it is moist?

LucidEye, 2023-06-08:
> Can't tell if trolling...

I'm being 100% serious.  This is one of the worst thought-out sensor designs I have ever seen.  I have been working with hydroponics for over 10 years. I have been a serious gardener for over 40 years.

LucidEye, 2023-06-08:
> Some haters just want to hate. Serious question though: What kind of gardener are you that you need to stick your hand 7-8 inches into the soil to see if it is moist?

How many roots do you see growing in the top 3-4 inches of soil in your plants?  Why would you measure soil moisture where there are no roots growing?

LucidEye, 2023-06-08:
> Do doctors typically have to shove a thermometer all the way up your ass to tell if you have a fever, or is that just a more accurate measure when there's uncertainty about your forehead? I expect this metaphor will be helpful for you given how far up your own ass you seem to be.

So you resort to personal attacks (and violating the code of conduct) instead of listening to reason from someone with a lifetime of real experience.  I think we can see who has the actual anal-cranial inversion syndrome issues here.

LucidEye, 2023-06-08:
> Ah, I see this is why we don't feed the trolls...

You're the only troll I see here.

foreverimagining, 2023-06-08:
Seems to me that if you can stick a finger down a couple inches to get a general reading of moisture past the crusty, dry top layer of soil, that an instrument could do the same thing, and because it's all relative anyways, you can determine whereabouts the reading from that instrument needs to be before you should water.

As for depth, I think this picture is sufficient, not that I would use a sensor on that plant. If I neglect it, it does well.
[image]

These moisture sensors are more often used for potted plants in generally indoor locations than outside in gardens (distance can be a problem as well as weather), in which case, a couple of inches of mulch is a little ridiculous. Since you're a hydroponics person and apparently interested in commercial-agriculture-level growing, I think you may have the wrong use case for this design. Your options, then, are to modify it to work for you or go elsewhere. The problem isn't the current design, but your intended use, which they are apparently not intended for.

LucidEye, 2023-06-08:
And what about real pots and planters for outdoor gardening and food production?
[image]
[image]
[image]
[image]

foreverimagining, 2023-06-08:
Again, not intended for your use case apparently. That doesn't mean it has a bad design. It means that the design won't work for you. So your options are still, modify it for yourself to make up for what you perceive are the shortcomings or go elsewhere and continue your search for something that matches your expectations. Like maybe those 12-inch long capacitive probes you were talking about. If the product you want is already out there in completed form, you should probably go with that.

LucidEye, 2023-06-08:
The idea would be to automate the above garden irrigation using a group of solar-powered sensor clusters, each capable of measuring and averaging soil moisture readings in several pots, and then telling the irrigation system to water only when the plants actually need it...
[image]
[image]
[image]
[image]

foreverimagining, 2023-06-08:
Yeah, that's along the lines of what everyone here is more or less trying to do with this project. It looks like you've got something going already, so why don't you pursue that rather than try to trash another project that might be following a different pathway for other different specific use cases? You could have started a constructive conversation hoping to find collaborators willing to consult and forward your goals with you, but instead you went on a rant claiming that everything is wrong, wrong, wrong.

LucidEye, 2023-06-08:
> Yeah, that's along the lines of what everyone here is more or less trying to do with this project. It looks like you've got something going already, so why don't you pursue that rather than try to [...]

Because I'm sick and tired of seeing so-called "developers" not actually putting any real-world thought into their "products" and selling junk that isn't really usable for solving real-world problems.
The industrial-grade agricultural sensors I mentioned above also come with an industrial-sized price tag.... several hundred dollars each for just the probes, and then you also have to buy their proprietary data-logger for another several hundred dollars.... (and their proprietary $oftware) not exactly practical or cost effective for the home gardener.
Then on the other end of the spectrum you have the "maker community" tinkerers like us and we are stuck with these dirt-cheap Chinese sensor boards from Amazon or AliExpress that are mostly DoA garbage.
There is really nothing in-between that is on an affordable scale for the average home user.  No one seems to even be thinking about designing some basic sensors like this that are designed well enough to be used in real-world, light duty applications in home-automation situations  Not everyone owns a 3D printer to make nice waterproof cases... most people use spray-on or dip-in conformal coatings for weather proofing.... so boards need to be designed with that in mind... clearly that is not the case here... so of what real use is this sensor board if it cant be use to automate any watering?

drspangle, 2023-06-08:
This isn't your personal blog, nobody here is shedding any tears for your plight -- especially after you come in here trashing the project and trashing the community that has contributed to it. Frankly what you're doing is worse than trolling. Issues aren't the appropriate forum for complaining (without any basis) about how wrong everything is and making demands for your own bespoke product. Nobody on this project gets paid for their contributions nor is anyone obliged to take you seriously when you're immediately aggressive and disrespectful from the outset.

If you don't like what this project is doing, move on. You are not contributing by doing what you're doing here and nobody is obligated to do anything about your complaining.

foreverimagining, 2023-06-08:
You do realize this is an open-source project, right? This is in no way a finished product, and nobody is selling them. You have to make or have them made for yourself, and all the necessary information is provided for free. You seem to be under the mistaken impression that your use case is the only one that matters, as well. This is actually a pretty fantastic middle-of-the-road, work-in-progress project. We're all dealing with the same unfortunate triangle that is one of the most annoying things inherent to our universe, and we can all commiserate on that part. You can have two out of three: a good product, a finished product, or a cheap product. You cannot have all three. This is a good project that is relatively cheap (or at least cheaper than the alternatives) that you have to put a little bit of work into. If it fits your use case, you just have to put in the work making them. If they don't quite fit what you want, you should have most of the information you need here to be able to adapt it and then go about making it. 

That's what you should be doing: stop complaining and do something about it.

I'll even be a little helpful since you don't seem to have really dived very deeply into this project. This is Home Assistant: https://www.home-assistant.io/. It's a home automation program. It can be used with b-parasites and irrigation systems (homemade or many on the market) to automate your watering schedule with just a little work. It even should work with your ESP32s using: https://www.home-assistant.io/integrations/esphome/ or perhaps MQTT: https://www.home-assistant.io/integrations/mqtt/. I haven't gotten into them yet so I can't say for sure. Here are the on-the-market irrigation systems automatically integratable with Home Assistant: https://www.home-assistant.io/integrations/#irrigation, but you could totally go DIY. Make up a simple (or not so simple if you want) automation joining the two parts, and boom, automated. 

As for 3D printing, a printer is a fantastic investment, and it has never been easier to learn to use them. That said, you don't actually need one as there are plenty of online 3D printing services, some of which are relatively cheap. If you had been nicer, you may also have found someone on here that would have been willing to help you out. I doubt you would even need a 3D printed case per se for your use case as it appears that you basically just need to modify the PCB board to be much longer, and then, remove nearly all of the hardware to another board. I'm not that far into electronics work yet, but seems pretty easy to me.

I'm not going to touch the conformal coatings stuff because there is already an issue open about that, and it is pretty much all experimental at this point.

I do, however, agree with @drspangle that this is not the proper forum for this, and it's time to move on. You should fork the project to work on for your own purposes and build upon the foundations of those that came before you, or at the very least, stop complaining here.