rbaron/b-parasite #138: Samples code compatibility with old hardware version

AlbanDurrheimer, 2023-06-12:
Hello,

Firstly, thank you for this project, I assembled my first board last week.

Concerning my question, I would like to know if the latest versions of the samples code (currently in the main branch) are compatible with older hardware versions (personally, I have 1.2.0 boards)?

I'd also like to know if the samples code specify a version somewhere? I've noticed that this information isn't passed on by the Zigbee code, and this would make it easy to find out if the firmware is up to date.

Thank you

rbaron, 2023-06-12:
Hi @AlbanDurrheimer,

As of now all the samples are compatible with all hardware versions from 1.0.0 to 2.0.0, with the exception of `samples/input`, because it showcases the push button that only 2.0.0 has.

We have a way of fetching the board revision via the `CONFIG_BOARD_REVISION` Kconfig. For example [here's where it's defined for nRF52840@1.2.0](https://github.com/rbaron/b-parasite/blob/426195184d2433f06f4b30ea665348985cdaffb3/code/nrf-connect/prstlib/boards/arm/bparasite_nrf52840/bparasite_nrf52840_1_2_0.conf#L1), but it's not currently used by any sample.

rbaron, 2023-06-14:
That's a good idea, I'll send a PR shortly. My original goal when setting the `CONFIG_BOARD_REVISION` was to put it in the `HWVersion` attribute, but it expects an integer version. We use a semantic ID type of versioning, and I'd rather not have two things meaning the same thing. I think a reasonable compromise is to put it together in the model identifier, like `"b-parasite 1.2.0"`.

[image]

AlbanDurrheimer, 2023-06-14:
Yes, I can see the problem. However, won't the solution with the model identifier create problems with the converter for zigbee2MQTT?

rbaron, 2023-06-14:
@AlbanDurrheimer that's a good point, I forgot about z2m. I closed #139.

rbaron, 2023-06-15:
I wanted to avoid having a different revision ID, but that may be a reasonable compromise. I opened #140 for it.

AlbanDurrheimer, 2023-06-15:
Alternatively, you could use the `SWBuildID` attribute to make a compound version from the hardware version and the firmware version (example: `[hardware version]-[firmware version]`, or vice versa)?  
(I don't know much about Zigbee, so sorry if this isn't relevant)

commit afa53f0e (2023-06-15): Introduce CONFIG_BOARD_REVISION_CODE

Beside having a `CONFIG_BOARD_REVISION` Kconfig string that represents
our semantic versioning for board revisions, the ZigBee sample requires
an integer representation of it. Instead of conditioning the integer
version on the string version only for that sample and having to touch
the ZigBee sample whenever we add a revision, I decided to have a
board-wide config, so that both string and int versions live close to
each other.

Fixes #138.
closed 2023-06-15 by rbaron