rbaron/b-parasite #142: Wiki instructions for docker build are outdated

Artanicus, 2023-06-19:
Would've made a PR but the wiki editing seems to be locked down. Here's a patch instead:

~~~
diff --git a/How-to-Build-the-Firmware-Samples.md b/How-to-Build-the-Firmware-Samples.md
index 648fa14..9e70284 100644
--- a/How-to-Build-the-Firmware-Samples.md
+++ b/How-to-Build-the-Firmware-Samples.md
@@ -10,7 +10,7 @@ We support [different nRF52 SoCs in different hardware versions](https://github.
 # With Docker
 To build the `ble` sample for the `nRF52840`, revision `1.2.0`, `cd` into `code/nrf-connect/` and run:
 ```
-$ docker run --rm -v ${PWD}:/code -w /code/samples/ble nordicplayground/nrfconnect-sdk:v2.2-branch west build --build-dir ./build --pristine --board bparasite_nrf52840@1.2.0
+docker run --rm -it -v ${PWD}/code:/code -w /code/nrf-connect/samples/ble nordicplayground/nrfconnect-sdk:v2.2-branch west build --build-dir ./build --pristine --board bparasite_nrf52840@1.2.0
 ```
 The output will be in `code/nrf-connect/samples/ble/build/zephyr/zephyr.hex`.
~~~

rbaron, 2023-06-19:
Hi @Artanicus, 

I think the issue is that the existing command must be run in the `code/nrf-connect/` directory (as written above there `[...] `cd` into `code/nrf-connect/` and run:`). With the change, the command must be run in the root of the repo, and that makes the line above it incorrect. The other change (`-it`) I believe has no effect, since we don't need an interactive shell.

Please let me know if I understood it correctly.

Artanicus, 2023-06-19:
You're absolutely right, I just was so used to running builds from the root of a project and didn't notice the guidance above the code snippet!

- If you prefer running the builds from deeper in the structure instead at the root, I would recommend moving the PWD change instead into the snippet as `cd code/nrf-connect` to make it more explicit and harder to miss.
- The `-it` under perfect circumstances has no effect, but if anything goes wrong it ensures the user sees the output and has control over it vs. output getting lost somewhere in a background task.

rbaron, 2023-06-20:
Great idea. I updated the instructions and made it more copy-and-paste friendly. Tks!

closed 2023-06-20 by rbaron

drspangle, 2023-12-28:
I hate to reraise a dead issue, but the current build instructions don't work and I actually used this suggested change as inspiration to fix them. When I simply copy and pasted the commands the docker image that was pulled didn't seem to have zephyr set up properly at all, and the `west` tool didn't have a `build` command. So instead I ran an interactive bash shell (rather than running the actual command):

```bash
# Make sure you start in the correct directory as the documentation currently states.
cd code/nrf-connect/

# Rather than running everything in one command, we're going to need to start an interactive shell.
docker run -it --rm -v ${PWD}:/code -w /code/samples/ble nordicplayground/nrfconnect-sdk:v2.5-branch bash

# Within the nordicplayground/nrfconnect-sdk:v2.5-branch container's shell, execute the following...
# This will take a while... 
# The nordicplayground/nrfconnect-sdk:v2.5-branch container seemingly doesn't have many of the deps set up already.
west init

# Now build the binary for the b-parasite.
# Remember to put your hardware revision at the end of this command, I happen to be using 1.2.0.
west build --build-dir ./build --pristine --board bparasite_nrf52840@1.2.0
```

When I did it this way it output the file as expected. It seemed the `west init` was necessary (and required an interactive shell) to get everything set up. I had expected the container image to have this already set up, but it clearly wasn't.

rbaron, 2023-12-28:
@drspangle indeed it seems like the Docker image changed how it handles freestanding Zephyr applications.

From [their docs](https://github.com/NordicPlayground/nrf-docker?tab=readme-ov-file#build-the-firmware) (under "freestanding application"), I think we don't have to run `west init` nor `west update`, but we can do it like this:

From the repo's root:
```
docker run --rm -v ${PWD}:/repo nordicplayground/nrfconnect-sdk:v2.5-branch west build --pristine --board bparasite_nrf52840@2.0.0 --build-dir /repo/code/nrf-connect/samples/ble/build /repo/code/nrf-connect/samples/ble
```

This is basically what I set up in the [GitHub action](https://github.com/rbaron/b-parasite/blob/37ea3a10622728b989bb62faf0f8d2a456f2aebd/.github/actions/build/action.yml#L20) that runs on every commit.

Could you try this one and let me know if it works for you too? I will then update the wiki.

drspangle, 2023-12-28:
@rbaron Confirming that the new command works. I purged my local docker image cache and cloned the repo from scratch, then executed the command. It took only a minute. Definitely saves a huge amount of time pulling down all the deps into the docker container. Great!