rbaron/b-parasite #144: Cannot connect to b-parasite using J-Link Commander

iwko, 2023-06-26:
Hello!

I am using J-Link EDU Mini to flash firmware into b-parasite 2.0. When I try to connect using J-Link Commander I get such an error.

```
SEGGER J-Link Commander V7.80c (Compiled Sep 27 2022 16:07:44)
DLL version V7.80c, compiled Sep 27 2022 16:06:20

Connecting to J-Link via USB...O.K.
Firmware: J-Link EDU Mini V1 compiled Mar 28 2023 17:01:59
Hardware version: V1.00
J-Link uptime (since boot): 0d 00h 21m 39s
S/N: 801048005
License(s): FlashBP, GDB
USB speed mode: Full speed (12 MBit/s)
VTref=3.299V

Type "connect" to establish a target connection, '?' for help
J-Link>connect
Please specify device / core. <Default>: CORTEX-M4
Type '?' for selection dialog
Device>
Please specify target interface:
  J) JTAG (Default)
  S) SWD
  T) cJTAG
TIF>S
Specify target interface speed [kHz]. <Default>: 4000 kHz
Speed>
Device "CORTEX-M4" selected.

Connecting to target via SWD
Failed to attach to CPU. Trying connect under reset.
Cannot connect to target.
J-Link>
```

I soldered jump wires to pins according to the instructions here https://github.com/rbaron/b-parasite/wiki/How-to-Flash-the-Firmware-Samples#using-a-cr2032-battery. Because my jump wires did not fit into j-link pin cable I cut some wire from resistors and soldered them to jump wires and then put them into the pin cable. Anyone had a similar problem?

rbaron, 2023-06-27:
hi @iwko. It looks like the JLink is correctly identified and the target is correctly powered by looking at the `VTref=3.299V` line.

One difference from my output is the `device / core` - it's identified as a `NRF52833_XXAA` for me, so you can try selecting that (or `NRF52840_XXAA` if you're using that):

```
% JLinkExe
SEGGER J-Link Commander V7.80c (Compiled Sep 27 2022 16:02:33)
DLL version V7.80c, compiled Sep 27 2022 16:02:25

Connecting to J-Link via USB...O.K.
Firmware: J-Link EDU Mini V1 compiled Mar 28 2023 17:01:59
Hardware version: V1.00
J-Link uptime (since boot): 0d 00h 10m 56s
S/N: 801036382
License(s): FlashBP, GDB
USB speed mode: Full speed (12 MBit/s)
VTref=2.504V

Type "connect" to establish a target connection, '?' for help
J-Link>connect
Please specify device / core. <Default>: NRF52833_XXAA
Type '?' for selection dialog
Device>NRF52833_XXAA
Please specify target interface:
  J) JTAG (Default)
  S) SWD
  T) cJTAG
TIF>S
Specify target interface speed [kHz]. <Default>: 4000 kHz
Speed>
Device "NRF52833_XXAA" selected.

Connecting to target via SWD
InitTarget() start
InitTarget() end
Found SW-DP with ID 0x2BA01477
DPIDR: 0x2BA01477
CoreSight SoC-400 or earlier
Scanning AP map to find all available APs
AP[2]: Stopped AP scan as end of AP map has been reached
AP[0]: AHB-AP (IDR: 0x24770011)
AP[1]: JTAG-AP (IDR: 0x02880000)
Iterating through AP map to find AHB-AP to use
AP[0]: Core found
AP[0]: AHB-AP ROM base: 0xE00FF000
CPUID register: 0x410FC241. Implementer code: 0x41 (ARM)
Found Cortex-M4 r0p1, Little endian.
FPUnit: 6 code (BP) slots and 2 literal slots
CoreSight components:
ROMTbl[0] @ E00FF000
[0][0]: E000E000 CID B105E00D PID 000BB00C SCS-M7
[0][1]: E0001000 CID B105E00D PID 003BB002 DWT
[0][2]: E0002000 CID B105E00D PID 002BB003 FPB
[0][3]: E0000000 CID B105E00D PID 003BB001 ITM
[0][4]: E0040000 CID B105900D PID 000BB9A1 TPIU
[0][5]: E0041000 CID B105900D PID 000BB925 ETM
Cortex-M4 identified.
J-Link>
```

I very rarely use this commander myself. If you're a beginner, I'd suggest using the VSCode extension for flashing / debugging. There's a nice tutorial [in the nrf docs](https://nrfconnect.github.io/vscode-nrf-connect/).

Good luck!

iwko, 2023-06-27:
I've tried to change the device in J-Link Commander to the same device as you but without a positive outcome. I also tried other nrf family devices but it had no effect. Then I tried to us VSC but i get error

```
C:\Windows\system32\cmd.exe /d /s /c "west flash -d c:\Users\admin\PycharmProjects\b-parasite\code\nrf-connect\samples\ble\build --skip-rebuild --dev-id 801048005 --erase"

-- west flash: using runner nrfjprog
-- runners.nrfjprog: mass erase requested
-- runners.nrfjprog: Flashing file: c:\Users\admin\PycharmProjects\b-parasite\code\nrf-connect\samples\ble\build\zephyr\zephyr.hex
[error] [ Client] - Encountered error -102: Command read_device_info executed for 138 milliseconds with result -102
[error] [ Worker] - An unknown error.
[error] [ Client] - Encountered error -102: Command read_memory_descriptors executed for 31 milliseconds with result -102
Failed to read device memories.
[error] [ Worker] - An unknown error.
ERROR: JLinkARM DLL reported an error. Try again. If error condition
ERROR: persists, run the same command again with argument --log, contact Nordic
ERROR: Semiconductor and provide the generated log.log file to them.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
FATAL ERROR: command exited with status 33: nrfjprog --program 'c:\Users\admin\PycharmProjects\b-parasite\code\nrf-connect\samples\ble\build\zephyr\zephyr.hex' --chiperase --verify -f NRF52 --snr 801048005
```