rbaron/b-parasite #145: Zigbee code does not update value

AlbanDurrheimer, 2023-06-27:
Hello,

I have some b-parasite boards and several it does not update value after some hours (almost every 2 days).  
To fix the problem, I have to restart the device by removing and reinserting the battery.

[image]

Does this happen to other people? And do you have a solution?

### Additional information:

B-Parasite Board: 1.2.0
Firmware: latest firmware in main branch (f6b46000448239a9e5288051953c34e94e617c5f)
Coordinator: Home Assistant ([2023.6.2](https://github.com/home-assistant/core/releases/tag/2023.6.2)) with Zigbee2MQTT ([1.31.2](https://github.com/Koenkk/zigbee2mqtt/releases/tag/1.31.2))

rbaron, 2023-06-27:
@AlbanDurrheimer this is unfortunately an open issue with the ZigBee sample. Recent attempts to fix and discussions are in #126, #130 and more recently on #136. We're having a hard time debugging this, as the problem sometimes happen in a few days, a few weeks, and on the other hand some of my sensors are running for multiple months without issues.

Do you have multiple b-parasites running ZigBee? Do they all experience this issue around the same time?

AlbanDurrheimer, 2023-06-27:
Thank you @rbaron for your reply.

Yes, I have three cards at the moment (but more are being soldered) with zigbee firmware. And no, they don't produce the problem at the same time, I have one that currently produces the problem every day or so and the others have also already produced the error, but for the moment it's calmed down.  

If I can help you find the source of this problem, please don't hesitate to let me know and explain how to proceed.

rbaron, 2023-06-27:
Thanks @AlbanDurrheimer. Were the batteries drained from any of the sensors after the problem occurred? Or did just resetting them fix it?

AlbanDurrheimer, 2023-06-27:
> Thanks @AlbanDurrheimer. Were the batteries drained from any of the sensors after the problem occurred? Or did just resetting them fix it?

@rbaron I haven't had any drained batteries, just one that I noticed was half empty, but I wonder if I'd used it before, so I don't know if it's related... In my case, a simple restart (removing-inserting the battery only once) is enough to get the sensor working again for at least 1 day.

AlbanDurrheimer, 2023-06-28:
It's happened again, and this time neither the reboot nor the factory reset was able to update the sensors values. The module just reconnected to the coordinator, but didn't send back any information from the sensors. I had to run a reconfiguration from Zigbee2MQTT to get the sensor to update again.

Regressor, 2023-07-05:
> It's happened again, and this time neither the reboot nor the factory reset was able to update the sensors values. The module just reconnected to the coordinator, but didn't send back any [...]

I've seen this behaviour with smart plug - it just stops sending measurements and the only way to fix this was a reconfiguration. It looks like this is a zigbee2mqtt problem not b-parasite.

AlbanDurrheimer, 2023-07-05:
> I've seen this behaviour with smart plug - it just stops sending measurements and the only way to fix this was a reconfiguration. It looks like this is a zigbee2mqtt problem not b-parasite.

Thanks for your feedback, I'll test this the next time it produces the problem.

AlbanDurrheimer, 2023-07-10:
I carried out several tests and here are the findings:  

### 1. Send a reconfiguration
 > I've seen this behaviour with smart plug - it just stops sending measurements and the only way to fix this was a reconfiguration. It looks like this is a zigbee2mqtt problem not b-parasite.

The board does not respond to the command (timeout).

### 2. Restart Zigbee2MQTT
No new data sent by the board.

### 3. Restart the b-parasite board
The board joins the Zigbee network again, but no data is sent back.

### 4. Restart the b-parasite board with Zigbee2MQTT enable joining
The board joins the Zigbee network again, but no data is sent back.

### 5. Reset the b-parasite board
The board does not join the Zigbee network.

### 6. Reset the b-parasite board with Zigbee2MQTT enable joining
The board joins the Zigbee network, but no data is sent back.

### 7. Remove the devices to Zigbee2MQTT network then enable joining
The board joins the Zigbee network, but no data is sent back.

### 8. Remove the devices to Zigbee2MQTT network and reset the board then enable joining
The board joins the Zigbee network, but no data is sent back.

### 9. Move the plant to the same room as the coordinator

Once it was working again, I moved the board to the same room as the coordinator (over a weekend). There was no problem until I moved the board back to its old location, where it crashed in less than 12 hours. (I should point out that in both rooms, the signal is greater than 200 lqi).

### 10. Conclusion

To conclude, I can no longer remove the error from my board. And yet, it still responds to commands such as `genBasic/hwVersion`.

rbaron, 2023-07-10:
Thanks for the detailed testing @AlbanDurrheimer. It's odd that the board doesn't work after a reboot. This I haven't experienced so far. Could it be some hardware issue when reading the sensors? Maybe some debug logging could shed some light into it. Did you experience it with multiple boards?

There's also #146 that fixes a bunch of issues with the zigbee firmware. It's too early to tell if they're gonna help in the long term, but so far I'm running it for 2w with no issues.

AlbanDurrheimer, 2023-07-10:
This is the same card with which I'd encountered problems earlier. Every time it crashed, a reboot or reset was enough to get it working again. It also worked all last weekend indoors, then I moved the plant outside for a night, then nothing

PR #150 (2023-07-23): [zigbee sample] Fix rare infinite loop bug & introduce watchdog
I think I finally managed to track down and fix the actual bug from https://github.com/rbaron/b-parasite/issues/113#issuecomment-1484903062, #126, #130, #145, #146.

Based on [this SHTC3 lib](https://github.com/adafruit/Adafruit_SHTC3/blob/ec670f90e4b523196afdd614223321fb1507ea2f/Adafruit_SHTC3.cpp#L153), we had the following code in [shtc3.c](https://github.com/rbaron/b-parasite/blob/95ec660cb6bdbe4c4dfd8ebc5f98a1141811ec9d/code/nrf-connect/prstlib/src/shtc3.c#L37):

```
  while (i2c_read_dt(&shtc3, buff, 6) != 0) {
    k_msleep( [... cut]
closed 2023-07-23 by rbaron